coolsnowwolf/lede #3029: 编译出错

xiexianghua, 2020-02-10:
make[4]: Entering directory `/home/xiexianghua95/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/u-boot-orangepi_zero/u-boot-2019.01'
  CHK     include/config/uboot.release
  CHK     include/generated/version_autogenerated.h
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  CHK     include/generated/generic-asm-offsets.h
  CHK     include/generated/asm-offsets.h
  HOSTCC  tools/mkenvimage.o
  HOSTLD  tools/mkenvimage
  HOSTCC  tools/fit_image.o
  HOSTCC  tools/image-host.o
  HOSTCC  tools/lib/crc16.o
In file included from tools/lib/crc16.c:1:0:
./tools/../lib/crc16.c: In function 'crc16_ccitt':
./tools/../lib/crc16.c:70:2: error: 'for' loop initial declarations are only allowed in C99 mode
  for (int i = 0;  i < len;  i++)
  ^
./tools/../lib/crc16.c:70:2: note: use option -std=c99 or -std=gnu99 to compile your code
make[5]: *** [tools/lib/crc16.o] Error 1
make[4]: *** [tools] Error 2
make[4]: Leaving directory `/home/xiexianghua95/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/u-boot-orangepi_zero/u-boot-2019.01'
make[3]: *** [/home/xiexianghua95/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/u-boot-orangepi_zero/u-boot-2019.01/.built] Error 2
make[3]: Leaving directory `/home/xiexianghua95/lede/package/boot/uboot-sunxi'
time: package/boot/uboot-sunxi/orangepi_zero/compile#0.68#0.17#1.39
make[2]: *** [package/boot/uboot-sunxi/compile] Error 2
make[2]: Leaving directory `/home/xiexianghua95/lede'
make[1]: *** [/home/xiexianghua95/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/xiexianghua95/lede'
make: *** [world] Error 2

danielwcd2, 2020-02-10:
the same!

make[4]: Entering directory `/home/chaodawu/lede/lede/build_dir/target-aarch64_cortex-a53_musl/u-boot-nanopi_neo2/u-boot-2019.01'
  CHK     include/config/uboot.release
  CHK     include/generated/version_autogenerated.h
  CHK     include/generated/timestamp_autogenerated.h
  CHK     include/generated/generic-asm-offsets.h
  CHK     include/generated/asm-offsets.h
  HOSTCC  tools/lib/crc16.o
In file included from tools/lib/crc16.c:1:0:
./tools/../lib/crc16.c: In function 'crc16_ccitt':
./tools/../lib/crc16.c:70:2: error: 'for' loop initial declarations are only allowed in C99 mode
  for (int i = 0;  i < len;  i++)
  ^
./tools/../lib/crc16.c:70:2: note: use option -std=c99 or -std=gnu99 to compile your code
make[5]: *** [tools/lib/crc16.o] Error 1
make[4]: *** [tools] Error 2
make[4]: Leaving directory `/home/chaodawu/lede/lede/build_dir/target-aarch64_cortex-a53_musl/u-boot-nanopi_neo2/u-boot-2019.01'
make[3]: *** [/home/chaodawu/lede/lede/build_dir/target-aarch64_cortex-a53_musl/u-boot-nanopi_neo2/u-boot-2019.01/.built] Error 2
make[3]: Leaving directory `/home/chaodawu/lede/lede/package/boot/uboot-sunxi'
time: package/boot/uboot-sunxi/nanopi_neo2/compile#0.34#0.15#1.83
make[2]: *** [package/boot/uboot-sunxi/compile] Error 2
make[2]: Leaving directory `/home/chaodawu/lede/lede'
make[1]: *** [/home/chaodawu/lede/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/chaodawu/lede/lede'
make: *** [world] 错误 2

AmadeusGhost, 2020-02-11:
你们的目标设备是什么？我ubuntu18.04随便编译了一个sunxi下的设备是过的

xiexianghua, 2020-02-11:
Orangepi zero plus

AmadeusGhost, 2020-02-11:
> Orangepi zero plus

[image]
ubuntu18.04一遍编译过

danielwcd2, 2020-02-11:
我的是NanoPi Neo2
系统Ubuntu  14.04 LTS x64

AmadeusGhost, 2020-02-11:
> 我的是NanoPi Neo2
> 系统Ubuntu 14.04 LTS x64

换ubuntu 18.04，出错再说

danielwcd2, 2020-02-11:
18.04最新代码可以编译过。
之前尝试在14.04上回退到了1月31号的版本可以编译过，最新代码就编译不过了

ch1993128, 2020-02-12:
下载不完整，工具和工具链编译出错
在make menuconfig以后
make download  -j4 V=s执行多次，直到所有包下载完成

closed 2020-02-12 by xiexianghua