coolsnowwolf/lede #3031: 希望大佬增加RK3328的支持

kl9800, 2020-02-10:
希望大佬增加对RK3328的支持，想用NanoPi R2s，体积超级小节省了大量空间

kl9800, 2020-02-10:
> 希望大佬增加对RK3328的支持，想用NanoPi R2s，体积超级小节省了大量空间

由于目前这个只有厂家编好的固件，不是很信任，原生openwrt再装一堆插件感觉很麻烦

lqhorochi, 2020-02-11:
> > 希望大佬增加对RK3328的支持，想用NanoPi R2s，体积超级小节省了大量空间
> 
> 由于目前这个只有厂家编好的固件，不是很信任，原生openwrt再装一堆插件感觉很麻烦

需要源码支持.你找一下源码给大雕吧...地址等等

kl9800, 2020-02-11:
> > > 希望大佬增加对RK3328的支持，想用NanoPi R2s，体积超级小节省了大量空间
> > 
> > 
> > 由于目前这个只有厂家编好的固件，不是很信任，原生openwrt再装一堆插件感觉很麻烦
> 
> 需要源码支持.你找一下源码给大雕吧...地址等等

去哪找源码？

ch1993128, 2020-02-12:
nano和树莓这种需要外挂无线网卡才好用，否则只能做旁路由，不推荐。。无线信号没天线太差了

hfharry, 2020-02-13:
> nano和树莓这种需要外挂无线网卡才好用，否则只能做旁路由，不推荐。。无线信号没天线太差了

NanoPi R2s 是双千兆端口的4核 A53 1.5GHz 开发板，体积极小，性价比高，可以配套交换机作为有线主路由使用。

coolsnowwolf, 2020-02-13:
OP 本身并没有支持这个，现在是厂商的 SDK ，不兼容

closed 2020-02-14 by kl9800

hanlihanshaobo, 2020-06-28:
> OP本身并没有支持这个，现在是厂商的SDK，不兼容

现在有人作了原生的，希望大佬支持下