coolsnowwolf/lede #3033: 进不去ssrplus页面

bjzhou, 2020-02-11:
点进去就报错

/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute function dispatcher target for entry '/admin/services/shadowsocksr'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/shadowsocksr/client'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/shadowsocksr/client.lua:18: bad argument #1 to 'upper' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

coolsnowwolf, 2020-02-11:
不要继承配置

bjzhou, 2020-02-11:
@coolsnowwolf 我的ssr订阅链接订阅不了，请问是链接的问题还是插件的问题，其他工具可以订阅

2020-02-11 16:40:44 发生错误, 正在恢复服务
	[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:354: in main chunk
	[C]: in function 'xpcall'
	/usr/share/shadowsocksr/subscribe.lua:307: in function </usr/share/shadowsocksr/subscribe.lua:226>
	[C]: in function 'foreach'
	/usr/share/shadowsocksr/subscribe.lua:356: in function </usr/share/shadowsocksr/subscribe.lua:354>
2020-02-11 16:40:44 stack traceback:
2020-02-11 16:40:44 /usr/share/shadowsocksr/subscribe.lua:319: attempt to concatenate field 'alias' (a nil value)
2020-02-11 16:40:44 成功解析节点数量: 39

oowizoo, 2020-02-12:
我也遇到类似的问题。这两天突然进不去SSR+的管理页面，错误提示：
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute function dispatcher target for entry '/admin/services/shadowsocksr'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/shadowsocksr/client'.
The called action terminated with an exception:
**/usr/lib/lua/luci/model/cbi/shadowsocksr/client.lua:18: bad argument #1 to 'upper' (string expected, got nil)**
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>
发现似乎是参数出现错误？是否更订阅的节点信息格式变化了相关

oowizoo, 2020-02-12:
报错位置的函数似乎与显示节点服务器相关，s.type出现错误，似乎变得不是string了？
uci:foreach(shadowsocksr, "servers", function(s)
        if s.alias then
                **server_table[s[".name"]] = "[%s]:%s" %{string.upper(s.type), s.alia**
        elseif s.server and s.server_port then
                server_table[s[".name"]] = "[%s]:%s:%s" %{string.upper(s.type), s.s
        end
end)

oowizoo, 2020-02-12:
问题已找到，而订阅的服务器最新一次更新后，服务器地区和别合为了一个字符串“[中文地区]+[别名]”，因此type参数失效，string.upper()就更加无用，只剩alias一个参数，因此导致此情况。
解决方法，修改client.lua：
uci:foreach(shadowsocksr, "servers", function(s)
        if s.alias then
                **server_table[s[".name"]] = "%s" %{s.alias}**
        elseif s.server and s.server_port then
                **server_table[s[".name"]] = "%s:%s" %{ s.server, s.server_port}**
        end
end)

coolsnowwolf, 2020-02-12:
没有 type 就运行不起来了，这个是机场的锅，让他们修复去

closed 2020-02-12 by coolsnowwolf

miroda, 2020-02-15:
> 问题已找到，订阅的服务器最新一次更新后，type参数失效，只剩alias一个参数，因此导致此情况。
> 解决方法，删除s.type参数，修改client.lua：
> uci:foreach(shadowsocksr, "servers", function(s)
> if s.alias then
> **server_table[s[".name"]] = "%s" %{s.alias}**
> elseif s.server and s.server_port then
> **server_table[s[".name"]] = "%s:%s" %{ s.server, s.server_port}**
> end
> end)
> 【此方法已知问题】：此方法会对一些采用原本的格式命名的订阅服务器产生影响，造成无法正常显示。
> 由于本人毫无LUA经验，希望作者能提供更好的解决方案。

那问题是，修复完成之后，我们怎么用啊