coolsnowwolf/lede #3035: NAT环回不生效

MarsGao, 2020-02-11:
防火墙→端口转发/etc/config/firewall中，NAT环回默认开启，但不生效。不知道是不是firewall写iptable时有问题。

yangsongli, 2020-02-11:
某个插件导致的 你自己排查

coolsnowwolf, 2020-02-11:
和 NET NS 内核选项有关，Docker 会自动选上这个

BROBIRD, 2020-02-11:
@coolsnowwolf `CONFIG_KERNEL_NET_NS`这一项需要设置为`n`吗？

trackycn, 2020-02-12:
我这边测试, 如果勾选 Docker, NAT 环回全部失效。

如果不勾选的话, 内外网相同端口的可以成功, 不相同的不行。

关闭 Flow Offloading 后, 一切正常

helllkz, 2020-02-12:
@trackycn 我也是这个问题，但是关闭Flow offloading过后，过段时间还是会失效，然后就是再启用然后禁用

trackycn, 2020-02-12:
> @trackycn 我也是这个问题，但是关闭Flow offloading过后，过段时间还是会失效，然后就是再启用然后禁用

我关了一切正常, 测试了六天了, 你如果关了还不行, 试试看在路由器里tcpdump抓包看看请求是在哪里断了。我之前打开 Flow offloading, 抓包发现不同端口下的 NAT 回环只有第一个 HTTP 请求有响应, 之后的都不回了。

helllkz, 2020-02-13:
> > @trackycn 我也是这个问题，但是关闭Flow offloading过后，过段时间还是会失效，然后就是再启用然后禁用
> 
> 我关了一切正常, 测试了六天了, 你如果关了还不行, 试试看在路由器里tcpdump抓包看看请求是在哪里断了。我之前打开 Flow offloading, 抓包发现不同端口下的 NAT 回环只有第一个 HTTP 请求有响应, 之后的都不回了。

@trackycn 一样的，我看nginx的日志也只有一个访问，之后的都没了

closed 2020-02-16 by coolsnowwolf