coolsnowwolf/lede #3041: R7800潘多拉R9.5.1 5GHz断流变信道且后面无法固定信道。

tangdashen1996, 2020-02-11:
系统日志：
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.508647] pdev attach 00000000 -1 
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.512206] 
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.512206] <=== cfg max peer id 1056 ====>
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.518270] htt_peer_map_timer_init Enter pdev d3a50000 hrtimer d3a54948
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.524760] 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.524760]  htt_alloc_peer_map_mem : Alloc Success : host q vaddr d4ce4000 paddr 56ce4000
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.534439] 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.534439]  htt_alloc_peer_map_mem : Flush Interval Configured to 256 pkts
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.543645] ol_txrx_pdev_attach: 2500 tx desc's allocated ; range starts from d4820000
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.551519] Firmware_Build_Number:83 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.554433] FW wireless modes: 0x7f9001
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.558093] num_rf_chain:0x00000004  ht_cap_info:0x0000185b  vht_cap_info:0x339b79fa  vht_supp_mcs:0x0000ffaa
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.567997] wmi_service_coex_gpio 0, wmi_service_4_wire_coex_support 0, coex_version 0
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.576056] 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.576056] Sending Ext resource cfg: HOST PLATFORM as 0
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.576056] fw_feature_bitmap as 50 to TGT
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.586908] ol_ath_service_ready_event: tt_support: 1
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.591793] ol_ath_service_ready_event: periodic_chan_stats: 1
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.597728] ol_ath_service_ready_event: sw_cal_support_check_flag: 1
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.604029] Peer Caching Enabled ; num_peers = 530, num_active_peers = 52 num_tids = 104, num_vdevs = 17
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.613485] EXT NSS Supported
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.616384] idx 1 req 2  num_units 1 num_unit_info 12 unit size 256 actual units 53 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.624193] ol_ath_alloc_host_mem_chunk req_id 2 idx 0 num_units 53 unit_len 256,
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.631538] idx 2 req 3  num_units 1 num_unit_info 12 unit size 1024 actual units 53 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.639482] ol_ath_alloc_host_mem_chunk req_id 3 idx 1 num_units 53 unit_len 1024,
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.646990] idx 3 req 4  num_units 1 num_unit_info 12 unit size 4096 actual units 53 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.654799] ol_ath_alloc_host_mem_chunk req_id 4 idx 2 num_units 53 unit_len 4096,
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.662279] idx 8 req 9  num_units 1 num_unit_info 8 unit size 819200 actual units 1 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.670225] ol_ath_alloc_host_mem_chunk req_id 9 idx 3 num_units 1 unit_len 819200,
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.677790] idx 0 req 1  num_units 0 num_unit_info 2 unit size 2192 actual units 531 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.685600] ol_ath_alloc_host_mem_chunk req_id 1 idx 4 num_units 531 unit_len 2192,
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.693258] idx 4 req 6  num_units 35 num_unit_info 0 unit size 3072 actual units 35 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.700985] ol_ath_alloc_host_mem_chunk req_id 6 idx 5 num_units 35 unit_len 3072,
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.708791] idx 5 req 7  num_units 1 num_unit_info 0 unit size 12288 actual units 1 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.716346] ol_ath_alloc_host_mem_chunk req_id 7 idx 6 num_units 1 unit_len 12288,
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.723905] idx 6 req 5  num_units 0 num_unit_info 2 unit size 2100 actual units 531 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.731631] ol_ath_alloc_host_mem_chunk req_id 5 idx 7 num_units 531 unit_len 2100,
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.739420] Support not added yet for Service 91
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.743937] Support not added yet for Service 92
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.748468] No EXT_MSG send INIT now
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.752032] chunk 0 len 13568 requested , ptr  0x59740000
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.757545] chunk 1 len 54272 requested , ptr  0x58a80000
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.762796] chunk 2 len 217088 requested , ptr  0x58ac0000
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.768392] chunk 3 len 819200 requested , ptr  0x58b00000
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.773809] chunk 4 len 1163952 requested , ptr  0x58400000
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.779272] chunk 5 len 107520 requested , ptr  0x58520000
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.784774] chunk 6 len 12288 requested , ptr  0x59fd0000
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.790111] chunk 7 len 1115100 requested , ptr  0x58600000
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.795706] chunk 8 len 0 requested , ptr  0xffffffff
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.800711] ol_ath_service_ready_event[4285] WAPI MBSSID 2 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.806404] smart_log_init: Smart logging Enabled buf=d4a40000 (size=65536)
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.877278] Version = 16777216 3  status = 0
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.880572] ol_ath_connect_htc() WMI is ready
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.884945] htt_h2t_frag_desc_bank_cfg_msg - HTT_H2T_MSG_TYPE_FRAG_DESC_BANK_CFG sent to FW for radio ID = 0
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.894762] target uses HTT version 2.2; host uses 2.2
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.902521] wmi_unified_register_event_handler : event handler already registered 0x900b
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.909658] wmi_unified_register_event_handler : event handler already registered 0x9042
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.917762] >>>> CB Set 00000000
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.921036] ol_ath_thermal_mitigation_attach: --
Tue Feb 11 21:41:03 2020 kern.debug kernel: [19455.925592] ol_if_dfs_setup: called 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.925671] ath_sysfs_diag_init: diag_fsattr 
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.929850] +hif_update_pipe_callback pipeid 8
Tue Feb 11 21:41:03 2020 kern.warn kernel: [19455.934304] -hif_update_pipe_callback
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.937962] ol_regdmn_init_channels: !avail mode 0x7f9001 (0x2) flags 0x2150
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.945002] ol_regdmn_init_channels: !avail mode 0x7f9001 (0x4) flags 0xa0
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.951814] ol_regdmn_init_channels: !avail mode 0x7f9001 (0x8) flags 0xc0
Tue Feb 11 21:41:03 2020 kern.info kernel: [19455.958699] ol_regdmn_init_channels: !avail mode 0x7f9001 (0x20) flags 0xd0
Tue Feb 11 21:41:04 2020 kern.info kernel: [19455.965662] ol_regdmn_init_channels: !avail mode 0x7f9001 (0x40) flags 0x150
Tue Feb 11 21:41:04 2020 kern.info kernel: [19455.972650] ol_regdmn_init_channels: !avail mode 0x7f9001 (0x800) flags 0x10080
[... 572 lines cut ...]
Tue Feb 11 21:44:25 2020 kern.info kernel: [19657.517416] dfs_deliver_event: dfs CAC_COMPLETED event delivered on chan 52.
Tue Feb 11 21:44:25 2020 kern.warn kernel: [19657.524495] su bfee 1 mu bfee 0 su bfer 1 mu bfer 1 impl bf 1 sounding dim 3
Tue Feb 11 21:44:25 2020 kern.warn kernel: [19657.531481] send_vdev_up_cmd_non_tlv for vap 0
Tue Feb 11 21:44:25 2020 kern.warn kernel: [19657.535921] Beacon mode set to staggered. Cannot enable FD
Tue Feb 11 21:44:25 2020 kern.warn kernel: [19657.541376] __ieee80211_smart_ant_init: Smart Antenna is not supported 
Tue Feb 11 21:44:26 2020 kern.warn kernel: [19658.182085] [wifi0] FWLOG: [81056] VDEV_MGR_HP_START_TIME ( 0x0, 0x1478, 0x43df000 )
Tue Feb 11 21:44:26 2020 kern.warn kernel: [19658.188837] [wifi0] FWLOG: [81056] RESMGR_OCS_GEN_PERIODIC_NOA ( 0x1 )
[... cut ...]
Tue Feb 11 21:44:36 2020 kern.warn kernel: [19668.220579] [wifi0] FWLOG: [91518] WAL_DBGID_SECURITY_ALLOW_DATA ( 0x457d60 )
Tue Feb 11 21:44:37 2020 kern.warn kernel: [19669.179861] [wifi0] FWLOG: [92799] WAL_DBGID_TX_BA_SETUP ( 0x457d60, 0x7c690006, 0x2, 0x20, 0x1 )
Tue Feb 11 21:44:37 2020 kern.warn kernel: [19669.187690] [wifi0] FWLOG: [92800] RATE: ChainMask 3, peer_mac 7c:69, phymode 6, ni_flags 0x00203006, vht_mcs_set 0x0000, ht_mcs_set 0xffff, legacy_rate_set 0x0ff0
Tue Feb 11 21:44:38 2020 kern.warn kernel: [19670.179677] [wifi0] FWLOG: [93026] WAL_DBGID_TX_BA_SETUP ( 0x457d60, 0x7c690000, 0x0, 0x20, 0x1 )

coolsnowwolf, 2020-02-12:
auto 信道跳信道

closed 2020-02-12 by coolsnowwolf

tangdashen1996, 2020-02-12:
感谢大雕回复，但是设置一直是固定44信道。