coolsnowwolf/lede #3045: 编译错误，求解决

lipwah, 2020-02-12:
make[3]: Leaving directory `/home/lipwah/lede/package/network/utils/iptables'
time: package/network/utils/iptables/compile#0.06#0.01#0.39
make[3]: Entering directory `/home/lipwah/lede/package/lean/openwrt-fullconenat'
touch /home/lipwah/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.prepared_e1ef44d511163f79685bd638757b9c03_6664517399ebbbc92a37c5bb081b5c53_check
. /home/lipwah/lede/include/shell.sh; xzcat /home/lipwah/lede/dl/fullconenat-2018-12-15-d4daedd0.tar.xz | tar -C /home/lipwah/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.. -xf -
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: *** [/home/lipwah/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.prepared_e1ef44d511163f79685bd638757b9c03_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory `/home/lipwah/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.04#0.00#0.28
make[2]: *** [package/lean/openwrt-fullconenat/compile] Error 2
make[2]: Leaving directory `/home/lipwah/lede'
make[1]: *** [/home/lipwah/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/lipwah/lede'
make: *** [world] 错误 2

coolsnowwolf, 2020-02-12:
梯子问题

closed 2020-02-12 by coolsnowwolf

lipwah, 2020-02-12:
> 梯子问题
感谢