coolsnowwolf/lede #3047: 编译报错，在线编辑的

saeson001, 2020-02-12:
弄了两次，都是这位置，每次两个多小时

cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/clamav-0.100.0/ipkg-install/usr/lib*/* /home/runner/work/lede/lede/build_dir/target-x86_64_musl/clamav-0.100.0/ipkg-x86_64/clamav/usr/lib/
install -d -m0755 /home/runner/work/lede/lede/build_dir/target-x86_64_musl/clamav-0.100.0/ipkg-x86_64/clamav/etc/config
install -m0600 ./files/clamav.config /home/runner/work/lede/lede/build_dir/target-x86_64_musl/clamav-0.100.0/ipkg-x86_64/clamav/etc/config/clamav
install -d -m0755 /home/runner/work/lede/lede/build_dir/target-x86_64_musl/clamav-0.100.0/ipkg-x86_64/clamav/etc/init.d/
install -m0755 ./files/clamav.init /home/runner/work/lede/lede/build_dir/target-x86_64_musl/clamav-0.100.0/ipkg-x86_64/clamav/etc/init.d/clamav
install -d -m0755 /home/runner/work/lede/lede/build_dir/target-x86_64_musl/clamav-0.100.0/ipkg-x86_64/clamav/usr/share/clamav
cp -fpR ./files/bytecode.cvd /home/runner/work/lede/lede/build_dir/target-x86_64_musl/clamav-0.100.0/ipkg-x86_64/clamav/usr/share/clamav/
find /home/runner/work/lede/lede/build_dir/target-x86_64_musl/clamav-0.100.0/ipkg-x86_64/clamav -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package clamav is missing dependencies for the following libraries:
libxml2.so.2
make[3]: *** [/home/runner/work/lede/lede/bin/packages/x86_64/packages/clamav_0.100.0-1_x86_64.ipk] Error 1
Makefile:124: recipe for target '/home/runner/work/lede/lede/bin/packages/x86_64/packages/clamav_0.100.0-1_x86_64.ipk' failed
time: package/feeds/packages/clamav/compile#0.28#0.10#0.37
make[3]: Leaving directory '/home/runner/work/lede/lede/feeds/packages/net/clamav'
make[2]: *** [package/feeds/packages/clamav/compile] Error 2
package/Makefile:107: recipe for target 'package/feeds/packages/clamav/compile' failed
make[2]: Leaving directory '/home/runner/work/lede/lede'
package/Makefile:103: recipe for target '/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
/home/runner/work/lede/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
**##[error]Process completed with exit code 2.**

zb479519891, 2020-02-12:
和你一样的情况

coolsnowwolf, 2020-02-13:
clamav 是干嘛的

jzh33, 2020-02-13:
杀毒软件吧。。。

cattypapa, 2020-02-18:
取消掉clamav全面的信号，别编译它

closed 2020-03-06 by coolsnowwolf