coolsnowwolf/lede #3060: 新代码ssrplus+ 游戏模式UDP中继服务器不起作用

bdhjiajia, 2020-02-12:
最近更新的版本游戏模式UDP中继服务器单独设置ss却不起作用了。

R3pl4c3r, 2020-02-17:
有同样的问题，2020年1月9日编译的版本的SSR plus+开启UDP并设置一个游戏节点，一切正常。但是在最近2月9日至2月14日编译的所有版本SSR plus+开启UDP设置游戏节点，游戏并不能被加速。但当我把UDP代理关闭并使用Windows下的Netch客户端添加游戏节点并监测游戏后，再打开游戏被加速。请问Lean大佬这是怎么回事呢？

bdhjiajia, 2020-02-17:
经LEAN回复得知是这个UDP模式和DOCKER有冲突导致，rm .config 后再重新生成编译脚本，重新编译，别选择DOCKER就正常了。

bdhjiajia, 2020-02-17:
> 有同样的问题，2020年1月9日编译的版本的SSR plus+开启UDP并设置一个游戏节点，一切正常。但是在最近2月9日至2月14日编译的所有版本SSR plus+开启UDP设置游戏节点，游戏并不能被加速。但当我把UDP代理关闭并使用Windows下的Netch客户端添加游戏节点并监测游戏后，再打开游戏被加速。请问Lean大佬这是怎么回事呢？

经LEAN回复得知是这个UDP模式和DOCKER有冲突导致，rm .config 后再重新生成编译脚本，重新编译，别选择DOCKER就正常了。

closed 2020-02-17 by bdhjiajia

R3pl4c3r, 2020-02-17:
> 经LEAN回复得知是这个UDP模式和DOCKER有冲突导致，rm .config 后再重新生成编译脚本，重新编译，别选择DOCKER就正常了。

十分感谢

yubanmeiqin9048, 2020-10-14:
我也是加速不了，没编译docker