coolsnowwolf/lede #3061: v2ray plugin编译提示404 not found  检查makefile里的链接已经是失效 的

feifan1113, 2020-02-12:
N1的 ，编译失败

tty228, 2020-02-12:
[v2ray-plugin-1.3.0.tar.gz](https://github.com/coolsnowwolf/lede/files/4192135/v2ray-plugin-1.3.0.tar.gz)
这个文件吗

AmadeusGhost, 2020-02-12:
这个链接实际是 https://codeload.github.com/shadowsocks/v2ray-plugin/tar.gz/v1.3.0 ，所以并没有失效

jxph1123, 2020-02-12:
Makefile:55: recipe for target '/home/etek/lede/build_dir/target-mipsel_24kc_musl/v2ray-plugin-1.3.0/.built' failed
make[3]: *** [/home/etek/lede/build_dir/target-mipsel_24kc_musl/v2ray-plugin-1.3.0/.built] Error 2
make[3]: Leaving directory '/home/etek/lede/package/lean/v2ray-plugin'
time: package/lean/v2ray-plugin/compile#9.26#3.61#27.44
package/Makefile:107: recipe for target 'package/lean/v2ray-plugin/compile' failed
make[2]: *** [package/lean/v2ray-plugin/compile] Error 2
make[2]: Leaving directory '/home/etek/lede'
package/Makefile:103: recipe for target '/home/etek/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/etek/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/etek/lede'
/home/etek/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

AmadeusGhost, 2020-02-12:
实测ipq40xx编译能过，估计失败是死在网络上了，设置go的国内代理试试：
```
export GO111MODULE=on
export GOPROXY=https://goproxy.cn
```

jxph1123, 2020-02-12:
按楼上的，还是不行。。。

AmadeusGhost, 2020-02-12:
> 按楼上的，还是不行。。。

你都不贴出具体报错，真就让人猜？

jxph1123, 2020-02-12:
上面贴了

AmadeusGhost, 2020-02-12:
> 上面贴了

你以为我没看到？具体报错的那个位置而不是结果！

lzx2009vb, 2020-02-12:
go: downloading github.com/v2ray/v2ray-core v4.22.1+incompatible
go: downloading github.com/golang/protobuf v1.3.2
go: extracting github.com/golang/protobuf v1.3.2
go: extracting github.com/v2ray/v2ray-core v4.22.1+incompatible
go: downloading golang.org/x/crypto v0.0.0-20190618222545-ea8f1a30c443
go: downloading github.com/refraction-networking/utls v0.0.0-20190909200633-43c36d3c1f57
go: downloading golang.org/x/sys v0.0.0-20190619223125-e40ef342dc56
go: downloading github.com/gorilla/websocket v1.4.1
go: extracting github.com/gorilla/websocket v1.4.1
go: extracting github.com/refraction-networking/utls v0.0.0-20190909200633-43c36d3c1f57
go: extracting golang.org/x/crypto v0.0.0-20190618222545-ea8f1a30c443
go: extracting golang.org/x/sys v0.0.0-20190619223125-e40ef342dc56
go: finding github.com/golang/protobuf v1.3.2
go: finding github.com/v2ray/v2ray-core v4.22.1+incompatible
go: finding github.com/v2ray/v2ray-core v4.22.1+incompatible
go: finding golang.org/x/sys v0.0.0-20190619223125-e40ef342dc56
go: finding github.com/refraction-networking/utls v0.0.0-20190909200633-43c36d3c1f57
go: finding github.com/gorilla/websocket v1.4.1
go: finding golang.org/x/crypto v0.0.0-20190618222545-ea8f1a30c443

Building targets
go: finding github.com/v2ray/v2ray-core v4.22.1+incompatible
go: finding github.com/v2ray/v2ray-core v4.22.1+incompatible
runtime/cgo
# runtime/cgo
gcc_mipsx.S: Assembler messages:
gcc_mipsx.S:25: Error: invalid operands `sw $18,12($29)'
gcc_mipsx.S:26: Error: invalid operands `sw $19,16($29)'
gcc_mipsx.S:27: Error: invalid operands `sw $20,20($29)'
gcc_mipsx.S:28: Error: invalid operands `sw $21,24($29)'
gcc_mipsx.S:29: Error: invalid operands `sw $22,28($29)'
gcc_mipsx.S:30: Error: invalid operands `sw $23,32($29)'
gcc_mipsx.S:31: Error: invalid operands `sw $30,36($29)'
gcc_mipsx.S:44: Error: invalid operands `jalr $20'
gcc_mipsx.S:48: Error: invalid operands `lw $18,12($29)'
gcc_mipsx.S:49: Error: invalid operands `lw $19,16($29)'
gcc_mipsx.S:50: Error: invalid operands `lw $20,20($29)'
gcc_mipsx.S:51: Error: invalid operands `lw $21,24($29)'
gcc_mipsx.S:52: Error: invalid operands `lw $22,28($29)'
gcc_mipsx.S:53: Error: invalid operands `lw $23,32($29)'
gcc_mipsx.S:54: Error: invalid operands `lw $30,36($29)'
gcc_mipsx.S:63: Error: invalid operands `lw $31,0($29)'

Cleaning module download cache (golang/go#27455)

make[3]: *** [/home/zcb/newifi3/lede/build_dir/target-mipsel_24kc_musl/v2ray-plugin-1.3.0/.built] Error 2
make[3]: Leaving directory `/home/zcb/newifi3/lede/package/lean/v2ray-plugin'
time: package/lean/v2ray-plugin/compile#4.84#0.98#17.39
make[2]: *** [package/lean/v2ray-plugin/compile] Error 2
make[2]: Leaving directory `/home/zcb/newifi3/lede'
make[1]: *** [/home/zcb/newifi3/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/zcb/newifi3/lede'
make: *** [world] 错误 2

是提示这个报错么？

AmadeusGhost, 2020-02-12:
是这个了，在package/lean/v2ray-plugin/Makefile加一行
```
PKG_USE_MIPS16:=0
```
试试

dovegq, 2020-02-12:
> 按楼上的，还是不行。。。

athens.azurefd.net  试试这个go的国内代理，不要走梯子

lzx2009vb, 2020-02-12:
> @lzx2009vb 是这个了
> 在package/lean/v2ray-plugin/Makefile加一行`PKG_USE_MIPS16:=0`试试
> 类似这样：
> 
> ```
> PKG_USE_MIPS16:=0
> 
> GO_PKG:=github.com/shadowsocks/v2ray-plugin
> GO_PKG_LDFLAGS:=-s -w
> ```

尝试了，可以了，谢谢

closed 2020-02-12 by coolsnowwolf