coolsnowwolf/lede #3066: 编译失败

ccc1924, 2020-02-13:
执行`./scripts/feeds install -a`时提示
```
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-mesh-openssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-mesh-wolfssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-basic', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-mini', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-openssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-wolfssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpa-supplicant', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpa-supplicant-mesh-openssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpa-supplicant-mesh-wolfssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpa-supplicant-basic', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpa-supplicant-mini', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpa-supplicant-p2p', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpa-supplicant-openssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpa-supplicant-wolfssl||', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_hostapd-basic', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_hostapd-mini', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_hostapd-openssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_hostapd-wolfssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-mesh-openssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-mesh-wolfssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-basic', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-mini', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-openssl', which does not exist
WARNING: Makefile 'package/network/services/hostapd/Makefile' has a dependency on 'PACKAGE_wpad-wolfssl||', which does not exist
```
查到[https://bugs.openwrt.org/index.php?do=details&task_id=2796](https://bugs.openwrt.org/index.php?do=details&task_id=2796)

大大可否帮忙解决？谢谢

AmadeusGhost, 2020-02-13:
```
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a
```

i6o6i, 2020-02-15:
貌似是openwrt的bug, 参考[git.openwrt.org Git - openwrt/openwrt.git/commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=766e778226f5d4c6ec49ce22b101a5dbd4306644)

[现在openwrt hostapd的Makefile](https://github.com/openwrt/openwrt/blob/master/package/network/services/hostapd/Makefile)已经删除了这两行
```
space :=
space+=
```
去`package/network/services/hostapd/Makefile` 删除这两行即可
目前我删除了这两行后没报错

hyird, 2020-02-16:
install 的时候 可能会提示  make的时候不提示就好 如果提示了 自己找问题

closed 2020-02-17 by ccc1924