coolsnowwolf/lede #3067: 让固件中的haproxy的web配置界面同时支持域名和IP两种写法的修改颁发

lqdflying, 2020-02-13:
`sed -i 's/o.datatype=\"ip4addr\"//g' ./build_dir/target-x86_64_musl/luci-app-haproxy-tcp/luasrc/model/cbi/haproxy.lua
sed -i 's/o.datatype=\"ip4addr\"//g' ./build_dir/target-x86_64_musl/luci-app-haproxy-tcp/ipkg-all/luci-app-haproxy-tcp/usr/lib/lua/luci/model/cbi/haproxy.lua
sed -i 's/o.datatype=\"ip4addr\"//g' ./build_dir/target-x86_64_musl/luci-app-haproxy-tcp/.pkgdir/luci-app-haproxy-tcp/usr/lib/lua/luci/controller/haproxy.lua
sed -i 's/o.datatype=\"ip4addr\"//g' ./build_dir/target-x86_64_musl/luci-app-haproxy-tcp/.pkgdir/luci-app-haproxy-tcp/usr/lib/lua/luci/model/cbi/haproxy.lua
sed -i 's/o.datatype=\"ip4addr\"//g' ./build_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/model/cbi/haproxy.lua
sed -i 's/o.datatype=\"ip4addr\"//g' ./build_dir/target-x86_64_musl/root.orig-x86/usr/lib/lua/luci/model/cbi/haproxy.lua
sed -i 's/o.datatype=\"ip4addr\"//g' ./package/lean/luci-app-haproxy-tcp/luasrc/model/cbi/haproxy.lua
sed -i 's/o.datatype=\"ip4addr\"//g' ./staging_dir/target-x86_64_musl/root-x86/usr/lib/lua/luci/model/cbi/haproxy.lua

sed -i '/frontend/i\resolvers mydns\n\ \ nameserver dns1 114.114.114.114:53\n\ \ nameserver dns2 223.5.5.5:53\n'  ./build_dir/target-x86_64_musl/luci-app-haproxy-tcp/ipkg-all/luci-app-haproxy-tcp/etc/haproxy_init.sh
sed -i '/frontend/i\resolvers mydns\n\ \ nameserver dns1 114.114.114.114:53\n\ \ nameserver dns2 223.5.5.5:53\n' ./build_dir/target-x86_64_musl/luci-app-haproxy-tcp/root/etc/haproxy_init.sh
sed -i '/frontend/i\resolvers mydns\n\ \ nameserver dns1 114.114.114.114:53\n\ \ nameserver dns2 223.5.5.5:53\n' ./build_dir/target-x86_64_musl/luci-app-haproxy-tcp/.pkgdir/luci-app-haproxy-tcp/etc/haproxy_init.sh
sed -i '/frontend/i\resolvers mydns\n\ \ nameserver dns1 114.114.114.114:53\n\ \ nameserver dns2 223.5.5.5:53\n' ./build_dir/target-x86_64_musl/root-x86/etc/haproxy_init.sh
sed -i '/frontend/i\resolvers mydns\n\ \ nameserver dns1 114.114.114.114:53\n\ \ nameserver dns2 223.5.5.5:53\n' ./build_dir/target-x86_64_musl/root.orig-x86/etc/haproxy_init.sh
sed -i '/frontend/i\resolvers mydns\n\ \ nameserver dns1 114.114.114.114:53\n\ \ nameserver dns2 223.5.5.5:53\n' ./package/lean/luci-app-haproxy-tcp/root/etc/haproxy_init.sh
sed -i '/frontend/i\resolvers mydns\n\ \ nameserver dns1 114.114.114.114:53\n\ \ nameserver dns2 223.5.5.5:53\n' ./staging_dir/target-x86_64_musl/root-x86/etc/haproxy_init.sh

sed -i 's/check/check\ resolvers\ mydns/g'  ./build_dir/target-x86_64_musl/luci-app-haproxy-tcp/ipkg-all/luci-app-haproxy-tcp/etc/haproxy_init.sh
sed -i 's/check/check\ resolvers\ mydns/g'  ./build_dir/target-x86_64_musl/luci-app-haproxy-tcp/root/etc/haproxy_init.sh
sed -i 's/check/check\ resolvers\ mydns/g'  ./build_dir/target-x86_64_musl/luci-app-haproxy-tcp/.pkgdir/luci-app-haproxy-tcp/etc/haproxy_init.sh
sed -i 's/check/check\ resolvers\ mydns/g'  ./build_dir/target-x86_64_musl/root-x86/etc/haproxy_init.sh
sed -i 's/check/check\ resolvers\ mydns/g'  ./build_dir/target-x86_64_musl/root.orig-x86/etc/haproxy_init.sh
sed -i 's/check/check\ resolvers\ mydns/g'  ./package/lean/luci-app-haproxy-tcp/root/etc/haproxy_init.sh
sed -i 's/check/check\ resolvers\ mydns/g'  ./staging_dir/target-x86_64_musl/root-x86/etc/haproxy_init.sh`

coolsnowwolf, 2020-02-13:
直接发个 PR 岂不是美滋滋？

lqdflying, 2020-02-13:
大佬的东西,虽然我测了好使,但不知道改的合适不合适,不敢随便发PR,哈哈

coolsnowwolf, 2020-02-19:
发就是了，有错误大家会改正

lqdflying, 2020-02-19:
the code has been merged

closed 2020-02-19 by lqdflying