coolsnowwolf/lede #3070: 最新版本antfs下载失败

helllkz, 2020-02-13:
更新到最新版本，修复antfs和ntfs-3g冲突那个bug的吧，现在找不到这个下载文件，404
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/antfs-2020-02-10-b41ba529.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/antfs-2020-02-10-b41ba529.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.

coolsnowwolf, 2020-02-13:
你被墙了

sky123456789sky, 2020-02-13:
我也没有 源文件里也是看不到这个文件

tty228, 2020-02-14:
解压，放到 DL 文件夹
[antfs.zip](https://github.com/coolsnowwolf/lede/files/4202351/antfs.zip)

ihuaer, 2020-02-14:
> 解压，放到 DL 文件夹
> [antfs.zip](https://github.com/coolsnowwolf/lede/files/4202351/antfs.zip)

多谢老哥，我编译试试

closed 2020-02-14 by coolsnowwolf