coolsnowwolf/lede #3074: 有大神能帮我看一下编译失败的原因吗？

wc5bk6, 2020-02-13:
/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/github.com/golang/protobuf/proto (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/github.com/golang/protobuf/proto (from $GOPATH)
src/v2ray.com/core/transport/internet/websocket/connection.go:10:2: cannot find package "github.com/gorilla/websocket" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/github.com/gorilla/websocket (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/github.com/gorilla/websocket (from $GOPATH)
src/v2ray.com/core/transport/internet/tls/tls.go:11:2: cannot find package "github.com/refraction-networking/utls" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/github.com/refraction-networking/utls (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/github.com/refraction-networking/utls (from $GOPATH)
src/v2ray.com/core/app/router/condition.go:8:2: cannot find package "go.starlark.net/starlark" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/go.starlark.net/starlark (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/go.starlark.net/starlark (from $GOPATH)
src/v2ray.com/core/app/router/condition.go:9:2: cannot find package "go.starlark.net/syntax" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/go.starlark.net/syntax (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/go.starlark.net/syntax (from $GOPATH)
src/v2ray.com/core/proxy/shadowsocks/config.go:11:2: cannot find package "golang.org/x/crypto/chacha20poly1305" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/chacha20poly1305 (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/golang.org/x/crypto/chacha20poly1305 (from $GOPATH)
src/v2ray.com/core/external/github.com/marten-seemann/qtls/13.go:26:2: cannot find package "golang.org/x/crypto/curve25519" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/curve25519 (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/golang.org/x/crypto/curve25519 (from $GOPATH)
src/v2ray.com/core/proxy/shadowsocks/config.go:12:2: cannot find package "golang.org/x/crypto/hkdf" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/hkdf (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/golang.org/x/crypto/hkdf (from $GOPATH)
src/v2ray.com/core/proxy/vmess/encoding/auth.go:10:2: cannot find package "golang.org/x/crypto/sha3" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/sha3 (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/golang.org/x/crypto/sha3 (from $GOPATH)
src/v2ray.com/core/app/dns/dnscommon.go:9:2: cannot find package "golang.org/x/net/dns/dnsmessage" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/net/dns/dnsmessage (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/golang.org/x/net/dns/dnsmessage (from $GOPATH)
src/v2ray.com/core/transport/internet/http/dialer.go:12:2: cannot find package "golang.org/x/net/http2" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/net/http2 (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/golang.org/x/net/http2 (from $GOPATH)
src/v2ray.com/core/transport/internet/http/hub.go:21:2: cannot find package "golang.org/x/net/http2/h2c" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/net/http2/h2c (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/golang.org/x/net/http2/h2c (from $GOPATH)
src/v2ray.com/core/transport/internet/sockopt_linux.go:6:2: cannot find package "golang.org/x/sys/unix" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/sys/unix (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/golang.org/x/sys/unix (from $GOPATH)
src/v2ray.com/core/app/commander/commander.go:12:2: cannot find package "google.golang.org/grpc" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/google.golang.org/grpc (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/google.golang.org/grpc (from $GOPATH)
src/v2ray.com/core/app/stats/command/command.pb.go:11:2: cannot find package "google.golang.org/grpc/codes" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/google.golang.org/grpc/codes (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/google.golang.org/grpc/codes (from $GOPATH)
src/v2ray.com/core/app/stats/command/command.pb.go:12:2: cannot find package "google.golang.org/grpc/status" in any of:
	/home/wc5bk6/openwrt180607/staging_dir/hostpkg/lib/go-cross/src/google.golang.org/grpc/status (from $GOROOT)
	/home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.go_work/build/src/google.golang.org/grpc/status (from $GOPATH)
make[3]: *** [Makefile:307: /home/wc5bk6/openwrt180607/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.built] Error 1
make[3]: Leaving directory '/home/wc5bk6/openwrt180607/package/v2ray'
time: package/v2ray/compile#0.16#0.67#0.86
make[2]: *** [package/Makefile:113: package/v2ray/compile] Error 2
make[2]: Leaving directory '/home/wc5bk6/openwrt180607'
make[1]: *** [package/Makefile:107: /home/wc5bk6/openwrt180607/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wc5bk6/openwrt180607'
make: *** [/home/wc5bk6/openwrt180607/include/toplevel.mk:218：world] 错误 2

hxlls, 2020-02-13:
openwrt180607？这什么玩意？

zh610902551, 2020-02-13:
cannot find package "github.com/gorilla/websocket" in any of:
找不到github.com/gorilla/websocket

dovegq, 2020-02-14:
export GO111MODULE=on
export GOPROXY=https://athens.azurefd.net

jess2020github, 2020-02-22:
aria2只能使用用户名密码方式，使用其他方式时，提示有一行参数未设置，不让保存

jess2020github, 2020-02-22:
aria2版本1.35

closed 2020-03-06 by coolsnowwolf