coolsnowwolf/lede #3077: 动态dns不能正常运行，已经安装了bind-host

zzsx644, 2020-02-13:
Error relocating /usr/bin/host: isc_socket_attach: symbol not found
Error relocating /usr/bin/host: dns_log_init: symbol not found
Error relocating /usr/bin/host: dns_rdata_freestruct: symbol not found
Error relocating /usr/bin/host: isc_buffer_free: symbol not found
Error relocating /usr/bin/host: isc_task_create: symbol not found
Error relocating /usr/bin/host: isc_socket_ipv6only: symbol not found
Error relocating /usr/bin/host: dst_lib_init: symbol not found
Error relocating /usr/bin/host: cfg_obj_asstring: symbol not found
Error relocating /usr/bin/host: dns_fixedname_initname: symbol not found
Error relocating /usr/bin/host: dns_rdataset_current: symbol not found
Error relocating /usr/bin/host: dns_compress_init: symbol not found
Error relocating /usr/bin/host: isc_app_block: symbol not found
Error relocating /usr/bin/host: isc_sockaddr_fromin6: symbol not found
Error relocating /usr/bin/host: isc_nonce_buf: symbol not found
Error relocating /usr/bin/host: dns_message_peekheader: symbol not found
Error relocating /usr/bin/host: isc_netaddr_totext: symbol not found
Error relocating /usr/bin/host: dns_rdata_tostruct: symbol not found
Error relocating /usr/bin/host: isc_result_totext: symbol not found
Error relocating /usr/bin/host: dns_rdataset_makequestion: symbol not found
Error relocating /usr/bin/host: dns_rdataclass_format: symbol not found
Error relocating /usr/bin/host: dns_name_copy: symbol not found
Error relocating /usr/bin/host: irs_resconf_getndots: symbol not found
Error relocating /usr/bin/host: dns_message_buildopt: symbol not found
Error relocating /usr/bin/host: cfg_parse_file: symbol not found
Error relocating /usr/bin/host: isc_timer_touch: symbol not found
Error relocating /usr/bin/host: dns_fixedname_name: symbol not found
Error relocating /usr/bin/host: dns_tsigkey_createfromkey: symbol not found
Error relocating /usr/bin/host: isc_netaddr_format: symbol not found
Error relocating /usr/bin/host: dns_message_findname: symbol not found
Error relocating /usr/bin/host: isc_commandline_reset: symbol not found
Error relocating /usr/bin/host: isc__mempool_get: symbol not found
Error relocating /usr/bin/host: isc_mempool_create: symbol not found
Error relocating /usr/bin/host: cfg_map_getname: symbol not found
Error relocating /usr/bin/host: isc_task_detach: symbol not found
Error relocating /usr/bin/host: isc_sockaddr_pf: symbol not found
Error relocating /usr/bin/host: isc_taskmgr_create: symbol not found
Error relocating /usr/bin/host: isc_socket_connect: symbol not found
Error relocating /usr/bin/host: dns_message_setopt: symbol not found
Error relocating /usr/bin/host: isc_event_allocate: symbol not found
Error relocating /usr/bin/host: dns_rdataset_first: symbol not found
Error relocating /usr/bin/host: bind9_getaddresses: symbol not found
 230746  WARN : Get registered/public IP for 'openwrt.zzsx644.xyz' failed - retry 5/0 in 60 seconds
 230847       : #> /usr/bin/host -t A openwrt.zzsx644.xyz  >/var/run/ddns/namecheap.dat 2>/var/run/ddns/namecheap.err
 230847 ERROR : BIND host error: '127'
 230847       : Error loading shared library libdns.so.1311: No such file or directory (needed by /usr/bin/host)
Error loading shared library libirs.so.1301: No such file or directory (needed by /usr/bin/host)
Error loading shared library libbind9.so.1302: No such file or directory (needed by /usr/bin/host)
Error loading shared library libisccfg.so.1302: No such file or directory (needed by /usr/bin/host)
Error loading shared library libisc.so.1310: No such file or directory (needed by /usr/bin/host)
Error relocating /usr/bin/host: dns_message_create: symbol not found
Error relocating /usr/bin/host: dst_key_setbits: symbol not found
Error relocating /usr/bin/host: isc_event_free: symbol not found
Error relocating /usr/bin/host: dns_message_settsigkey: symbol not found
Error relocating /usr/bin/host: isc_socket_detach: symbol not found
Error relocating /usr/bin/host: dns_message_destroy: symbol not found
Error relocating /usr/bin/host: isc_app_finish: symbol not found
Error relocating /usr/bin/host: dns_tsigkey_create: symbol not found
Error relocating /usr/bin/host: dns_tsig_verify: symbol not found
Error relocating /usr/bin/host: isc_task_send: symbol not found
Error relocating /usr/bin/host: dns_name_invalidate: symbol not found
Error relocating /usr/bin/host: isc_commandline_index: symbol not found
Error relocating /usr/bin/host: dns_name_format: symbol not found
Error relocating /usr/bin/host: isc_hex_totext: symbol not found
Error relocating /usr/bin/host: dst_lib_destroy: symbol not found
Error relocating /usr/bin/host: dns_message_nextname: symbol not found
Error relocating /usr/bin/host: isc_buffer_getuint16: symbol not found
Error relocating /usr/bin/host: dns_rdata_init: symbol not found
Error relocating /usr/bin/host: isc_net_disableipv6: symbol not found
Error relocating /usr/bin/host: isc_sockaddr_format: symbol not found
Error relocating /usr/bin/host: isc_mem_stats: symbol not found
[... 155 lines cut ...]
Error relocating /usr/bin/host: isc_result_totext: symbol not found
Error relocating /usr/bin/host: dns_rdataset_makequestion: symbol not found
Error relocating /usr/bin/host: dns_rdataclass_format: symbol not found
Error relocating /usr/bin/host: dns_name_copy: symbol not found
Error relocating /usr/bin/host: irs_resconf_getndots: symbol not found
Error relocating /usr/bin/host: dns_message_buildopt: symbol not found
Error relocating /usr/bin/host: cfg_parse_file: symbol not found
Error relocating /usr/bin/host: isc_timer_touch: symbol not found
Error relocating /usr/bin/host: dns_fixedname_name: symbol not found
Error relocating /usr/bin/host: dns_tsigkey_createfromkey: symbol not found
Error relocating /usr/bin/host: isc_netaddr_format: symbol not found
Error relocating /usr/bin/host: dns_message_findname: symbol not found
Error relocating /usr/bin/host: isc_commandline_reset: symbol not found
Error relocating /usr/bin/host: isc__mempool_get: symbol not found
Error relocating /usr/bin/host: isc_mempool_create: symbol not found
Error relocating /usr/bin/host: cfg_map_getname: symbol not found
Error relocating /usr/bin/host: isc_task_detach: symbol not found
Error relocating /usr/bin/host: isc_sockaddr_pf: symbol not found
Error relocating /usr/bin/host: isc_taskmgr_create: symbol not found
Error relocating /usr/bin/host: isc_socket_connect: symbol not found
Error relocating /usr/bin/host: dns_message_setopt: symbol not found
Error relocating /usr/bin/host: isc_event_allocate: symbol not found
Error relocating /usr/bin/host: dns_rdataset_first: symbol not found
Error relocating /usr/bin/host: bind9_getaddresses: symbol not found
 230925  WARN : Get registered/public IP for 'openwrt.zzsx644.xyz' failed - retry 6/0 in 60 seconds

coolsnowwolf, 2020-02-14:
编译时加入 bind （其实没必要

zzsx644, 2020-02-14:
> 编译时加入 bind （其实没必要

感谢，虽然我还是不太明白

closed 2020-02-14 by coolsnowwolf