coolsnowwolf/lede #3079: 更新后所用dockerman，无法启动容器。

cnekol, 2020-02-14:
[image]
网络使用的bridge。

------

[image]
使用host无法创建容器

coolsnowwolf, 2020-02-14:
不支持你这个版本的 luci

cnekol, 2020-02-14:
#3053 
是拉取的这个更新，未作任何修改，直接在make menuconfig 内勾选的。这次更新中luciapp里dockerman替换了之前的docker。

lisaac, 2020-02-14:
你这个问题和dockerman无关
联系关闭

cnekol, 2020-02-14:
> 你这个问题和dockerman无关
> 联系关闭

我会全新安装一次，之前一直是升级上来的。docker分区以及其内容都没动过。理论上应该没影响，不过这次docker的版本好像更新了。我会把盘全部删除重新创建试试。

elonj, 2020-02-14:
> 
> 
> > 你这个问题和dockerman无关
> > 联系关闭
> 
> 我会全新安装一次，之前一直是升级上来的。docker分区以及其内容都没动过。理论上应该没影响，不过这次docker的版本好像更新了。我会把盘全部删除重新创建试试。

我全新安装也遇到了这个。而且重启之后docker没有自动启动，试了三次。在启动项里关了再开才实现docker自动启动。另一个问题是，docker只启动了13个container中的3个，需要手动启动剩余的container。不知道是什么原因

unifreq, 2020-02-14:
可以启动，没问题，不过iptables1.8.2的版本可能需要升级到1.8.3

cnekol, 2020-02-15:
> > > 你这个问题和dockerman无关
> > > 联系关闭
> > 
> > 
> > 我会全新安装一次，之前一直是升级上来的。docker分区以及其内容都没动过。理论上应该没影响，不过这次docker的版本好像更新了。我会把盘全部删除重新创建试试。
> 
> 我全新安装也遇到了这个。而且重启之后docker没有自动启动，试了三次。在启动项里关了再开才实现docker自动启动。另一个问题是，docker只启动了13个container中的3个，需要手动启动剩余的container。不知道是什么原因

已经全新安装，固件用最新编译的带。我是在虚拟环境下，删除了之前的盘全部新建，只是需要手动格式并化挂载分区为docker所用。也碰到过一次 #3088 中拉取镜像后不显示的问题。后来重新上传固件不保存配置更新一次后正常了。

portianer 无法连接到本地docker不知是否是因为被dockerman占用的情况。目前直接dockerman管理，使用正常。我会多重启几次再多测试下。

closed 2020-02-15 by cnekol