coolsnowwolf/lede #3084: ssrplus导入SSR/V2RAY的配置信息不能生成配置，其中：ssr的配置谢信息能识别成功但无配置信息出来！！！

zxlhhyccc, 2020-02-14:
导入ssr配置信息：
[image]
导入v2ray配置信息：
[image]

SharerMax, 2020-02-14:
不能导入的URL 提供一下

zxlhhyccc, 2020-02-14:
> 不能导入的URL 提供一下

如何提供给你。。。。？

SharerMax, 2020-02-14:
> > 不能导入的 URL 提供一下
> 
> 如何提供给你。。。。？

复制，粘贴 到这里

zxlhhyccc, 2020-02-15:
> > > 不能导入的 URL 提供一下
> > 
> > 
> > 如何提供给你。。。。？
> 
> 复制，粘贴 到这里
提供一个v2ray：

vmess://ewoidiI6ICIyIiwKInBzIjogIjIzM3YyLmNvbV80Ny4yNDAuMTAzLjg2IiwKImFkZCI6ICI0Ny4yNDAuMTAzLjg2IiwKInBvcnQiOiAiMjYwNDEiLAoiaWQiOiAiOTgxZjdkMDMtYWI3NS00YWY0LTliNzQtZTEzNjAwZDUxMWUwIiwKImFpZCI6ICIyMzMiLAoibmV0IjogIndzIiwKInR5cGUiOiAibm9uZSIsCiJob3N0IjogIiIsCiJwYXRoIjogIiIsCiJ0bHMiOiAiIgp9Cg==

SharerMax, 2020-02-15:
> vmess://ewoidiI6ICIyIiwKInBzIjogIjIzM3YyLmNvbV80Ny4yNDAuMTAzLjg2IiwKImFkZCI6ICI0Ny4yNDAuMTAzLjg2IiwKInBvcnQiOiAiMjYwNDEiLAoiaWQiOiAiOTgxZjdkMDMtYWI3NS00YWY0LTliNzQtZTEzNjAwZDUxMWUwIiwKImFpZCI6ICIyMzMiLAoibmV0IjogIndzIiwKInR5cGUiOiAibm9uZSIsCiJob3N0IjogIiIsCiJwYXRoIjogIiIsCiJ0bHMiOiAiIgp9Cg==

我这导入正常 用的默认主题(Bootstrap)

badcrazy, 2020-02-15:
测试正常导入

zxlhhyccc, 2020-02-15:
> 测试正常导入

我试试默认主题

zxlhhyccc, 2020-02-16:
> 测试正常导入

我测试导入不成功。。。我是官方19.07，可能是不兼容。。。。

zxlhhyccc, 2020-02-16:
> 我这导入正常 用的默认主题(Bootstrap)

closed 2020-03-06 by coolsnowwolf