coolsnowwolf/lede #3086: antfs-mount与ntfs-3g冲突

505799690, 2020-02-14:
Collected errors:
 * check_data_file_clashes: Package ntfs-3g wants to install file /workspace/lede/build_dir/target-mipsel_24kc_musl/root-ramips/sbin/mount.ntfs
        But that file is already provided by package  * antfs-mount
 * opkg_install_cmd: Cannot install package ntfs-3g.

EnnawYang, 2020-02-14:
取消一个不就好了？

505799690, 2020-02-14:
> 取消一个不就好了？

我不了解antfs-mount这个是干嘛的啊，名字上看是挂载ntfs啊。我想读写ntfs。。

505799690, 2020-02-14:
> 取消一个不就好了？

antfs-mount 挂载了也能读写嘛=？

EnnawYang, 2020-02-14:
> > 取消一个不就好了？
> 
> 我不了解antfs-mount这个是干嘛的啊，名字上看是挂载ntfs啊。我想读写ntfs。。

新的ntfs驱动，两个功能重复

505799690, 2020-02-14:
> > > 取消一个不就好了？
> > 
> > 
> > 我不了解antfs-mount这个是干嘛的啊，名字上看是挂载ntfs啊。我想读写ntfs。。
> 
> 新的ntfs驱动，两个功能重复

感谢 找到了commit与源码库，antfs-mount基于ntfs-3g开发 可读写ntfs。

closed 2020-02-14 by 505799690