coolsnowwolf/lede #3088: 新版Docker(docker manager)pull error及解决方法

R3pl4c3r, 2020-02-14:
新版docker在docker pull时
[image]
虽然显示pull成功，但是找不到images。
[image]
在命令行pull之后发现
[image]
是没有文件夹，创建文件夹并pull后
[image]
可以pull成功，但仍存在一个问题，就是在docker run时，会显示
`docker: Error response from daemon: failed to update store for object type *libnetwork.endpointCnt: Key not found in store.`

R3pl4c3r, 2020-02-14:
此时输入
`rm -rf /var/lib/docker`
再重启docker，开启container即可解决问题

bdhjiajia, 2020-02-14:
谢谢分享

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

coolsnowwolf, 2020-02-14:
mark

KFERMercer, 2020-02-14:
👍

RealKiro, 2020-02-18:
[image]

我的luci界面不正常，重编译升级还是一样，该怎么办？

R3pl4c3r, 2020-02-23:
@coolsnowwolf 其实原来要是装过docker并挂载过/opt，其实可以新装之后先挂载原有的/opt，之后直接在启动项重启docker就恢复正常了，没必要执行上面的那些步骤
The issue was closed due to solution.

closed 2020-02-23 by R3pl4c3r