coolsnowwolf/lede #3091: ssrplus+的广告屏蔽文件无法更新啦！

abclai, 2020-02-14:
ssrplus+的广告屏蔽文件无法更新啦！
更新过好几次发现，ssrplus+的广告屏蔽文件无法更新啦！
原因应该是文件过大（2.6M），下载地址为德国服务器，就算用VPS下载也好几分钟
下载完之后好像还要上传转换，但是无法转换！也就无法生效
可否提供ad.conf里面屏蔽的格式？自己手动上传ad.conf到lede?

abclai, 2020-02-14:
我有尝试把hanti-AD Filters的dnsmasq屏蔽文件
https://anti-ad.win/dnsmasq.conf这个的改名为ad.conf之后
上传到/etc/dnsmasq.ssr/替代同名文件，但是只能识别出数据数量但是无法屏蔽
所以希望能提供屏蔽的格式或是能改成新的dnsmasq屏蔽文件 比如hanti-AD Filters的

hplee0120, 2020-02-14:
我也感觉可以考虑将ssr plus的广告屏蔽规则换成别的规则。因为这个ssr plus的广告屏蔽和广告屏蔽大师的hosts规则好像都是采用easylistchina+easylist这个规则，两个插件都开的话规则重叠了似乎。adbyby plus就变成了普通的adbyby

hplee0120, 2020-02-14:
刚看了下有人pr了广告屏蔽可以自定义订阅，感觉挺好的。

hcym, 2020-02-15:
这个规则是域名还是dnsmasq？有范本就好了，不用联网自己定制更方便

abclai, 2020-02-15:
> 这个规则是域名还是dnsmasq？有范本就好了，不用联网自己定制更方便

转化之后为格式我不清楚，因为我就没有更新成功过！转化之前都是域名啦！包括dnsmasq的https://anti-ad.win/dnsmasq.conf 也是域名啦！只是域名前要什么东西域名后是否要/0.0.0.0就不清楚啦 现在只能等看是否能更新看下格式啦

abclai, 2020-02-15:
> 刚看了下有人pr了广告屏蔽可以自定义订阅，感觉挺好的。

自行订阅问题不大，主要要屏蔽的格式对啦！如果转化完格式不对，没用！最少我直接改文件名是没用，只有识别出数据的条数没有屏蔽啦

hplee0120, 2020-02-15:
我也好奇过这个问题，所以之前试过把插件的规则订阅地址改成yhosts
https://github.com/vokins/yhosts
之类的规则项目地址，似乎也是不成功

hplee0120, 2020-02-15:
> 我也好奇过这个问题，所以之前试过把插件hosts的规则订阅地址改成yhosts
> https://github.com/vokins/yhosts
> 之类的规则项目地址，似乎也是不成功，不过你说的原来的广告屏蔽不能更新我倒是没遇到过。在我这是正常更新的。5857条

abclai, 2020-02-15:
> 我也好奇过这个问题，所以之前试过把插件的hosts规则订阅地址改成yhosts
> https://github.com/vokins/yhosts
> 之类的规则项目地址，似乎也是不成功。不过你说的广告屏蔽不能更新我这边倒是没遇到过。5857条。是正常更新的

你更新的是是否adbyby plus有开着？我有种怀疑就是这个SSPLUS+的广告更新需要adbyby plus有编译和开启才能用啦，不过我感觉这样就没意义啦

coolsnowwolf, 2020-02-15:
可用了

hcym, 2020-02-15:
要是和adby一起玩就没意思了，我都是dnsmasq和hosts简单粗暴

closed 2020-02-15 by coolsnowwolf