coolsnowwolf/lede #3094: /usr/share/shadowsocksr/subscribe.lua:321: attempt to concatenate field 'alias' (a nil value)

miroda, 2020-02-15:
[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:356: in main chunk
	[C]: in function 'xpcall'
	/usr/share/shadowsocksr/subscribe.lua:309: in function </usr/share/shadowsocksr/subscribe.lua:228>
	[C]: in function 'foreach'
	/usr/share/shadowsocksr/subscribe.lua:358: in function </usr/share/shadowsocksr/subscribe.lua:356>
2020-02-15 07:58:40 stack traceback:
2020-02-15 07:58:40 /usr/share/shadowsocksr/subscribe.lua:321: attempt to concatenate field 'alias' (a nil value)

miroda, 2020-02-15:
找到了解决方法，ssh连接到openwrt服务器，注释这一行即可
` /usr/share/shadowsocksr/subscribe.lua`

SharerMax, 2020-02-15:
求复现步骤

coolsnowwolf, 2020-02-15:
最好附上订阅 URL 或者发到邮箱，这个不属于解决

Q0x18c75bcf, 2020-02-15:
https://ji-rss-aa.pw/link_diy/uwfzSdn0vSp0hNQT?t=1 试试这个 我也遇到这种情况

SharerMax, 2020-02-15:
> https://ji-rss-aa.pw/link_diy/uwfzSdn0vSp0hNQT?t=1 试试这个 我也遇到这种情况

--! 我这还是好使 无法复现

Q0x18c75bcf, 2020-02-15:
> > https://ji-rss-aa.pw/link_diy/uwfzSdn0vSp0hNQT?t=1 试试这个 我也遇到这种情况
> 
> --! 我这还是好使 无法复现

先手动添加一个链接，然后再添加订阅url 更新看看

SharerMax, 2020-02-16:
> > > https://ji-rss-aa.pw/link_diy/uwfzSdn0vSp0hNQT?t=1 试试这个 我也遇到这种情况
> > 
> > 
> > --! 我这还是好使 无法复现
> 
> 先手动添加一个链接，然后再添加订阅 url 更新看看

建议你更新到最新到固件，删除所有节点后重新添加订阅试试。
我还是不能复现。

closed 2020-02-17 by coolsnowwolf