coolsnowwolf/lede #3100: 光猫桥接模式，openwrt拨号连接，有ipv6地址但是没网

BlacK201, 2020-02-15:
[image]

现在不仅内网设备ipv6没网，连入openwrt的ssh后台，直接
`ping6 [240c::6666]`
也ping不通
[image]
有大佬遇到过这样的问题吗

nivalxer, 2020-02-15:
刚试了一下，正常的。
看看是不是运营商的问题。
现在貌似电信的IPV6路由还不太好，所以dnsmsq那边暂时还是过滤了IPV6地址返回
[image]

BlacK201, 2020-02-15:
@nivalxer 我这里是移动的宽带，如果使用电脑直接拨号，ipv6网站和ping就什么问题都没有，路由器拨号就会出问题。。

BlacK201, 2020-02-15:
尝试过的办法：
根据[https://www.v2ex.com/t/251153](url)中提供的方法修改了文件`/etc/config/firewall`，添加了如下配置
`config defaults
        option disable_ipv6 1`
重启后问题并没有得到解决

klever1988, 2020-02-15:
[image]

试试中继模式

BlacK201, 2020-02-16:
@klever1988 也不行，今天在tg群也看见几个哥们跟我一样的情况，很奇怪

georip, 2020-02-18:
最近这几个版本编译出来好像都这样。能获得IPV6地址，但无法上网。前几个版本 IPV6正常上网。最近的版本PPPOE拨号也延迟时间了。 以前看内核日志基本在50-60秒以内pppoe就拨号完成，现在延迟了快1分钟了。
[   42.052374] ath: country maps to regdmn code: 0x3b
[   42.052379] ath: Country alpha2 being used: AU
[   42.052383] ath: Regpair used: 0x3b
[   42.052388] ath: regdomain 0x8024 dynamically updated by user
[   45.903509] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   47.559999] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   47.561986] br-lan: port 2(wlan1) entered blocking state
[   47.564829] br-lan: port 2(wlan1) entered disabled state
[   47.570862] device wlan1 entered promiscuous mode
[   47.583813] br-lan: port 3(wlan0) entered blocking state
[   47.583912] br-lan: port 3(wlan0) entered disabled state
[   47.589119] device wlan0 entered promiscuous mode
[   48.069090] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   48.069350] br-lan: port 3(wlan0) entered blocking state
[   48.074600] br-lan: port 3(wlan0) entered forwarding state
[   48.358201] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   48.358471] br-lan: port 2(wlan1) entered blocking state
[   48.363725] br-lan: port 2(wlan1) entered forwarding state
[   95.378775] pppoe-vwan3: renamed from ppp0
[   95.464495] pppoe-vwan2: renamed from ppp1
[   95.504443] pppoe-vwan1: renamed from ppp2
[  106.472557] hrtimer: interrupt took 417896 ns

Ge0Lin, 2020-02-18:
请查看：
DHCP/DNS>>>>
高级设置>>>
禁止解释 IPv6 DNS 记录>>**这条是否被启用**

BlacK201, 2020-02-19:
@georip 唉，手欠把以前的固件删了，没法测试了。。

BlacK201, 2020-02-19:
@georip 唉，手欠把以前的固件删了，没法测试了。。

> 请查看：
> DHCP/DNS>>>>
> 高级设置>>>
> 禁止解释 IPv6 DNS 记录>>**这条是否被启用**

你好。。这个在群里的朋友也提醒过我，但是这个选项的开关与否只影响了我ping一个解析到v6的域名时能否正确解析。。但是就算解析出来了地址，也是全部loss无法ping通。。

you-mu, 2020-02-19:
2月13日新装了一个,同样移动宽带. 一切正常.
你修改过哪些配置?

BlacK201, 2020-02-19:
> 2月13日新装了一个,同样移动宽带. 一切正常.
> 你修改过哪些配置?

无论是修改了一大堆配置的还是什么都没改过的，我都试过好几遍。。你是哪里的移动宽带啊

you-mu, 2020-02-19:
> > 2月13日新装了一个,同样移动宽带. 一切正常.
> > 你修改过哪些配置?
> 
> 无论是修改了一大堆配置的还是什么都没改过的，我都试过好几遍。。你是哪里的移动宽带啊

广东深圳。

BlacK201, 2020-02-19:
> > > 2月13日新装了一个,同样移动宽带. 一切正常.
> > > 你修改过哪些配置?
> > 
> > 
> > 无论是修改了一大堆配置的还是什么都没改过的，我都试过好几遍。。你是哪里的移动宽带啊
> 
> 广东深圳。

我这里是广西移动，就算是光猫拨号，也是只有光猫有v6网络，而光猫下的设备只有地址没有网络。
可是openwrt路由器拨号的话，就连路由器自己也是没有v6网络的。。如果用padavan老毛子固件就是和光猫一样的情况

nivalxer, 2020-02-19:
> > > > 2月13日新装了一个,同样移动宽带. 一切正常.
> > > > 你修改过哪些配置?
> > > 
> > > 
> > > 无论是修改了一大堆配置的还是什么都没改过的，我都试过好几遍。。你是哪里的移动宽带啊
> > 
> > 
> > 广东深圳。
> 
> 我这里是广西移动，就算是光猫拨号，也是只有光猫有v6网络，而光猫下的设备只有地址没有网络。
> 可是openwrt路由器拨号的话，就连路由器自己也是没有v6网络的。。如果用padavan老毛子固件就是和光猫一样的情况

刚编译了最新的固件，正常的。
有尝试恢复默认配置吗？

BlacK201, 2020-02-19:
> > > > > 2月13日新装了一个,同样移动宽带. 一切正常.
> > > > > 你修改过哪些配置?
> > > > 
> > > > 
> > > > 无论是修改了一大堆配置的还是什么都没改过的，我都试过好几遍。。你是哪里的移动宽带啊
> > > 
> > > 
> > > 广东深圳。
> > 
> > 
> > 我这里是广西移动，就算是光猫拨号，也是只有光猫有v6网络，而光猫下的设备只有地址没有网络。
> > 可是openwrt路由器拨号的话，就连路由器自己也是没有v6网络的。。如果用padavan老毛子固件就是和光猫一样的情况
> 
> 刚编译了最新的固件，正常的。
> 有尝试恢复默认配置吗？

我都重新编译重刷过不知道多少次了(；一_一)

nivalxer, 2020-02-19:
> > > > > > 2月13日新装了一个,同样移动宽带. 一切正常.
> > > > > > 你修改过哪些配置?
> > > > > 
> > > > > 
> > > > > 无论是修改了一大堆配置的还是什么都没改过的，我都试过好几遍。。你是哪里的移动宽带啊
> > > > 
> > > > 
> > > > 广东深圳。
> > > 
> > > 
> > > 我这里是广西移动，就算是光猫拨号，也是只有光猫有v6网络，而光猫下的设备只有地址没有网络。
> > > 可是openwrt路由器拨号的话，就连路由器自己也是没有v6网络的。。如果用padavan老毛子固件就是和光猫一样的情况
> > 
> > 
> > 刚编译了最新的固件，正常的。
> > 有尝试恢复默认配置吗？
> 
> 我都重新编译重刷过不知道多少次了(；一_一)

ifconfig能看到br-lan上面有240e开头的ipv6地址是吧？
做一下路由跟踪试试
traceroute -d 240C::6666

you-mu, 2020-02-19:
> > > > > > 2月13日新装了一个,同样移动宽带. 一切正常.
> > > > > > 你修改过哪些配置?
> > > > > 
> > > > > 
> > > > > 无论是修改了一大堆配置的还是什么都没改过的，我都试过好几遍。。你是哪里的移动宽带啊
> > > > 
> > > > 
> > > > 广东深圳。
> > > 
> > > 
> > > 我这里是广西移动，就算是光猫拨号，也是只有光猫有v6网络，而光猫下的设备只有地址没有网络。
> > > 可是openwrt路由器拨号的话，就连路由器自己也是没有v6网络的。。如果用padavan老毛子固件就是和光猫一样的情况
> > 
> > 
> > 刚编译了最新的固件，正常的。
> > 有尝试恢复默认配置吗？
> 
> 我都重新编译重刷过不知道多少次了(；一_一)

软件选了哪些，有mwan吗？

BlacK201, 2020-02-19:
> > > > > > > 2月13日新装了一个,同样移动宽带. 一切正常.
> > > > > > > 你修改过哪些配置?
> > > > > > 
> > > > > > 
> > > > > > 无论是修改了一大堆配置的还是什么都没改过的，我都试过好几遍。。你是哪里的移动宽带啊
> > > > > 
> > > > > 
> > > > > 广东深圳。
> > > > 
> > > > 
> > > > 我这里是广西移动，就算是光猫拨号，也是只有光猫有v6网络，而光猫下的设备只有地址没有网络。
> > > > 可是openwrt路由器拨号的话，就连路由器自己也是没有v6网络的。。如果用padavan老毛子固件就是和光猫一样的情况
> > > 
> > > 
> > > 刚编译了最新的固件，正常的。
> > > 有尝试恢复默认配置吗？
> > 
> > 
> > 我都重新编译重刷过不知道多少次了(；一_一)
> 
> 软件选了哪些，有mwan吗？

现在我用的这个版本没有，之前也编译过有mwan的，然后按照教程关闭了负载均衡

BlacK201, 2020-02-19:
@nivalxer 我已经把ipv6关了。。一会有空重新开启试试

whlsxl, 2020-02-19:
> 请查看：
> DHCP/DNS>>>>
> 高级设置>>>
> 禁止解释 IPv6 DNS 记录>>**这条是否被启用**

我的原因是这个，谢谢！

whlsxl, 2020-02-19:
> > > > 2月13日新装了一个,同样移动宽带. 一切正常.
> > > > 你修改过哪些配置?
> > > 
> > > 
> > > 无论是修改了一大堆配置的还是什么都没改过的，我都试过好几遍。。你是哪里的移动宽带啊
> > 
> > 
> > 广东深圳。
> 
> 我这里是广西移动，就算是光猫拨号，也是只有光猫有v6网络，而光猫下的设备只有地址没有网络。
> 可是openwrt路由器拨号的话，就连路由器自己也是没有v6网络的。。如果用padavan老毛子固件就是和光猫一样的情况

看看能不能超管登陆光猫，改成自己拨号。光猫下发的IPv6 是不是直接就是/64的了

whlsxl, 2020-02-19:
> > > > > 2月13日新装了一个,同样移动宽带. 一切正常.
> > > > > 你修改过哪些配置?
> > > > 
> > > > 
> > > > 无论是修改了一大堆配置的还是什么都没改过的，我都试过好几遍。。你是哪里的移动宽带啊
> > > 
> > > 
> > > 广东深圳。
> > 
> > 
> > 我这里是广西移动，就算是光猫拨号，也是只有光猫有v6网络，而光猫下的设备只有地址没有网络。
> > 可是openwrt路由器拨号的话，就连路由器自己也是没有v6网络的。。如果用padavan老毛子固件就是和光猫一样的情况
> 
> 看看能不能超管登陆光猫，改成自己拨号。光猫下发的IPv6 是不是直接就是/64的了

我新建了个WAN6，设备选择pppoe-wan，用它获取IPv6 前缀，现在还没完全通，http://test-ipv6.com/说MTU过大，还得查查。
但是ping6 dns 都是通了的

klever1988, 2020-02-20:
LZ还是换上运营商光猫路由排查一下是不是本地的ipv6路由问题

BlacK201, 2020-02-21:
在尝试了多种方法无果之后，我放弃使用IPV6网络了。。谢谢各位老哥的热心帮助

closed 2020-02-21 by BlacK201

whlsxl, 2020-02-21:
> 在尝试了多种方法无果之后，我放弃使用IPV6网络了。。谢谢各位老哥的热心帮助

已经通了，肯定是可以的！
可以试试我上条留言

RicardoKlose, 2020-02-22:
我这里其实也一直有ipv6的问题，我用henet那边提供的6in4方式ipv6，路由端和客户端都可以ping通ipv6的地址，但是ipv6测试页也是一直提示MTU问题，那些ipv6的站点也一直打不开

whlsxl, 2020-02-22:
> 我这里其实也一直有ipv6的问题，我用henet那边提供的6in4方式ipv6，路由端和客户端都可以ping通ipv6的地址，但是ipv6测试页也是一直提示MTU问题，那些ipv6的站点也一直打不开

本来已经通了，今天又挂了。
这些运营商啊。。