coolsnowwolf/lede #3103: 求救（antfs），

kl9800, 2020-02-15:
antfs-2020-02-10-b41ba529.tar.xz（243 KB）已存在于dl文件夹，但编译的时候还是出错。。。

[ ! -d ./src/ ] || cp -fpR ./src/. /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount
touch /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.prepared_b41fe2a0784b77c0f444d4bc00ab7183_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.configured_*
rm -f /home/xpp/lede/staging_dir/target-x86_64_musl/stamp/.antfs-mount_installed
touch /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.built
touch /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.built_check
touch /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.built
rm -rf /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.pkgdir/antfs-mount.installed /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.pkgdir/antfs-mount
mkdir -p /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.pkgdir/antfs-mount
install -d -m0755 /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.pkgdir/antfs-mount/sbin
install -m0755 ./files/mount.ntfs /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.pkgdir/antfs-mount/sbin
touch /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.pkgdir/antfs-mount.installed
mkdir -p /home/xpp/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/xpp/lede/tmp/.root-copy.flock -c 'cp -fpR /home/xpp/lede/build_dir/target-x86_64_musl/antfs-mount/.pkgdir/antfs-mount/. /home/xpp/lede/staging_dir/target-x86_64_musl/root-x86/'
cp: not writing through dangling symlink '/home/xpp/lede/staging_dir/target-x86_64_musl/root-x86/./sbin/mount.ntfs'
Makefile:28: recipe for target '/home/xpp/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.antfs-mount_installed' failed
make[3]: *** [/home/xpp/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.antfs-mount_installed] Error 1
make[3]: Leaving directory '/home/xpp/lede/package/lean/antfs-mount'
time: package/lean/antfs-mount/compile#0.09#0.02#0.11
package/Makefile:107: recipe for target 'package/lean/antfs-mount/compile' failed
make[2]: *** [package/lean/antfs-mount/compile] Error 2
make[2]: Leaving directory '/home/xpp/lede'
package/Makefile:103: recipe for target '/home/xpp/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/xpp/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xpp/lede'
/home/xpp/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

coolsnowwolf, 2020-02-15:
去掉 ntfs-3g

supppig, 2020-02-15:
> 去掉 ntfs-3g

赞，刚好我也遇到这个问题，谢谢老大~~

kl9800, 2020-02-15:
谢谢大佬

closed 2020-02-15 by kl9800