coolsnowwolf/lede #3104: 关于增加驱动

zhangyining215, 2020-02-15:
大神，手里现有一块英特尔540-t2的万兆双电口卡，刷了几个系统后发现需要额外安装驱动，可不可以下次更新时把这个驱动搞进去或者教下怎么手动搞进去吗？？？

hxlls, 2020-02-15:
自己选就完事了

zhangyining215, 2020-02-15:
关键是不会啊，不知道怎么编译

dolphinpaopao, 2020-02-15:
不知道编译，油管上面那么多教程，真的是连搜索一下都不愿意的吗？如此下去永远都不会，我就以 openwrt 添加网卡驱动 为关键字在百度里面搜索第一个网页都是答案。真的服了。学习是需要成本的。

coolsnowwolf, 2020-02-15:
默认是支持的，直接action 里面下载我的固件即可

zhangyining215, 2020-02-16:
> 不知道编译，油管上面那么多教程，真的是连搜索一下都不愿意的吗？如此下去永远都不会，我就以openwrt添加网卡驱动为关键字在百度里面搜索第一个网页都是答案。真的服了。学习是需要成本的。

找不到我需要的啊，现在手里有linux的驱动安装包，不知道要怎么弄

Aes64X, 2020-02-16:
> > 不知道编译，油管上面那么多教程，真的是连搜索一下都不愿意的吗？如此下去永远都不会，我就以openwrt添加网卡驱动为关键字在百度里面搜索第一个网页都是答案。真的服了。学习是需要成本的。
> 
> 找不到我需要的啊，现在手里有linux的驱动安装包，不知道要怎么弄

作者说的还不够明确吗？还是你压根不想动脑子？驱动默认是有的，直接下作者的固件或者自己搞，这还看不明白吗？issue 不是讨论这种问题的地方，麻烦您关闭这个 issue，在这里问下去没有意义，或许你去论坛里求一个还有人给你，但在这里，不好意思，没有。

ZqinKing, 2020-02-16:
这个82599的应该是默认支持的吧

zhangyining215, 2020-02-16:
> 这个82599的应该是默认支持的吧

X540-T2确定支持吗？

ZqinKing, 2020-02-16:
> > 这个82599的应该是默认支持的吧
> 
> X540-T2确定支持吗？

你看下网卡芯片型号

zhangyining215, 2020-02-16:
> > > 这个82599的应该是默认支持的吧
> > 
> > 
> > X540-T2确定支持吗？
> 
> 你看下网卡芯片型号

Intel x540芯片组

ZqinKing, 2020-02-16:
> > > > 这个82599的应该是默认支持的吧
> > > 
> > > 
> > > X540-T2确定支持吗？
> > 
> > 
> > 你看下网卡芯片型号
> 
> Intel x540芯片组

查过了，你这个和82559是一个驱动，ixgbe.ko。大雕的x64固件已经默认带了

zhangyining215, 2020-02-16:
> > > > > 这个82599的应该是预设支持的吧
> > > > 
> > > > 
> > > > X540-T2确定支持吗？
> > > 
> > > 
> > > 你看下网卡芯片型号
> > 
> > 
> > 英特尔x540芯片组
> 
> 查过了，你这个和82599是一个驱动，ixgbe.ko。大雕的x64固件已经重置带了

具体怎么操作？加我qq咱们讨论下把 306783892

closed 2020-02-16 by coolsnowwolf