coolsnowwolf/lede #3105: 最新版R20.2.15编译出错

jxph1123, 2020-02-15:
清除了配置文件，重新配置编译也出错。错误信息如下：
make[3]: Leaving directory '/home/etek/lede/package/lean/ddns-scripts_aliyun'
time: package/lean/ddns-scripts_aliyun/compile#0.21#0.10#0.29
make[3]: Entering directory '/home/etek/lede/package/lean/ddns-scripts_dnspod'
mkdir -p /home/etek/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/etek/lede/tmp/.root-copy.flock -c 'cp -fpR /home/etek/lede/build_dir/target-mipsel_24kc_musl/ddns-scripts_dnspod-1.0.1/.pkgdir/ddns-scripts_dnspod/. /home/etek/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/etek/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.ddns-scripts_dnspod_installed
echo "ddns-scripts_dnspod" >> /home/etek/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/ddns-scripts_dnspod.default.install
make[3]: Leaving directory '/home/etek/lede/package/lean/ddns-scripts_dnspod'
time: package/lean/ddns-scripts_dnspod/compile#0.20#0.10#0.29
make[3]: Entering directory '/home/etek/lede/package/lean/default-settings'
rm -rf /home/etek/lede/build_dir/target-mipsel_24kc_musl/default-settings/.pkgdir/default-settings.installed /home/etek/lede/build_dir/target-mipsel_24kc_musl/default-settings/.pkgdir/default-settings
mkdir -p /home/etek/lede/build_dir/target-mipsel_24kc_musl/default-settings/.pkgdir/default-settings
install -d -m0755 /home/etek/lede/build_dir/target-mipsel_24kc_musl/default-settings/.pkgdir/default-settings/usr/lib/lua/luci/i18n
install -d -m0755 /home/etek/lede/build_dir/target-mipsel_24kc_musl/default-settings/.pkgdir/default-settings/etc/uci-defaults
install -m0644 /home/etek/lede/build_dir/target-mipsel_24kc_musl/default-settings/*.lmo /home/etek/lede/build_dir/target-mipsel_24kc_musl/default-settings/.pkgdir/default-settings/usr/lib/lua/luci/i18n/
install: cannot stat '/home/etek/lede/build_dir/target-mipsel_24kc_musl/default-settings/*.lmo': No such file or directory
Makefile:47: recipe for target '/home/etek/lede/build_dir/target-mipsel_24kc_musl/default-settings/.pkgdir/default-settings.installed' failed
make[3]: *** [/home/etek/lede/build_dir/target-mipsel_24kc_musl/default-settings/.pkgdir/default-settings.installed] Error 1
make[3]: Leaving directory '/home/etek/lede/package/lean/default-settings'
time: package/lean/default-settings/compile#0.19#0.10#0.27
package/Makefile:107: recipe for target 'package/lean/default-settings/compile' failed
make[2]: *** [package/lean/default-settings/compile] Error 2
make[2]: Leaving directory '/home/etek/lede'
package/Makefile:103: recipe for target '/home/etek/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/etek/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/etek/lede'
/home/etek/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

dlj390, 2020-02-15:
same issue

jackiezhang29, 2020-02-15:
同样出错了

EstFoxs, 2020-02-15:
make[3]: *** [Makefile:51: /home/bbq/lede/build_dir/target-x86_64_musl/default-settings/.pkgdir/default-settings.installed] Error 1
make[3]: Leaving directory '/home/bbq/lede/package/lean/default-settings'
time: package/lean/default-settings/compile#0.10#0.01#0.20
make[2]: *** [package/Makefile:109: package/lean/default-settings/compile] Error 2
make[2]: Leaving directory '/home/bbq/lede'
make[1]: *** [package/Makefile:103: /home/bbq/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bbq/lede'
make: *** [/home/bbq/lede/include/toplevel.mk:218: world] Error 2
bbq@bbq:~/lede$

lqhorochi, 2020-02-15:
同样编译出错,一样的问题

hgh188, 2020-02-15:
同样编译出错,一样的问题

xiaozizi58, 2020-02-15:
同样编译出错,一样的问题

helianling, 2020-02-15:
**install -m0644 /home/aaa/lede/build_dir/target-x86_64_musl/default-settings/*.lmo /home/aaa/lede/build_dir/target-x86_64_musl/default-settings/.pkgdir/default-settings/usr/lib/lua/luci/i18n/
install: cannot stat '/home/aaa/lede/build_dir/target-x86_64_musl/default-settings/*.lmo': No such file or directory**

arfaWong, 2020-02-15:
同样编译出错,一样的问题

waterun, 2020-02-15:
yes, same issue

hyird, 2020-02-15:
Makefile 第43行 删掉即可

PR #3107 (2020-02-15): default-settings: fix po2lmo error
fix #3105
closed 2020-02-15 by coolsnowwolf