coolsnowwolf/lede #3115: 编译出错

snakwu, 2020-02-15:
/home/snakwu/openwrt-master/openwrt/staging_dir/host/bin/upx --lzma --best /home/snakwu
/openwrt-master/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-1.3.0/.go_work/build/
bin/v2ray-plugin                                                                       
bash: /home/snakwu/openwrt-master/openwrt/staging_dir/host/bin/upx: No such file or dir
ectory                                                                                 
Makefile:59: recipe for target '/home/snakwu/openwrt-master/openwrt/build_dir/target-x8
6_64_musl/v2ray-plugin-1.3.0/.built' failed                                            
make[3]: *** [/home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/v2ray-pl
ugin-1.3.0/.built] Error 127                                                           
make[3]: Leaving directory '/home/snakwu/openwrt-master/openwrt/package/lean/v2ray-plug
in'                                                                                    
time: package/lean/v2ray-plugin/compile#5.37#1.10#14.20                                
package/Makefile:111: recipe for target 'package/lean/v2ray-plugin/compile' failed     
make[2]: *** [package/lean/v2ray-plugin/compile] Error 2                               
make[2]: Leaving directory '/home/snakwu/openwrt-master/openwrt'                       
package/Makefile:107: recipe for target '/home/snakwu/openwrt-master/openwrt/staging_di
r/target-x86_64_musl/stamp/.package_compile' failed                                    
make[1]: *** [/home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/stamp/
.package_compile] Error 2                                                              
make[1]: Leaving directory '/home/snakwu/openwrt-master/openwrt'                       
/home/snakwu/openwrt-master/openwrt/include/toplevel.mk:225: recipe for target 'world' 
failed                                                                                 
make: *** [world] Error 2

snakwu, 2020-02-15:
我已经关掉upx压缩的了

snakwu, 2020-02-15:
ubuntu 18.0.4
upx 3.96

SharerMax, 2020-02-16:
v2ray-plugin 没有upx的配置选项，默认使用了upx压缩，如果可能的话请使用upx

snakwu, 2020-02-16:
但问题是，我在Project V里面打开了upx，也同样错误

snakwu, 2020-02-16:
原因找到了,原来是tools里面没有生成upx引起的,希望可以默认是关了upx压缩的,担供选项,让有需要的人使用,又或可以使用编译系统里面已安装的upx来进行压缩,谢谢!!!!

Atroc-0625, 2020-02-16:
> 原因找到了,原来是tools里面没有生成upx引起的,希望可以默认是关了upx压缩的,担供选项,让有需要的人使用,又或可以使用编译系统里面已安装的upx来进行压缩,谢谢!!!!

你好，请问要怎么操作，我也是这个问题

lizhenglin2002, 2020-02-16:
我也是这个问题，应该怎么解决

closed 2020-02-16 by snakwu

agutools, 2020-02-16:
已经有编译好的的了，不想压缩就不勾选v2ray相关，直接用原版