coolsnowwolf/lede #3116: 你们的luci-app-dockerman正常？？？

lloyd18, 2020-02-16:
拉不到代码，启动也失败。终端里运行代码没问题

SharerMax, 2020-02-16:
#3088

lloyd18, 2020-02-16:
> #3088

不一样，终端运行代码一点问题没有,与新建文件夹没关系。启Containers: start netdata...fail code:500 driver failed programming external connectivity on endpoint netdata (783b6063eae8e627b243372ec4e84cdf4548e14337e6c61bb4c6690a0fbb6803): (iptables failed: iptables --wait -t nat -A DOCKER -p tcp -d 0/0 --dport 19999 -j DNAT --to-destination 172.17.0.2:19999 ! -i docker0: iptables: No chain/target/match by that name. (exit status 1))动错误也不同

cnekol, 2020-02-16:
#3079 你也是直接升级上来的？

tcsr200722, 2020-02-16:
除了bridge不能连外网，其他的都还好

wgedu, 2020-02-16:
启动失败的，把文件/etc/init.d/dockerd中的START=25改成START=99就可以启动成功了

lloyd18, 2020-02-18:
> 启动失败的，把文件/etc/init.d/dockerd中的START=25改成START=99就可以启动成功了

大佬，方法有用，docker的锅

closed 2020-02-18 by lloyd18