coolsnowwolf/lede #3121: 建议屏蔽PDNSD的AAAA记录

MeIsReallyBa, 2020-02-16:
虽然现在dnsmasq可以屏蔽AAAA记录，但是一刀切后会影响连接国内某些正常的ipv6网站。现在的op应该是国内走运营商dns，国外或者gfwlist走pdnsd。可以调整pdnsd的配置文件，屏蔽AAAA记录。如
server{
label="ssr-usrdns";
ip=8.8.4.4;
port=53;
timeout=6;
uptest=none;
interval=10m;
purge_cache=off;
reject=::/0;
reject_policy=negate;
reject_recursively=on;
}
server{
label="ssr-pdnsd";
ip=208.67.222.222, 208.67.220.220;
port=5353;
timeout=6;
uptest=none;
interval=10m;
purge_cache=off;
reject=::/0;
reject_policy=negate;
reject_recursively=on;
}
server加入这三行即可，这样屏蔽国外的v6，也不影响国内的v6解析
reject=::/0;
reject_policy=negate;
reject_recursively=on;

closed 2020-02-16 by MeIsReallyBa

iky1905, 2021-08-29:
> 并没有用好像。

这个得改对应app的启动脚本，passwall的得改他自己的脚本 ssrplus同理
他们一般是在每次启动的时候会根据启动脚本生成各自的临时配置文件供相关插件用，比如pdnsd dns2socks，所以一般情况下直接改pdnsd的配置文件没啥用
passwall的启动脚本在/usr/share/passwall/app.sh，查找定位到有关pdnsd的地方，加入这几行代码就能用了

blueveryday, 2021-11-29:
> > 并没有用好像。
> 
> 这个得改对应app的启动脚本，passwall的得改他自己的脚本 ssrplus同理 他们一般是在每次启动的时候会根据启动脚本生成各自的临时配置文件供相关插件用，比如pdnsd dns2socks，所以一般情况下直接改pdnsd的配置文件没啥用 passwall的启动脚本在/usr/share/passwall/app.sh，查找定位到有关pdnsd的地方，加入这几行代码就能用了

请问ssrplus如何修改呢？

iky1905, 2022-12-05:
> 

在 `/etc/init.d/shadowsocksr` 里面找类似的文本

iky1905, 2022-12-05:
> > > 并没有用好像。
> > 
> > 
> > 这个得改对应app的启动脚本，passwall的得改他自己的脚本 ssrplus同理 他们一般是在每次启动的时候会根据启动脚本生成各自的临时配置文件供相关插件用，比如pdnsd dns2socks，所以一般情况下直接改pdnsd的配置文件没啥用 passwall的启动脚本在/usr/share/passwall/app.sh，查找定位到有关pdnsd的地方，加入这几行代码就能用了
> 
> 请问ssrplus如何修改呢？

在 `/etc/init.d/shadowsocksr` 里面找，这个文件是ssrplus的启动脚本，跟passwall的那个app.sh是一样地位的