coolsnowwolf/lede #3123: git 安装失败

z5984, 2020-02-16:
我想在我的软路由上安装git服务端
但出现了以下报错

root@OpenWrt:/# opkg install git
Installing git (2.25.0-1) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/packages/git_2.25.0-1_x86_64.ipk
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (librt and librt) providing same name marked HOLD or PREFER. Using latest.
Installing libopenssl1.1 (1.1.1d-2) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/base/libopenssl1.1_1.1.1d-2_x86_64.ipk
Collected errors:
 * check_data_file_clashes: Package libopenssl1.1 wants to install file /usr/lib/libcrypto.so.1.1
        But that file is already provided by package  * libopenssl
 * check_data_file_clashes: Package libopenssl1.1 wants to install file /usr/lib/libssl.so.1.1
        But that file is already provided by package  * libopenssl
 * opkg_install_cmd: Cannot install package git.

SharerMax, 2020-02-16:
建议你用luci界面安装

AmadeusGhost, 2020-02-16:
编译的时候选上git就行了，opkg install容易冲突

slitilanber, 2020-02-16:
opkg install git --force-overwrite

z5984, 2020-02-16:
> opkg install git --force-overwrite

root@OpenWrt:~# opkg install git --force-overwrite
Installing git (2.25.0-1) to root...
Collected errors:
 * opkg_download_pkg: Package git is not available from any configured src.
 * opkg_install_pkg: Failed to download git. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package git.
root@OpenWrt:~#

z5984, 2020-02-16:
> 编译的时候选上git就行了，opkg install容易冲突

我用了别人编译好的固件

z5984, 2020-02-16:
> 建议你用luci界面安装

Thank you !

closed 2020-02-16 by z5984

servicehost, 2020-11-19:
@zhou-daxia 请问一下你是怎么解决的？我现在也遇到这个问题。

zhou-daxia, 2020-11-20:
> @zhou-daxia 请问一下你是怎么解决的？我现在也遇到这个问题。

我是在web上的 系统->软件包 里安装的

servicehost, 2020-11-20:
> 
> 
> > @zhou-daxia 请问一下你是怎么解决的？我现在也遇到这个问题。
> 
> 我是在web上的 系统->软件包 里安装的

谢谢，但现在官方源不知怎么了，git、node、libopenssl1.1.. 等全消失了.. 连安装都安不了。