coolsnowwolf/lede #3128: 希望默认v2ray-plugin的upx关掉,又或者提供选项,谢谢

snakwu, 2020-02-16:
希望默认v2ray-plugin的upx关掉,又或者提供选项,让有需要的人使用,又或者优先使用编译系统自身安装的upx来压缩,谢谢!

dolphinpaopao, 2020-02-16:
自己编译前更改一下y为n保存不就完事了吗。。。。

snakwu, 2020-02-16:
> 自己编译前更改一下y为n保存不就完事了吗。。。。

并不行

dolphinpaopao, 2020-02-16:
根据作者默认压缩的更改，自行改回原来的就是，刚看了一下，应该还要删除makefile里面的第51行$(STAGING_DIR_HOST)/bin/upx --lzma --best $(GO_PKG_BUILD_BIN_DIR)/v2ray-plugin

snakwu, 2020-02-16:
因为,考虑到网络原因,因为,google.golang.org是经常抽风的,也同时考虑到用虚拟机的朋友,可以不需机upx,所以,希望提供选项,不默认关闭也行

snakwu, 2020-02-16:
因为,考虑到网络原因,因为,google.golang.org是经常抽风的,也同时考虑到用虚拟机的朋友,可以不需机upx,所以,,不默认关闭也行,希望提供选项!

panther706, 2020-02-16:
你那是什么型号？我的是网件WNDR4300，v2ray也是开不起。查看了UPX已经是3.96了还是不行，
一直在”应该更改“那里转圏圈。查看了下系统进程，CPU占用很高

8620 | root | /usr/bin/v2ray/v2ray -config /var/etc/v2-ssr-retcp.json | 50% | 16% |   |   |  
-- | -- | -- | -- | -- | -- | -- | --
8623 | root | /bin/sh /etc/rc.common /etc/init.d/shadowsocksr reload | 0% | 1% |   |   |  
8625 | root | /usr/bin/v2ray/v2ray -version | 33% | 16%

pwh@pwh-virtual-machine:~$ upx
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2020
UPX 3.96        Markus Oberhumer, Laszlo Molnar & John Reiser   Jan 23rd 2020

Usage: upx [-123456789dlthVL] [-qvfk] [-o file] file..

Commands:
  -1     compress faster                   -9    compress better
  -d     decompress                        -l    list compressed file
  -t     test compressed file              -V    display version number
  -h     give more help                    -L    display software license
Options:
  -q     be quiet                          -v    be verbose
  -oFILE write output to 'FILE'
  -f     force compression of suspicious files
  -k     keep backup files
file..   executables to (de)compress

Type 'upx --help' for more detailed help.

UPX comes with ABSOLUTELY NO WARRANTY; for details visit https://upx.github.io

snakwu, 2020-02-16:
> 你那是什么型号？我的是网件WNDR4300，v2ray也是开不起。查看了UPX已经是3.96了还是不行，
> 一直在”应该更改“那里转圏圈。查看了下系统进程，CPU占用很高
>
> 8620	root	/usr/bin/v2ray/v2ray -config /var/etc/v2-ssr-retcp.json	50%	16%
> 8623	root	/bin/sh /etc/rc.common [...]

我是x86,和你情况不一样,我是编译时,tools没有生成upx文件引起的,这个upx最近引起了好多鬼怪问题

d975xbx, 2020-02-16:
正好相反，没有upx。我开不了v2，现在一起正常了。

snakwu, 2020-02-16:
> 正好相反，没有upx。我开不了v2，现在一起正常了。

那是因为你的路由内存少的原因吧?所以,才希望提供一个选项,让大家按需所取!

d975xbx, 2020-02-16:
x64.2g内存

d975xbx, 2020-02-16:
现在的源码，你不用管什么upx，什么都不用做，默认编译就ok，最起码我是能编译，v2也正常启动

dolphinpaopao, 2020-02-16:
> 现在的源码，你不用管什么upx，什么都不用做，默认编译就ok，最起码我是能编译，v2也正常启动

是的，我一直很少遇到什么莫名其妙的问题，就算是升级，也是升级后恢复出厂重新配置。虽然繁琐一点，但是很稳妥。

AmadeusGhost, 2020-02-16:
v2ray是有提供是否使用upx的选项的，v2ray-plugin要可以写一个，但是v2ray-plugin是基于v2ray的一个扩展（SIP003 plugin for shadowsocks, based on v2ray），不是vmess用的
> 因为,考虑到网络原因,因为,google.golang.org是经常抽风的,也同时考虑到用虚拟机的朋友,可以不需机upx,所以,希望提供选项,不默认关闭也行

这个说的也不对，upx跟网络无关，而且go可以使用国内代理的
```
export GO111MODULE=on
export GOPROXY=https://goproxy.cn
```
[image]

Aes64X, 2020-02-16:
我不是很确定。
我使用EA4500 V1 kirkwood 1.2GHz
包含UPX时生成的固件会无法刷入，从原厂刷入，没有任何的错误记录，提示错误。
去掉v2ray后生成的固件则正常写入。

snakwu, 2020-02-16:
我网络渣，就是因为网络差所以tools里面没生成upx，所以报错，希望可以加个开关，再说我用的是x86也不需要upx，谢谢！

d975xbx, 2020-02-16:
可能你的不需要，但我的x64必须要upx，昨天以前我是手动修改文件才能启动v2，昨天更新源码以后，一切都好了。

yyssxx, 2020-02-16:
newifi3,最新R20.2.15源码，v2ray不能启动，关掉upx再编译，就可以了

snakwu, 2020-02-16:
所以各说各的，才有这需求，加个选项多好

AmadeusGhost, 2020-02-16:
> 我网络渣，就是因为网络差所以tools里面没生成upx，所以报错，希望可以加个开关，再说我用的是x86也不需要upx，谢谢！

网差不是理由啊。。。编译不挂梯很多东西都下不下来，实在网络有问题就github云编译啊。

snakwu, 2020-02-16:
哦，那随便了，加不加无所谓了

snakwu, 2020-02-16:
只是说应该加个开关，对应各需求的人，有些人说打开能运行，有些说关了能运行，加个开关，可以满足各需求，但如果麻烦，当我没说过，网络差不差我不是什么理由，但我在确是这样，所以才引起tools没生成upx，当然我自己解决了，只是考虑大家才提的问题，但看你口气不高兴了，你高兴就行，当我没提过，我自关，打扰，再见

closed 2020-02-16 by snakwu

AmadeusGhost, 2020-02-16:
> 只是说应该加个开关，对应各需求的人，有些人说打开能运行，有些说关了能运行，加个开关，可以满足各需求，但如果麻烦，当我没说过，网络差不差我不是什么理由，但我在确是这样，所以才引起tools没生成upx，当然我自己解决了，只是考虑大家才提的问题，但看你口气不高兴了，你高兴就行，当我没提过，我自关，打扰，再见

不是，我没有不高兴…
只是网差容易引起编译失败，而且github actions也是一个好选择

dolphinpaopao, 2020-02-16:
> 只是说应该加个开关，对应各需求的人，有些人说打开能运行，有些说关了能运行，加个开关，可以满足各需求，但如果麻烦，当我没说过，网络差不差我不是什么理由，但我在确是这样，所以才引起tools没生成upx，当然我自己解决了，只是考虑大家才提的问题，但看你口气不高兴了，你高兴就行，当我没提过，我自关，打扰，再见

太敏感了吧，人家也只是陈述一个事实，况且C也是建议全程FQ编译，国内网络环境众所周知。

snakwu, 2020-02-16:
网络问题，其实不难解决，但是会解决的也应该还有人在，我提问题，只是出发大众的角度，希望这个项目更好的兼容人群！

andrewstar, 2020-02-16:
newifi3 D2,最新R20.2.15源码，v2ray不能启动，日志是这样的：2020-02-16 17:12:18  Started!
2020-02-16 17:12:04 Try restart current server.
2020-02-16 17:10:56  Started!
2020-02-16 17:11:57 Current server error, try to switch another server.
2020-02-16 17:11:54 Main server is NOT avilable.Continue using current server.
2020-02-16 17:11:42 Current server is not default Main server, try to switch back.
2020-02-16 17:10:41 Try restart current server.
2020-02-16 17:09:44  Started!
2020-02-16 17:10:36 Current server error, try to switch another server.
2020-02-16 17:10:32 Main server is NOT avilable.Continue using current server.
2020-02-16 17:10:20 Current server is not default Main server, try to switch back.
2020-02-16 17:09:20 Try restart current server.
2020-02-16 17:08:13  Started!
2020-02-16 17:09:14 Current server error, try to switch another server.
2020-02-16 17:09:11 Main server is NOT avilable.Continue using current server.
2020-02-16 17:08:59 Current server is not default Main server, try to switch back.
2020-02-16 17:07:59 Try restart current server.
2020-02-16 17:06:41  Started!
2020-02-16 17:07:53 Current server error, try to switch another server.
2020-02-16 17:07:49 Main server is NOT avilable.Continue using current server.
2020-02-16 17:07:37 Current server is not default Main server, try to switch back.
2020-02-16 17:06:37 Try restart current server.
2020-02-16 17:05:20  Started!
2020-02-16 17:06:31 Current server error, try to switch another server.
2020-02-16 17:06:28 Main server is NOT avilable.Continue using current server.
2020-02-16 17:06:16 Current server is not default Main server, try to switch back.
2020-02-16 17:05:16 Try restart current server.
2020-02-16 17:04:08  Started!
2020-02-16 17:05:10 Current server error, try to switch another server.
2020-02-16 17:05:07 Main server is NOT avilable.Continue using current server.
2020-02-16 17:04:55 Current server is not default Main server, try to switch back.
2020-02-16 17:03:54 Try restart current server.
2020-02-16 17:02:38  Started!
CPU占用特别高，几乎 99，luci非常卡，几乎进不去管理界面了

snakwu, 2020-02-16:
> > 只是说应该加个开关，对应各需求的人，有些人说打开能运行，有些说关了能运行，加个开关，可以满足各需求，但如果麻烦，当我没说过，网络差不差我不是什么理由，但我在确是这样，所以才引起tools没生成upx，当然我自己解决了，只是考虑大家才提的问题，但看你口气不高兴了，你高兴就行，当我没提过，我自关，打扰，再见
> 
> 不是，我没有不高兴…
> 只是网差容易引起编译失败，而且github actions也是一个好选择

那是我误会你了，在这和你说过对不起，我自己解决了，谢谢！

panther706, 2020-02-16:
> newifi3 D2,最新R20.2.15源码，v2ray不能启动，日志是这样的：2020-02-16 17:12:18 Started!
> 2020-02-16 17:12:04 Try restart current server.
> 2020-02-16 17:10:56 Started!
> 2020-02-16 17:11:57 Current server error, try to [...]

我的情况和你一样

AmadeusGhost, 2020-02-16:
> newifi3 D2,最新R20.2.15源码，v2ray不能启动，日志是这样的：2020-02-16 17:12:18 Started!
> CPU占用特别高，几乎 99，luci非常卡，几乎进不去管理界面了

v2ray问题请另开issue

AmadeusGhost, 2020-02-16:
> 那是我误会你了，在这和你说过对不起，我自己解决了，谢谢！

没事，网差还是用云编译好点
https://github.com/coolsnowwolf/lede/actions
https://github.com/P3TERX/Actions-OpenWrt

snakwu, 2020-02-16:
> > 那是我误会你了，在这和你说过对不起，我自己解决了，谢谢！
> 
> 没事，网差还是用云编译好点
> https://github.com/coolsnowwolf/lede/actions
> https://github.com/P3TERX/Actions-OpenWrt

云编是给没条件的朋友准备的，就时间上来说，云编一修固件要6小时，本地编译，make clean后用j5，两小时出固件，普通更新，10分多钟一个固件，不是能比的