coolsnowwolf/lede #3132: ssrplus V2RAY模式不能启动，直接假死。。

jxph1123, 2020-02-16:
如题，V2RAY模式直接在保存应用界面无响应假死，也没有日志。。。SSR日志和系统日志里面都没有错误信息，就是死了。。。切换到SSR模式就正常，最新版本。

coolsnowwolf, 2020-02-16:
什么架构？是否启用 upx 压缩

yumibing, 2020-02-16:
我也是,设备k2p

jxph1123, 2020-02-16:
MTK，默认启用了upx 压缩

yumibing, 2020-02-16:
L大给解决下，谢谢

yumibing, 2020-02-16:
> 什么架构？是否启用 upx 压缩
L大给解决下

coolsnowwolf, 2020-02-16:
取消 UPX 压缩试试，有可能是 UPX 在 MIPS 上面的问题

yumibing, 2020-02-16:
怎么操作取消，取消了东西放不下，绝望

panther706, 2020-02-16:
我的是网件WNDR4300，v2ray也是开不起。
一直在”应用更改“那里转圏圈。查看了下系统进程，CPU占用很高,有时100

8620 | root | /usr/bin/v2ray/v2ray -config /var/etc/v2-ssr-retcp.json | 50% | 16% |   |   |  
-- | -- | -- | -- | -- | -- | -- | --
8623 | root | /bin/sh /etc/rc.common /etc/init.d/shadowsocksr reload | 0% | 1% |   |   |  
8625 | root | /usr/bin/v2ray/v2ray -version | 33% | 16% |   |

yumibing, 2020-02-16:
> 我的是网件WNDR4300，v2ray也是开不起。
> 一直在”应用更改“那里转圏圈。查看了下系统进程，CPU占用很高,有时100
> 
> 8620	root	/usr/bin/v2ray/v2ray -config /var/etc/v2-ssr-retcp.json	50%	16%	 	 	 
> 8623	root	/bin/sh /etc/rc.common /etc/init.d/shadowsocksr reload	0%	1%	 	 	 
> 8625	root	/usr/bin/v2ray/v2ray -version	33%	16%

我也是这个情况，一直转圈，没应用成功

panther706, 2020-02-16:
> > 我的是网件WNDR4300，v2ray也是开不起。
> > 一直在”应用更改“那里转圏圈。查看了下系统进程，CPU占用很高,有时100
> > 8620	root	/usr/bin/v2ray/v2ray -config /var/etc/v2-ssr-retcp.json	50%	16%	 	 	 
> > 8623	root	/bin/sh /etc/rc.common /etc/init.d/shadowsocksr reload	0%	1%	 	 	 
> > 8625	root	/usr/bin/v2ray/v2ray -version	33%	16%
> 
> 我也是这个情况，一直转圈，没应用成功

你看是不是也是这两个进程

panther706, 2020-02-16:
> 什么架构？是否启用 upx 压缩

原始代码编译

jxph1123, 2020-02-16:
应该就是UPX的问题，我取消了重新编译，测试后再来反馈。

yumibing, 2020-02-16:
> 应该就是UPX的问题，我取消了重新编译，测试后再来反馈。

怎么取消，取消了固件会变大吧

yumibing, 2020-02-16:
> 你看是不是也是这两个进程

是的

yumibing, 2020-02-16:
希望大佬修复下，小内存k2p伤不起，谢谢

jxph1123, 2020-02-16:
是不是新版本的UPX的原因？

coolsnowwolf, 2020-02-16:
在 v2ray 的config 中，选择不进行 upx 压缩，然后试试是否可以运行 /usr/bin/v2ray/v2ray -version

dolphinpaopao, 2020-02-16:
> 希望大佬修复下，小内存k2p伤不起，谢谢

不是修复不修复，这个在编译的时候可以自主选择y/n的啊。

yyssxx, 2020-02-16:
关闭upx，再编译就可以了，v2ray正常使用

yyssxx, 2020-02-16:
关掉upx再编译，v2ray就正常启动运行了

d975xbx, 2020-02-16:
每个人硬件和使用环境不一样，为难死大神们了。
我是希望不要关闭upx，就现在这套源码最好，开启upx，v2才能正常启动。
关闭upx，我v2不能启动。x86-64。

hgh188, 2020-02-16:
取消upx 压缩后一切正常

d975xbx, 2020-02-16:
有人关闭upx压缩才可以，有人开启才可以。就这样了吧。

jxph1123, 2020-02-16:
MIPS要关闭，x86-64要开启。。。硬件不一样

coolsnowwolf, 2020-02-16:
修复了，MIPS 太老了。。。

yumibing, 2020-02-16:
感谢大佬，我去编译一个试试

jxph1123, 2020-02-16:
感谢大佬修复

closed 2020-02-16 by coolsnowwolf