coolsnowwolf/lede #3133: 请问能否在SSR的v2ray配置中提供自定义JSON的选项？

diligentpig, 2020-02-16:
各位好，最近开始关注 Lean大的OpenWrt，想使用其中v2ray的功能。在电脑端我可以通过v2ray的客户端JSON文件里配置多个outbound object和routing之后，

* 实现socks+ws+tls替代vmess+ws+tls以提升速度： https://github.com/v2ray/discussion/issues/513
* 实现根据不同域名分流到多个服务器的功能（不同vps的价格有高低。。。）

但目前发现ShadowSocksR Plus+应用里的v2ray不能添加自定的JSON配置文件。在repo里找到里一些相同需求issue，

* https://github.com/coolsnowwolf/lede/issues/562
* https://github.com/coolsnowwolf/lede/issues/713
* https://github.com/coolsnowwolf/lede/issues/1232#issuecomment-496877981

但似乎发现都没有明确的方法。想请问大家，类似的思路是否能在目前的SSR Plus+应用里实现呢？如果不行的话, @coolsnowwolf 大，请问是否考虑加入v2ray的自定义JSON配置的形式呢？这样除了订阅和相对简单的配置，也可以支持自己VPS上比较灵活的设置。谢谢！

coolsnowwolf, 2020-02-16:
不提供，那样就复杂化了，SSR Plus 主体思想就是 keep it simple

diligentpig, 2020-02-20:
@coolsnowwolf 谢谢回复。那如果ssh之后找到v2ray运行时的配置文件位置，手动将自己配置文件里outbounds部分以及routing部分加进去，这个思路是否可行呢？这样UI上也不需要暴露出来复杂的参数设置……

tanst, 2020-02-22:
> @coolsnowwolf 谢谢回复。那如果ssh之后找到v2ray运行时的配置文件位置，手动将自己配置文件里outbounds部分以及routing部分加进去，这个思路是否可行呢？这样UI上也不需要暴露出来复杂的参数设置……

同需求，哥们儿解决没有，这里给你参考一下：https://github.com/coolsnowwolf/lede/issues/1134

closed 2020-03-06 by coolsnowwolf

zhudan, 2020-03-30:
同求…