coolsnowwolf/lede #3134: 源码文件包为什么总是下载２次？第一次已经下载好了，又开始重新下载过。。。

haxc, 2020-02-16:
源码文件包为什么总是下载２次？第一次已经下载好了，又开始重新下载过。。。

--2020-02-16 12:23:17--  (try: 4)  http://117.128.6.16/cache/www.lwfinger.com/b43-firmware/broadcom-wl-5.100.138.tar.bz2?ich_args2=521-16202202058079_92d7c004d0acb32181a2edb2901b3ee4_10001002_9c896c2ddfc6f4d6953c518939a83798_f4d0e4a295fd91f37dc390036fb2d473
Connecting to 117.128.6.16:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 13514651 (13M) [application/x-tar]
Saving to: 'STDOUT'

-                             100%[================================================>]  12.89M  5.38KB/s    in 32m 53s 

2020-02-16 12:56:26 (6.69 KB/s) - written to stdout [13514651/13514651]

Hash of the downloaded file does not match (file: b0f320e0d1fed5d5ee8b159dbe83d8c99a2a7ede0d03b327e59f63280ecc21b6, requested: f1e7067aac5b62b67b8b6e4c517990277804339ac16065eb13c731ff909ae46f) - deleting download.
Can't exec "curl": No such file or directory at /home/haxc/Desktop/openwrt/scripts/download.pl line 77.
+ wget --tries=5 --timeout=20 --no-check-certificate --output-document=- https://sources.openwrt.org/broadcom-wl-5.100.138.tar.bz2
--2020-02-16 12:56:26--  https://sources.openwrt.org/broadcom-wl-5.100.138.tar.bz2
Resolving sources.openwrt.org (sources.openwrt.org)... 2a01:4f8:210:5087::2, 148.251.151.136
Connecting to sources.openwrt.org (sources.openwrt.org)|2a01:4f8:210:5087::2|:443... failed: Connection timed out.
Connecting to sources.openwrt.org (sources.openwrt.org)|148.251.151.136|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 13514651 (13M) [application/octet-stream]
Saving to: 'STDOUT'

-                             100%[================================================>]  12.89M   969KB/s    in 15s     

2020-02-16 12:57:07 (894 KB/s) - written to stdout [13514651/13514651]

mkdir -p /home/haxc/Desktop/openwrt/dl

star-o0O, 2020-02-16:
Hash of the downloaded file does not match (file: b0f320e0d1fed5d5ee8b159dbe83d8c99a2a7ede0d03b327e59f63280ecc21b6, requested: f1e7067aac5b62b67b8b6e4c517990277804339ac16065eb13c731ff909ae46f)

coolsnowwolf, 2020-02-16:
hash 不对，第一次的

closed 2020-02-16 by haxc