coolsnowwolf/lede #3135: 内核级 Samba 文件共享服务器 CIFSD 正式开始测试

coolsnowwolf, 2020-02-16:
CISFD 是一个使用内核模块的 SAMBA 1/2/3 服务器，可以提供比 Samba3/4 这种用户态的程序提供高得多的性能和低延迟，大并发

配合最近的内核态 antfs 的 NTFS 读写文件驱动，应该是补上了 OpenWrt 文件共享的性能问题

大家编译的时候选上这个 luci-app-cifsd 测试下反馈。

注意，和 Samba 冲突，需要取消。 如果效果好以后就是默认了

[image]

hcym, 2020-02-16:
感谢大雕

unifreq, 2020-02-16:
反馈：环境：Arm64环境
测试1:默认配置下报错：
[   94.277834] ksmbd: kill_server_store:437: kill command received
[   94.331114] ksmbd: ksmbd_tcp_set_interfaces:596: Unknown interface: 
[  113.961601] ksmbd: kill_server_store:437: kill command received
[  835.181268] ksmbd: ksmbd_tcp_set_interfaces:596: Unknown interface: 
[ 1093.833659] ksmbd: ksmbd_tcp_set_interfaces:596: Unknown interface: 
[ 1307.474413] ksmbd: ksmbd_tcp_set_interfaces:596: Unknown interface: 
[ 1385.562898] ksmbd: kill_server_store:437: kill command received
[ 1443.247842] ksmbd: kill_server_store:437: kill command received
[ 1473.398408] ksmbd: ksmbd_tcp_set_interfaces:596: Unknown interface: 

smb.conf部分内容如下：
        interfaces = eth0 
        bind interfaces only = yes
        ipc timeout = 20
        deadtime = 15
        map to guest = Bad User
        smb2 max read = 64K
        smb2 max write = 64K
        smb2 max trans = 64K
        cache read buffers = no
        cache trans buffers = no

ps -ef|grep ksmb
23186 root      1168 S    /usr/sbin/ksmbd.mountd --n
23188 root      1336 S    /usr/sbin/ksmbd.mountd --n

lsmod|grep ksmb
ksmbd                 249856  0

netstat -an|grep 445 无结果

测试2：
把/etc/ksmbd/ksmbd.config.example 拷贝成 /etc/ksmbd/smb.conf同样不行：
daemon.notice ksmbd: Starting Ksmbd userspace service.

smb.conf如下：
config globals
        option 'description'    'Ksmbd on OpenWrt'

config share
        option name 'testshare'
        option path '/tmp'
        option read_only 'no'
        option guest_ok 'yes'
        option create_mask '0666'
        option dir_mask '0777'
        option force_root '1'

ps -ef|grep ksmb
23974 root      1168 S    /usr/sbin/ksmbd.mountd --n

lsmod|grep ksmb
ksmbd                 249856  0

netstat -an|grep 445 无结果

jy3344511, 2020-02-16:
> CISFD 是一个使用内核模块的 SAMBA 1/2/3 服务器，可以提供比 Samba3/4 这种用户态的程序提供高得多的性能和低延迟，大并发
> 
> 配合最近的内核态 antfs 的 NTFS 读写文件驱动，应该是补上了 OpenWrt 文件共享的性能问题
> 
> 大家编译的时候选上这个 luci-app-cifsd 测试下反馈。
> 
> 注意，和 Samba 冲突，需要取消。 如果效果好以后就是默认了
> 
> [image]

编译完，win10共享不了CISFD指定的目录，只用Samba 4的时候，可以共享。

tofuliang, 2020-02-16:
和NFS性能对比 测试有么？

monsterzzzz, 2020-02-17:
反馈：
windows 10 无法扫描到共享，通过\\ip，\\计算机名可以访问
原因是wsdd2的service报错，日志如下：
Mon Feb 17 08:25:56 2020 user.info wsdd2[29465]: starting.
Mon Feb 17 08:25:56 2020 user.err wsdd2[29465]: cannot read workgroup from testparm
Mon Feb 17 08:25:56 2020 user.err wsdd2[29465]: cannot read netbios name from testparm
Mon Feb 17 08:26:42 2020 user.err wsdd2[29465]: Interrupt received.
Mon Feb 17 08:26:42 2020 user.info wsdd2[29465]: terminating.
如果在SSH下直接运行wsdd2，会显示：
cannot read workgroup from testparm
cannot read netbios name from testparm
但是进程没有退出，这时可以在windows 10中扫描到
速度方面，感觉和samba4相当，但是读写的时候系统显示 system load超级高

monsterzzzz, 2020-02-17:
/etc/init.d/wsdd2里面$board_model变量的值含有冒号，没处理，因此wsdd2程序不能启动

kltk, 2020-02-17:
上次添加的时候只能设置共享为小写，不知道现在怎么样？

chacha20, 2020-02-17:
我说怎么编译报错了，原来是勾选了samba

jxph1123, 2020-02-18:
测试环境：WIN7
路由器硬件：MIPS
测试情况：CISFD拷贝速度6M/S，原samba速度2M/S，快了三倍左右。
存在问题：1.拷贝时负载过高，达到60多，原samba只有负载1.5，原samba拷贝负载涨幅不大。
                  2.不能自动共享，只能手动挂载共享。

georip, 2020-02-18:
体验挺好，在win7下   网络邻居 不能发现 。只有通过ip直接访问可以。

unifreq, 2020-02-18:
反馈：在Arm64平台上，第一次开机启动后服务不可用，内核日志：
[   27.161191] ksmbd: kill_server_store:480: kill command received
手动重启ksmbd之后服务可用

在modprobe ksmbd之前延时1秒后问题解决。
sed -e 's/modprobe ksmbd/sleep 1 && modprobe ksmbd/' -i /etc/init.d/ksmbd

但速度好象不如cifsd2.0.5快， 老版本的读取速度能达到100m，新版本只有60-80m而且波动较大。

monsterzzzz, 2020-02-18:
感觉负载高只是表象，实际CPU占用甚至比samba4略低，速度完全相当
两者都是读100（接近机械硬盘上限），写60-90波动，ext4分区，x86_64

coolsnowwolf, 2020-02-19:
system load 高是正常的，因为跑在内核，实际 CPU 占用不高

jxph1123, 2020-02-19:
反馈一个问题，安卓手机打开共享时，只能看到初级目录，打开后里面什么都没有。电脑正常。

jxph1123, 2020-02-19:
反馈一个问题，安卓手机打开共享时，只能看到初级目录，打开后里面什么都没有。电脑正常。手机用的是ES浏览器。

chenxiccc, 2020-02-19:
你好，上文提到与samba有冲突，需要取消Samba。不过我在编译make menuconfig时，luci-app-samba是默认选中且不可取消状态，请问需要怎么处理才能取消luci-app-samba的选择呢？

jxph1123, 2020-02-20:
要取消掉

> 你好，上文提到与samba有冲突，需要取消Samba。不过我在编译make menuconfig时，luci-app-samba是替换并且不可取消状态，请问如何处理才能取消luci-app-samba的选择呢？

要取消掉 Autosamba

guochan2019, 2020-02-20:
x86系统，昨晚测试，编译正常，启动后，系统日志显示：
ksmbd: kill_server_store:480: kill command received
貌似服务没启动，windows10在网络共享看不到共享文件夹，运行\\IP可以访问，但无法进入目标文件夹

zhg-em, 2020-02-20:
我也测试了一下，windows10在网络共享能看到，但打不开
[image]
安卓手机打开共享时，只能看到初级目录，打开后里面什么都没有。

hgh188, 2020-02-20:
Windows Registry Editor Version 5.00
[HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\LanmanWorkstation\Parameters]

"AllowInsecureGuestAuth"=dword:00000001

windows10下要把上面文本保存成reg文件导入就可以了

zhg-em, 2020-02-20:
> Windows Registry Editor Version 5.00
> [HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\LanmanWorkstation\Parameters]
> 
> "AllowInsecureGuestAuth"=dword:00000001
> 
> windows10下要把上面文本保存成reg文件导入就可以了

谢谢！那这样电脑上就可以了。但安卓手机不正常！

hgh188, 2020-02-20:
我安卓和IOS都正常

jxph1123, 2020-02-20:
我也是安卓手机不行，楼上的如何配置的呢？

wspzf, 2020-02-21:
试了一下，win10和手机都能访问。
sv3-26026拷贝电影大概速率是24MB/s
请问怎么添加非匿名用户？百度了一下 ksmbd添加非匿名用户的命令是：   ksmbd.adduser -a 用户名
但是系统里没有ksmbd.adduser这个命令

wspzf, 2020-02-22:
> 试了一下，win10和手机都能访问。
> 软路由sv3-26026 拷贝电影
> USB接口移动硬盘读写速率是25MB/s
> SATA接口硬盘读写稳定在70MB/s
> 
> 请问怎么添加非匿名用户？百度了一下 ksmbd添加非匿名用户的命令是： ksmbd.adduser -a 用户名
> 但是系统里没有ksmbd.adduser这个命令

解决了 装一下 ksmbd-utils 这个包就有ksmbd.adduser命令了

zhg-em, 2020-02-22:
> 我安卓和IOS都正常
> 
> 昨晚更新后我的安卓也不行了
> 
> [17630.727945] ksmbd: ksmbd_smb1_check_message:258: Invalid word count, 13 not -22. cmd 73

是这样的，我的华为安卓手机，用系统自带文件管理器，打不开，显示上面的错误信息。

用cx文件管理器，可以看到共享目录，但打不开，显示下面的错误信息

[ 1090.947546] ksmbd: smb2_query_dir:3468: no right to enumerate directory (/)

hgh188, 2020-02-22:
[ 7597.828501] ksmbd: kill_server_store:480: kill command received
[ 8734.010848] ksmbd: ksmbd_smb1_check_message:258: Invalid word count, 13 not -22. cmd 73
[ 8819.201225] ksmbd: ksmbd_smb1_check_message:258: Invalid word count, 13 not -22. cmd 73
[ 8879.108250] ksmbd: ksmbd_smb1_check_message:258: Invalid word count, 13 not -22. cmd 73
[ 8910.246631] ksmbd: ksmbd_smb1_check_message:258: Invalid word count, 13 not -22. cmd 73
win10和IOS都能访问   安卓的不行
Sat Feb 22 16:43:33 2020 kern.err kernel: [ 8879.108250] ksmbd: ksmbd_smb1_check_message:258: Invalid word count, 13 not -22. cmd 73
Sat Feb 22 16:44:04 2020 kern.err kernel: [ 8910.246631] ksmbd: ksmbd_smb1_check_message:258: Invalid word count, 13 not -22. cmd 73

coolsnowwolf, 2020-02-22:
看来和 SMBv1 客户端还是有兼容性问题

unifreq, 2020-02-22:
华为Mate9的文件管理没问题

monsterzzzz, 2020-02-22:
这个项目默认没打开SMB1，需要在配置文件里设置

xxp2277, 2020-02-23:
> 
> 
> 这个项目默认没打开SMB1，需要在配置文件里设置

咨询下，配置文件应该怎么写打开SMBv1？网上没搜索到。

pk49800, 2020-02-23:
Win10可以匿名访问设置好的共享文件夹（使用KODI），IOS使用Infuse（视频播放软件）可发现路由器但访问报错

monsterzzzz, 2020-02-23:
> > 这个项目默认没打开SMB1，需要在配置文件里设置
> 
> 咨询下，配置文件应该怎么写打开SMBv1？网上没搜索到。

https://github.com/namjaejeon/cifsd-tools/blob/master/Documentation/configuration.txt

yunmi323, 2020-02-24:
win10默认关闭SMB1，SMB2/3默认打开的，如果SMB1有兼容问题应该不要支持SMB1，反正SMB1不安全

lyin888, 2020-02-24:
同问SMB1设置参数

loungee, 2020-02-26:
默认状态下win 2016 server 无法使用，试了试安卓手机的kodi可以看见和使用，但好像总有些莫名其妙的问题，好像和nfs有冲突

closed 2020-03-06 by coolsnowwolf

123834, 2020-08-04:
> 试了一下，win10和手机都能访问。
> 软路由sv3-26026 拷贝电影
> USB接口移动硬盘读写速率是25MB/s
> SATA接口硬盘读写稳定在70MB/s
> 请问怎么添加非匿名用户？百度了一下 ksmbd添加非匿名用户的命令是： ksmbd.adduser -a 用户名
> 但是系统里没有ksmbd.adduser这个命令
> 
> 解决了 装一下 ksmbd-utils 这个包就有ksmbd.adduser命令了
ksmbd-utils这个包在哪，我编译的时候没找到呢