coolsnowwolf/lede #3137: luci-app-frpc检测到递归依赖项

ITdesk01, 2020-02-16:
### 情景
    将大雕的package/lean文件夹放到了openwrt 官方master的源码中，编译报错
    编译机型：r7800
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-frpc:
 * 	frpc
 * opkg_install_cmd: Cannot install package luci-app-frpc.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/zhang/Openwrt/op/lede'
package/Makefile:108: recipe for target '/home/zhang/Openwrt/op/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_install' failed
make[1]: *** [/home/zhang/Openwrt/op/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/zhang/Openwrt/op/lede'
/home/zhang/Openwrt/op/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

### make menuconfig
tmp/.config-package.in:133984:error:` recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:133984:	symbol PACKAGE_frpc is selected by PACKAGE_luci-app-frpc
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:74050:	symbol PACKAGE_luci-app-frpc depends on PACKAGE_frpc
tmp/.config-package.in:134722:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:134722:	symbol PACKAGE_nginx-ssl-util depends on PACKAGE_nginx-util
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:134761:	symbol PACKAGE_nginx-util depends on PACKAGE_nginx-ssl-util
configuration written to .config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

haxc, 2020-02-17:
https://www.right.com.cn/forum/thread-306836-1-1.html
用这个

ITdesk01, 2020-02-17:
> https://www.right.com.cn/forum/thread-306836-1-1.html
> 用这个

我还是比较想用大雕原版的frp，感谢

haxc, 2020-02-18:
这个就是根据大雕修改的

coolsnowwolf, 2020-02-20:
官方自己就有 frpc ，重复了大概

ITdesk01, 2020-02-20:
> 官方自己就有frpc，重复了大概

还真是，没想到官方竟然有，感谢回复

closed 2020-02-20 by ITdesk01