coolsnowwolf/lede #3138: DDNS报错wget 报错，代码8，无法解析

authon, 2020-02-16:
231253       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/MYDDNS_IPV4.dat -o /var/run/ddns/MYDDNS_IPV4.err --no-proxy 'http://alidns.aliyuncs.com/?Action=DescribeSubDomainRecords&SubDomain=****.com&Format=JSON&Version=2015-01-09&AccessKeyId=***************&SignatureMethod=HMAC-SHA1&Timestamp=2020-02-16T15%3A12%3A53Z&SignatureVersion=1.0&SignatureNonce=************&Signature='

 231254 ERROR : wget 错误代码: '8'

 231254       : http://alidns.aliyuncs.com/?Action=DescribeSubDomainRecords&SubDomain=****.com&Format=JSON&Version=2015-01-09&AccessKeyId=--------------------------&SignatureMethod=HMAC-SHA1&Timestamp=2020-02-16T15%3A12%3A53Z&SignatureVersion=1.0&SignatureNonce=*******************&Signature=:

2020-02-16 23:12:54 ERROR 400: Bad Request.

 231254  WARN : 传输失败 - 4/0 在 60 秒后重试

coolsnowwolf, 2020-02-16:
设置问题

authon, 2020-02-16:
[image]

就是这样正常设置的

authon, 2020-02-16:
[image]

设置是这样的

jzh33, 2020-02-17:
你的设置错了，跟插件没关系。
option domain xxx@xxx
你改一下吧

authon, 2020-02-17:
> 你的设置错了，跟插件没关系。
> option domain xxx@xxx
> 你改一下吧

[image]

插件是没问题，我这包还有一个人用，他解析没问题，我就不知道为什么我这个不行。，不知道哪里出错，找你的改了一下，不管是lookup的还是option的，都是wget 报错，bad request

trackycn, 2020-02-20:
参数填错了, 看一下阿里云的API文档吧
[https://help.aliyun.com/document_detail/29778.html?spm=a2c4g.11186623.6.653.6baf5eb4VJVyRA](url)

> 以参数SubDomain为a.www.example.com为示例：
> 
> 如果参数DomainName为空，则查询域名为example.com、主机名为”a.www“的解析记录。
> 
> 如果参数DomainName为 www.example.com，则查询域名为www.example.com、主机名为 "a"的解析记录。
> 
> 如果参数DomainName为 a.www.example.com，则查询域名为a.www.example.com、主机名为 "@"的解析记录。

kuretru, 2020-03-01:
相同问题已解决，其实作者已经提示了

> option domain   - 完整的域名。建议主机与域名之间使用 @符号 分隔，否则将以第一个 .符号 之前的内容作为主机名

我是一个二级域名
xxx.ddns.github.com，主机被错误的解析为`xxx`，而不是`xxx.ddns`，因此改为xxx.ddns@github.com即可

closed 2020-03-05 by coolsnowwolf

hb1707, 2020-07-07:
阿里云那边要给dns的修改权限，不是domain权限

NagaseKouichi, 2022-02-20:
今天发现这个问题和openssl-util的版本有关系，qsdk固件上的openssl-util装的是1.1.1l-1，使用aliddns就会报这个错误8，把openssl-util升级到1.1.1m-1，就可以正常使用了

qwnyx, 2023-01-05:
[update_aliyun_com.sh](https://github.com/coolsnowwolf/lede/discussions/10732)

1WuNeiGui, 2023-07-16:
> 231253 : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/MYDDNS_IPV4.dat -o /var/run/ddns/MYDDNS_IPV4.err --no-proxy [...]

老哥你可以去设置页面看看你的用户名和密码前面是不是有空格，我直接从阿里云复制过去的就出现空格了，然后我把密码前面的空格删掉就好了，不报wget8错误了

chigua2248, 2024-01-04:
> 1WuNeiGui

牛逼,还真就是多了个空格