coolsnowwolf/lede #3159: 最近几个版本编译armv8的固件，xfsprogs出错

liwenjie119, 2020-02-17:
16T19:18:07.9121537Z make[3]: Entering directory '/home/runner/work/lede/lede/package/utils/xfsprogs'
2020-02-16T19:18:08.0164712Z rm -f /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/.built
2020-02-16T19:18:08.0185175Z touch /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/.built_check
2020-02-16T19:18:08.0205129Z make -C /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/libxfs crc32table.h CFLAGS="-O2 -I/home/runner/work/lede/lede/staging_dir/host/include -I/home/runner/work/lede/lede/staging_dir/hostpkg/include -I/home/runner/work/lede/lede/staging_dir/target-aarch64_generic_musl/host/include -I/home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include"
2020-02-16T19:18:08.0259430Z make[4]: Entering directory '/home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/libxfs'
2020-02-16T19:18:08.0261246Z aarch64-openwrt-linux-musl-gcc -MM -O2 -I/home/runner/work/lede/lede/staging_dir/host/include -I/home/runner/work/lede/lede/staging_dir/hostpkg/include -I/home/runner/work/lede/lede/staging_dir/target-aarch64_generic_musl/host/include -I/home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include cache.c crc32.c defer_item.c init.c kmem.c list_sort.c logitem.c radix-tree.c rdwr.c trans.c util.c xfs_ag_resv.c xfs_alloc.c xfs_alloc_btree.c xfs_attr.c xfs_attr_leaf.c xfs_attr_remote.c xfs_bit.c xfs_bmap.c xfs_bmap_btree.c xfs_btree.c xfs_da_btree.c xfs_da_format.c xfs_defer.c xfs_dir2.c xfs_dir2_block.c xfs_dir2_data.c xfs_dir2_leaf.c xfs_dir2_node.c xfs_dir2_sf.c xfs_dquot_buf.c xfs_ialloc.c xfs_inode_buf.c xfs_inode_fork.c xfs_ialloc_btree.c xfs_log_rlimit.c xfs_refcount.c xfs_refcount_btree.c xfs_rmap.c xfs_rmap_btree.c xfs_rtbitmap.c xfs_sb.c xfs_symlink_remote.c xfs_trans_resv.c linux.c | /home/runner/work/lede/lede/staging_dir/host/bin/sed -e 's,^\([^:]*\)\.o,\1.lo,' > .ltdep
2020-02-16T19:18:08.0932060Z In file included from defer_item.c:36:0:
2020-02-16T19:18:08.0932911Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.0933423Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.0933539Z           ^~~~~~~~~~~~~~~~~
2020-02-16T19:18:08.0933647Z compilation terminated.
2020-02-16T19:18:08.1099904Z In file included from init.c:32:0:
2020-02-16T19:18:08.1101179Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.1101678Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.1101976Z           ^~~~~~~~~~~~~~~~~
2020-02-16T19:18:08.1102135Z compilation terminated.
2020-02-16T19:18:08.1494132Z In file included from logitem.c:28:0:
2020-02-16T19:18:08.1494815Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.1494986Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.1495096Z           ^~~~~~~~~~~~~~~~~
2020-02-16T19:18:08.1495218Z compilation terminated.
2020-02-16T19:18:08.1737094Z In file included from rdwr.c:30:0:
2020-02-16T19:18:08.1738026Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.1738460Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.1738771Z           ^~~~~~~~~~~~~~~~~
2020-02-16T19:18:08.1739060Z compilation terminated.
2020-02-16T19:18:08.1871538Z In file included from trans.c:29:0:
2020-02-16T19:18:08.1872417Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.1873078Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.1873380Z           ^~~~~~~~~~~~~~~~~
2020-02-16T19:18:08.1873665Z compilation terminated.
2020-02-16T19:18:08.2006078Z In file included from util.c:31:0:
2020-02-16T19:18:08.2006670Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.2006849Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.2006954Z           ^~~~~~~~~~~~~~~~~
2020-02-16T19:18:08.2007054Z compilation terminated.
2020-02-16T19:18:08.2305865Z In file included from xfs_alloc.c:28:0:
2020-02-16T19:18:08.2306709Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.2307132Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.2307449Z           ^~~~~~~~~~~~~~~~~
2020-02-16T19:18:08.2307732Z compilation terminated.
2020-02-16T19:18:08.2599158Z In file included from xfs_attr.c:30:0:
2020-02-16T19:18:08.2600294Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.2600476Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.2600609Z           ^~~~~~~~~~~~~~~~~
2020-02-16T19:18:08.2600953Z compilation terminated.
2020-02-16T19:18:08.2741422Z In file included from xfs_attr_leaf.c:30:0:
2020-02-16T19:18:08.2742293Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.2742703Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.2742992Z           ^~~~~~~~~~~~~~~~~
2020-02-16T19:18:08.2743328Z compilation terminated.
2020-02-16T19:18:08.2879812Z In file included from xfs_attr_remote.c:30:0:
2020-02-16T19:18:08.2880751Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.2881192Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.2881511Z           ^~~~~~~~~~~~~~~~~
2020-02-16T19:18:08.2881933Z compilation terminated.
2020-02-16T19:18:08.3156023Z In file included from xfs_bmap.c:31:0:
2020-02-16T19:18:08.3157118Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.3157574Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.3157901Z           ^~~~~~~~~~~~~~~~~
2020-02-16T19:18:08.3158215Z compilation terminated.
2020-02-16T19:18:08.3291626Z In file included from xfs_bmap_btree.c:27:0:
2020-02-16T19:18:08.3292471Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.3292897Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.3293198Z           ^~~~~~~~~~~~~~~~~
2020-02-16T19:18:08.3293483Z compilation terminated.
2020-02-16T19:18:08.3428329Z In file included from xfs_btree.c:27:0:
2020-02-16T19:18:08.3428991Z /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/xfsprogs-4.11.0/include/xfs_inode.h:23:10: fatal error: xfs_inode_buf.h: No such file or directory
2020-02-16T19:18:08.3429495Z  #include "xfs_inode_buf.h"
2020-02-16T19:18:08.3429805Z           ^~~~~~~~~~~~~~~~~
[... 163 lines cut ...]
2020-02-16T19:18:10.3755052Z                  from ../include/input.h:24,
2020-02-16T19:18:10.3755154Z                  from copy_file_range.c:23:
2020-02-16T19:18:10.3755814Z /mnt/openwrt/staging_dir/toolchain-aarch64_generic_gcc-7.5.0_musl/include/unistd.h:192:9: note: previous declaration of 'copy_file_range' was here
2020-02-16T19:18:10.3755982Z  ssize_t copy_file_range(int, off_t *, int, off_t *, size_t, unsigned);
2020-02-16T19:18:10.3756109Z          ^~~~~~~~~~~~~~~
2020-02-16T19:18:10.3797367Z make[6]: *** [copy_file_range.o] Error 1
2020-02-16T19:18:10.3798048Z ../include/buildrules:59: recipe for target 'copy_file_range.o' failed
2020-02-16T19:18:10.3803283Z make[5]: *** [io] Error 2
2020-02-16T19:18:10.3803773Z include/buildrules:35: recipe for target 'io' failed
2020-02-16T19:18:10.3804870Z make[4]: *** [default] Error 2
2020-02-16T19:18:10.3805290Z Makefile:74: recipe for target 'default' failed
[... cut ...]
2020-02-16T19:18:10.3844203Z package/Makefile:103: recipe for target '/home/runner/work/lede/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
2020-02-16T19:18:10.3861030Z make[1]: Leaving directory '/home/runner/work/lede/lede'
2020-02-16T19:18:10.3862503Z make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
2020-02-16T19:18:10.3872521Z make: *** [world] Error 2
2020-02-16T19:18:10.3872949Z /home/runner/work/lede/lede/include/toplevel.mk:216: recipe for target 'world' failed
2020-02-16T19:18:10.3985682Z ##[error]Process completed with exit code 2.

AmadeusGhost, 2020-02-17:
https://github.com/AmadeusGhost/lede/commit/57380ffd6068990e1bf99d52cb4f56816c627042
试一下这样改了能正常编译不

unifreq, 2020-02-17:
需要升级到19.07的xfsprogs

coolsnowwolf, 2020-02-18:
fixed

closed 2020-02-18 by coolsnowwolf