coolsnowwolf/lede #3169: 下载错误u-boot-2018.03.tar.bz2

Alice-space, 2020-02-18:
```

+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://mirror2.openwrt.org/sources/u-boot-2018.03.tar.bz2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure ftp://ftp.denx.de/pub/u-boot/u-boot-2018.03.tar.bz2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 11.7M  100 11.7M    0     0  12651      0  0:16:12  0:16:12 --:--:--     0
Warning: Transient problem: timeout Will retry in 1 seconds. 5 retries left.
100 11.7M  100 11.7M    0     0  11537      0  0:17:46  0:17:46 --:--:--     0
Warning: Transient problem: timeout Will retry in 2 seconds. 4 retries left.
 43 11.7M   43 5284k    0     0  11519      0  0:17:48  0:07:49  0:09:59  7720
```
ftp下载结束后没有正常返回继续，而是超时退出。
http链接挂了

Alice-space, 2020-02-18:
```
make[3]: *** [/home/alice/lede/dl/u-boot-2018.03.tar.bz2] Interrupt
time: tools/mkimage/compile#0.02#0.03#2605.06
tools/Makefile:158: recipe for target '/home/alice/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynyyynynnyyyynynyyy' failed
make[1]: *** [/home/alice/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynyyynynnyyyynynyyy] Interrupt
/home/alice/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Interrupt
```

Alice-space, 2020-02-18:
http://sources.buildroot.net/uboot/u-boot-2018.03.tar.bz2
亲测成功，建议替换

AmadeusGhost, 2020-02-18:
第一个下载地址是 https://sources.openwrt.org/u-boot-2018.03.tar.bz2 ，这个是对的

closed 2020-02-18 by coolsnowwolf