coolsnowwolf/lede #3170: 最新这版AR7XX似乎有问题

yishan2019, 2020-02-18:
最新的源码编译了841（品胜003n替代）和703，（AR71XX里修改了config4.14和image里的对应MK文件的内存为16M，以往都是这样做的全都成功编译出来15M左右的刷机包），出现了生成不了有效的刷机文件的问题，无报错通过，但生成的都是那几个无用文件，以前出现这样的情况只要精简插件减小固件即可。但是这次缩减到了极致也不出刷机包，应该不是文件超大的问题了，实验了N多遍不行。找不到问题所在，是不是源码里面的C程序外的源码部分有改动？（应该不是C文件，841的C语言文件我以前改过这次用的原来的无变化）

yishan2019, 2020-02-18:
试验了一下，把luci里内容全部清空，仍然出不来有效刷机包！肯定不是超大问题了！

EnnawYang, 2020-02-18:
我觉得你可以把生成的文件列表列贴出来，编译最后的日志贴出来，你这么说谁知道是什么问题

coolsnowwolf, 2020-02-18:
实测可以编译出来

[image]

rainqk, 2020-02-18:
建议楼主先尝试清理一下。
make clean

closed 2020-02-18 by coolsnowwolf

yishan2019, 2020-02-18:
都重建VPS了，比清除干净多了，但是还是出不来呢，问题是以前都是这样配置的，这次还减了很多插件，又试了一次仍然不出来

yishan2019, 2020-02-18:
建议大雕先生不要用自己的环境，在谷歌上建一个实例然后用它编译，所有需要的东西都当时在VPS上下载，这样避免了用原来的部分源码。测试效果肯定不一样的。我都是在VPS上编译的，当时下载最新的。