coolsnowwolf/lede #3172: 请求大佬帮修改OpenWRT Ffmpeg 4.2.2 Makefile 支持HTTPS协议

feizaibufei, 2020-02-18:
请求大佬帮修改OpenWRT Ffmpeg 4.2.2 Makefile 支持HTTPS协议
我自己修改编译老是出错

feizaibufei, 2020-02-18:
#
# Copyright (C) 2006-2017 OpenWrt.org
# Copyright (C) 2017 Ian Leonard <antonlacon@gmail.com>
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=ffmpeg
PKG_VERSION:=4.0.5
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
PKG_SOURCE_URL:=https://ffmpeg.org/releases/
PKG_HASH:=13f258aa54e181cb3c49e35c12ab842f05e3b1d1588100567b2e53ebb0ef3972
PKG_MAINTAINER:=Ted Hess <thess@kitschensync.net>, \
		Ian Leonard <antonlacon@gmail.com>

PKG_LICENSE:=LGPL-2.1+ GPL-2+ LGPL-3
PKG_LICENSE_FILES:=COPYING.GPLv2 COPYING.GPLv3 COPYING.LGPLv2.1 COPYING.LGPLv3

FFMPEG_CUSTOM_ENCODERS:= \
	ac3 \
	jpegls \
	mpeg1video \
	mpeg2video \
	mpeg4 \
	pcm_s16be \
	pcm_s16le \
	png \
	vorbis \
	zlib \

FFMPEG_CUSTOM_DECODERS:= \
	aac \
	ac3 \
	alac \
	amrnb \
	amrwb \
	ape \
	atrac3 \
	flac \
	gif \
	h264 \
	hevc \
	jpegls \
	mp2 \
	mp3 \
	mpeg1video \
	mpeg2video \
	mpeg4 \
	mpegvideo \
	mpc7 \
	mpc8 \
	pcm_s16be \
	pcm_s16le \
	png \
	vc1 \
	vorbis \
	wavpack \
	wmav1 \
	wmav2 \
	zlib \

FFMPEG_CUSTOM_MUXERS:= \
	ac3 \
	ffm \
	h264 \
[... 593 lines cut ...]
endef

# Soft float is LGPL (no libpostproc); Hard float is GPL (yes libpostproc)
define Package/libffmpeg-full/install
	$(INSTALL_DIR) $(1)/usr/lib
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/lib{avcodec,avdevice,avfilter,avformat,avresample,avutil,swresample,swscale}.so.* $(1)/usr/lib/
ifneq ($(CONFIG_SOFT_FLOAT),y)
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/libpostproc.so.* $(1)/usr/lib/
endif
endef

define Package/libffmpeg-mini/install
	$(INSTALL_DIR) $(1)/usr/lib
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/lib{avcodec,avformat,avutil}.so.* $(1)/usr/lib/
endef

Package/libffmpeg-audio-dec/install = $(Package/libffmpeg-custom/install)

$(eval $(call BuildPackage,ffmpeg))
$(eval $(call BuildPackage,ffprobe))
$(eval $(call BuildPackage,ffserver))
$(eval $(call BuildPackage,libffmpeg-audio-dec))
$(eval $(call BuildPackage,libffmpeg-full))
$(eval $(call BuildPackage,libffmpeg-mini))
$(eval $(call BuildPackage,libffmpeg-custom))

feizaibufei, 2020-02-18:
改PKG_VERSION:=4.0.5
编译通过
改PKG_VERSION:=4.2.2
老是出错

feizaibufei, 2020-02-18:
又出现新的问题
satisfy_dependencies_for: Cannot satisfy the following dependencies for ffmpeg:
 * 	lame-lib
 * opkg_install_cmd: Cannot install package ffmpeg.

feizaibufei, 2020-02-18:
mkdir -p "/home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-install/usr/lib/pkgconfig"
install -m 644 libavutil/libavutil.pc "/home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-install/usr/lib/pkgconfig"
make[4]: Leaving directory '/home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2'
touch /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.built
rm -rf /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg.installed /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg
mkdir -p /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg
install -d -m0755 /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg/usr/bin
cp -fpR /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-install/usr/bin/ffmpeg /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg/usr/bin/
touch /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg.installed
mkdir -p /home/openwrt/lede/staging_dir/target-aarch64_generic_musl/root-armvirt/stamp
SHELL= flock /home/openwrt/lede/tmp/.root-copy.flock -c 'cp -fpR /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/.pkgdir/ffmpeg/. /home/openwrt/lede/staging_dir/target-aarch64_generic_musl/root-armvirt/'
touch /home/openwrt/lede/staging_dir/target-aarch64_generic_musl/root-armvirt/stamp/.ffmpeg_installed
mkdir -p /home/openwrt/lede/bin/targets/armvirt/64/packages /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-aarch64_generic/libffmpeg-full/CONTROL /home/openwrt/lede/staging_dir/target-aarch64_generic_musl/pkginfo
install -d -m0755 /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-aarch64_generic/libffmpeg-full/usr/lib
cp -fpR /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-install/usr/lib/lib{avcodec,avdevice,avfilter,avformat,avresample,avutil,swresample,swscale}.so.* /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-aarch64_generic/libffmpeg-full/usr/lib/
find /home/openwrt/lede/build_dir/target-aarch64_generic_musl/ffmpeg-4.2.2/ipkg-aarch64_generic/libffmpeg-full -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package libffmpeg-full is missing dependencies for the following libraries:
libpostproc.so.55
make[3]: *** [Makefile:808: /home/openwrt/lede/bin/packages/aarch64_generic/packages/libffmpeg-full_4.2.2-1_aarch64_generic.ipk] Error 1
make[3]: Leaving directory '/home/openwrt/lede/feeds/packages/multimedia/ffmpeg'
time: package/feeds/packages/ffmpeg/full/compile#545.18#77.51#620.68
make[2]: *** [package/Makefile:109: package/feeds/packages/ffmpeg/compile] Error 2
make[2]: Leaving directory '/home/openwrt/lede'
make[1]: *** [package/Makefile:103: /home/openwrt/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
make: *** [/home/openwrt/lede/include/toplevel.mk:218：world] 错误 2

coolsnowwolf, 2020-02-19:
发个 PR ?

coolsnowwolf, 2020-02-19:
FFMPEG 的 HTTPS 用在什么场合？

feizaibufei, 2020-02-19:
计划任务下载https协议华人直播节目

closed 2020-03-06 by coolsnowwolf