coolsnowwolf/lede #3174: ADP 怎么编译加入规则订阅？

hxlls, 2020-02-18:
https://github.com/privacy-protection-tools/anti-AD 这个规则看起来不错，并且已经有多种支持，想加入adp使 去广告过滤效果更好点

goukey, 2020-02-18:
装个adguard home添加规则用吧

hxlls, 2020-02-18:
不需要ADG，浪费资源，本来就支持dnsmasq的配置，为啥要装个这么占用资源的东西

coolsnowwolf, 2020-02-18:
这个去广告的订阅目前迁移到 SSR Plus+ 中了

例如你这个需求，在 SSR Plus+ 的去广告订阅中，输入

https://gitee.com/privacy-protection-tools/anti-ad/raw/master/easylist.txt

[image]

然后更新去广告条目：
[image]

然后应用重启 SSR Plus+ 即可马上生效

hxlls, 2020-02-18:
SSRP的广告订阅在国内也可以生效么？

coolsnowwolf, 2020-02-18:
当然，全局生效

hxlls, 2020-02-18:
OK,这太好了

hcym, 2020-02-18:
新玩具，感谢各位帅哥

lltcom, 2020-02-18:
订阅的规则可以设置几条么，默认是1条

kawon7317, 2020-02-18:
太好了

jxph1123, 2020-02-18:
建议可以自己再新加地址，只有一条，能自己多增加就更好了，可以淘汰ADBYBY了。

coolsnowwolf, 2020-02-18:
加多条支持只会被滥用而已，好用的一条即可

d975xbx, 2020-02-18:
https的过滤效果感觉不如浏览器插件

d975xbx, 2020-02-18:
[image]
去不掉，浏览器插件可以

dolphinpaopao, 2020-02-18:
个人使用证明，目前浏览器插件才是最完善的，路由器开启这些功能或多或少都会出现一下莫名其妙的影响。测试过，最不受人待见的D2550，不开过滤广告什么插件，完全可以满足500M宽带+油管15W↑。

china0532, 2020-02-18:
给大家推荐个浏览器去广告插件adblock-plus过滤广告很强，就是不能在路由器上用，只有手机版，和浏览器版

SharerMax, 2020-02-18:
> 这个去广告的订阅目前迁移到 SSR Plus+ 中了
> 
> 例如你这个需求，在 SSR Plus+ 的去广告订阅中，输入
> 
> https://gitee.com/privacy-protection-tools/anti-ad/raw/master/easylist.txt
> 
> [image]
> 
> 然后更新去广告条目：
> [image]
> 
> 然后应用重启 SSR Plus+ 即可马上生效

有点问题，更新的规则生成的ad.conf 在/etc/dnsmasq.ssr 下，dnsmasq 运行时使用的是/tmp/etc/dnsmasq.ssr 下的，更新完可能要重启一下才能生效

jackiezhang29, 2020-02-18:
是不是adblock plus在编译的时候，就不需要添加了？现在用大雕集成广告过滤的ssr+就行？

hxlls, 2020-02-18:
> 是不是adblock plus在编译的时候，就不需要添加了？现在用大雕集成广告过滤的ssr+就行？

按照道理上是可以的，我看了下规则好像比较详尽，并且利用的也是dnsmasq 来做HOST 类型的去广告，并且很大部分规则和ADP应该是有重复的。所以觉得取消ADP应该也没太大问题

jxph1123, 2020-02-19:
SSR Plus+ 去广告还有点问题，会导致有些网站打不开或者打开延时，我又取消了。

closed 2020-02-19 by coolsnowwolf