coolsnowwolf/lede #3178: 小白请教 编译 多拨相关问题

Guxing0321, 2020-02-18:
Hi, 大佬,

本人刚学编译固件，发现编译最后是成功的，但是始终没有 多线多拨 选项
网上有推荐 编译时候勾上  luci-app-syncdial, 尝试过多次， 依然没有看到 

BR
Robin

jh0072110, 2020-02-18:
勾选完保存了吗

Guxing0321, 2020-02-18:
我确定保存了，而且试过多次

我都是web 升级，保存原有配置的，这个有影响吗？

jh0072110, 2020-02-18:
你把mwan3helper也够选上试试（好像是这个名字……）

Guxing0321, 2020-02-18:
这个有的，我应该是选上了，你看下是不是这个

[image]

Guxing0321, 2020-02-18:
我现在在想，web 升级保留原来配置会不会有影响

Guxing0321, 2020-02-18:
我现在在想，web 升级保留原来配置会不会有影响。
我是改了用户名，不用root，之前几次升级过后，需要再去修改一个文件，重启才能在网页登录
最近几次web升级过后都没有问题，所以我也很好奇，在想会不会升级哪里验证没通过，或者看到什么信息一致没有刷进新的固件

jh0072110, 2020-02-18:
额，不知道了…你选的插件没问题，不然你试试重置……哈哈哈哈

Guxing0321, 2020-02-18:
好的，我这边再试试。谢谢哈

mcucore, 2020-02-18:
昨天加过这个，发现编译的固件WEB页面反应很慢，不知是否MWAN引起。

gudumuyu, 2020-02-19:
要勾选mwan3的

Guxing0321, 2020-02-19:
我这个问题，尝试了在breed里面重新刷，是解决了
[image]

但是，我又遇到一个问题：
刷了固件或者web升级之后，第一天基本都会出现一次机器莫名重启，或者无法访问网络
有时候过一会就行了，有时候需要重启下路由器，

这个是挺烦的，刚我这边又出现了一次，这个可以通过哪边去查问题？

closed 2020-02-19 by coolsnowwolf