coolsnowwolf/lede #3179: 编译报错 不知道为什么

chiyu521, 2020-02-18:
go: finding golang.org/x/crypto v0.0.0-20190308221718-c2843e01d9a2
go: finding github.com/gorilla/websocket v1.4.1
build v2ray.com/core/main: cannot load github.com/golang/protobuf/proto: github.com/golang/protobuf@v1.3.2: stream error: stream ID 59; INTERNAL_ERROR

Building targets
build .: cannot find module for path .

Cleaning module download cache (golang/go#27455)

Makefile:307: recipe for target '/home/chiyu/lede/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.built' failed
make[3]: *** [/home/chiyu/lede/build_dir/target-mipsel_24kc_musl/v2ray-core-4.22.1/.built] Error 1
make[3]: Leaving directory '/home/chiyu/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#2.31#1.34#1039.16
package/Makefile:107: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/chiyu/lede'
package/Makefile:103: recipe for target '/home/chiyu/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/chiyu/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/chiyu/lede'
/home/chiyu/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

PandaEgg, 2020-02-18:
我今天也是 明明什么都没添加

GreatGarlic, 2020-02-18:
7:00:59.2497150Z x86 instruction decoder selftest (X86_DECODER_SELFTEST) [N/y/?] (NEW) aborted!
2020-02-18T07:00:59.2497451Z 
2020-02-18T07:00:59.2498271Z Console input/output is redirected. Run 'make oldconfig' to update configuration.
2020-02-18T07:00:59.2498609Z 
2020-02-18T07:00:59.2499428Z scripts/kconfig/Makefile:37: recipe for target 'silentoldconfig' failed
2020-02-18T07:00:59.2500360Z Makefile:562: recipe for target 'silentoldconfig' failed
2020-02-18T07:00:59.2807302Z 
2020-02-18T07:00:59.2817422Z The present kernel configuration has modules disabled.
2020-02-18T07:00:59.2827053Z Type 'make config' and enable loadable module support.
2020-02-18T07:00:59.2834705Z Then build a kernel with module support enabled.
2020-02-18T07:00:59.2843742Z 
2020-02-18T07:00:59.2852333Z make[5]: *** [modules] Error 1
2020-02-18T07:00:59.2852930Z Makefile:1296: recipe for target 'modules' failed
2020-02-18T07:00:59.2853583Z make[5]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.9.212'
2020-02-18T07:00:59.2854700Z make[4]: *** [/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.9.212/.modules] Error 2
2020-02-18T07:00:59.2855125Z Makefile:29: recipe for target '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.9.212/.modules' failed
2020-02-18T07:00:59.2858147Z make[3]: *** [compile] Error 2
2020-02-18T07:00:59.2858581Z make[4]: Leaving directory '/home/runner/work/lede/lede/target/linux/x86'
2020-02-18T07:00:59.2859311Z time: target/linux/compile#18.09#6.49#23.74
2020-02-18T07:00:59.2859836Z Makefile:13: recipe for target 'compile' failed
2020-02-18T07:00:59.2860605Z make[3]: Leaving directory '/home/runner/work/lede/lede/target/linux'
2020-02-18T07:00:59.2883366Z make[2]: *** [target/linux/compile] Error 2
2020-02-18T07:00:59.2883878Z target/Makefile:23: recipe for target 'target/linux/compile' failed
2020-02-18T07:00:59.2884474Z make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
2020-02-18T07:00:59.2917832Z make[2]: Leaving directory '/home/runner/work/lede/lede'
2020-02-18T07:00:59.2918637Z make: *** [world] Error 2
2020-02-18T07:00:59.2923775Z target/Makefile:18: recipe for target '/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.target_compile' failed
2020-02-18T07:00:59.2926066Z make[1]: Leaving directory '/home/runner/work/lede/lede'
2020-02-18T07:00:59.2926839Z /home/runner/work/lede/lede/include/toplevel.mk:216: recipe for target 'world' failed
2020-02-18T07:00:59.2929582Z ##[error]Process completed with exit code 2.

1026831402, 2020-02-18:
用ubuntu 16.04 lts编译试试？ 编译前先make download下载dl库在make V=99 ,建议不要加-j参数多线程编译，容易出错

dovegq, 2020-02-19:
> go: finding golang.org/x/crypto v0.0.0-20190308221718-c2843e01d9a2
> go: finding github.com/gorilla/websocket v1.4.1
> build v2ray.com/core/main: cannot load github.com/golang/protobuf/proto: [...]
go语言的代理加一个就好了
export GO111MODULE=on
export GOPROXY=https://athens.azurefd.net

dovegq, 2020-02-19:
> go: finding golang.org/x/crypto v0.0.0-20190308221718-c2843e01d9a2
> go: finding github.com/gorilla/websocket v1.4.1
> build v2ray.com/core/main: cannot load github.com/golang/protobuf/proto: [...]

go语言的服务器不稳，换个镜像就好了。
export GO111MODULE=on
export GOPROXY=https://athens.azurefd.net

closed 2020-02-19 by coolsnowwolf