coolsnowwolf/lede #3186: 从今天起再发垃圾 PR 的一律永久拉黑！有完没完的？！

coolsnowwolf, 2020-02-19:
[image]

82kg, 2020-02-19:
估计是小白不会操作

snakwu, 2020-02-19:
其实是不是只给一些维护的人开RP权限比较好?要提交RP的申请权限,这样好管理?

hcym, 2020-02-19:
基本码盲，不会吧，还是关闭非维护者的操作简单

dolphinpaopao, 2020-02-19:
其实这个和是否小白码盲没有多大关系，先搜索学习做好准备再操作是一个基本的流程，伸手党习惯了不思考不学习懒得多看一眼，只管索取。让他们付出一点学习成本是应该的。

dolphinpaopao, 2020-02-19:
> 估计是小白不会操作

前阵还看到一位一天PR好多次的，但凡有点思考学习能力的正常人士不会这样子捣乱来吧。

lloyd18, 2020-02-19:
我作为一个小白 吓得瑟瑟发抖 都不敢说话了，结果我还是说了

whlsxl, 2020-02-19:
> > 估计是小白不会操作
> 
> 前阵还看到一位一天PR好多次的，但凡有点思考学习能力的正常人士不会这样子捣乱来吧。

小白进化老白也要时间的么，能提PR已经比伸手党好太多了。

dolphinpaopao, 2020-02-19:
> > > 估计是小白不会操作
> > 
> > 
> > 前阵还看到一位一天PR好多次的，但凡有点思考学习能力的正常人士不会这样子捣乱来吧。
> 
> 小白进化老白也要时间的么，能提PR已经比伸手党好太多了。

我并没有说学习不需要时间，但是我坚持先搜索做好准备再开始操作，这样就算是做错了也能自己及时纠正，而不是不停的在原项目上面乱来一通，让作者花费很多不必要的精力来管理。本来这个就有一定的门槛，大家的时间都很值价，自己的项目上就随意，没人去说什么。

whlsxl, 2020-02-19:
> > > > 估计是小白不会操作
> > > 
> > > 
> > > 前阵还看到一位一天PR好多次的，但凡有点思考学习能力的正常人士不会这样子捣乱来吧。
> > 
> > 
> > 小白进化老白也要时间的么，能提PR已经比伸手党好太多了。
> 
> 我并没有说学习不需要时间，但是我坚持先搜索做好准备再开始操作，这样就算是做错了也能自己及时纠正，而不是不停的在原项目上面乱来一通，让作者花费很多不必要的精力来管理。本来这个就有一定的门槛，大家的时间都很值价，自己的项目上就随意，没人去说什么。

我怀疑他们还不知道PR的含义吧。。

closed 2020-02-21 by coolsnowwolf