coolsnowwolf/lede #3189: 最近这几天的版本保留配置更新后SSR 不自动运行

tone88, 2020-02-19:
基本L大每次更新都会编译升级，最近这几天的版本保留配置更新后发现SSR 不自动运行，手动重启一次后正常。   

感谢L大的无私贡献！

lqhorochi, 2020-02-19:
一样，同一问题

coolsnowwolf, 2020-02-19:
那就不保留

lqhorochi, 2020-02-19:
> 那就不保留

梁静茹给你的勇气！

xiaozizi58, 2020-02-19:
那你不保留 或者重新启动

dolphinpaopao, 2020-02-19:
版本更新从来不保留配置升级，因为不管是硬路由还是软路由，保留配置升级偶尔都会带来莫名其妙的问题，升级完成后来一次恢复出厂操作重新配置。虽然这样会繁琐一点。但是路由器的本职工作不就是稳定使用吗？折腾那就要做好心理准备。

coolsnowwolf, 2020-02-19:
我测试没有重现，重启后可以运行

dolphinpaopao, 2020-02-19:
> > 那就不保留
> 
> 梁静茹给你的勇气！

这个建议有什么问题吗？人家只是提出一个可行的方案，采不采纳随你？什么态度，欠你什么了？

tone88, 2020-02-19:
> 版本更新从来不保留配置升级，因为不管是硬路由还是软路由，保留配置升级偶尔都会带来莫名其妙的问题，升级完成后来一次恢复出厂操作重新配置。虽然这样会繁琐一点。但是路由器的本职工作不就是稳定使用吗？折腾那就要做好心理准备。

以前也是每次更新都是新建虚拟机全部重新配置，最近懒了而已。

dolphinpaopao, 2020-02-19:
> > 版本更新从来不保留配置升级，因为不管是硬路由还是软路由，保留配置升级偶尔都会带来莫名其妙的问题，升级完成后来一次恢复出厂操作重新配置。虽然这样会繁琐一点。但是路由器的本职工作不就是稳定使用吗？折腾那就要做好心理准备。
> 
> 以前也是每次更新都是新建虚拟机全部重新配置，最近懒了而已。

新版本可能会有功能设置的改善，保留配置的话会出现不兼容或者设置错误从而带来不便。

tone88, 2020-02-19:
> 我测试没有重现，重启后可以运行

我是需要手动重启一次才正常，也许是我最近一直保留配置升级的过，晚点重刷一次，下次更新的时候再观察一下。  感谢l大测试

DragonOneY, 2020-02-19:
> > 那就不保留
> 
> 梁静茹给你的勇气！

你对软件作者就是这么说话的吗？

hxlls, 2020-02-19:
> > 那就不保留
> 
> 梁静茹给你的勇气！

这话对于你更适用，你没任何贡献，一白嫖，你可以选择不用，梁静茹给你的勇气？来这里喷？

closed 2020-02-20 by coolsnowwolf