coolsnowwolf/lede #3197: SSRplus+的广告过滤数据库

sky137818221, 2020-02-19:
点更新数据库，就只显示正在更新请稍后 过了两个小时 都没反应 昨晚点了之后今天早上起床看了一下
5900+的数据 看了一下这个列表 有3W+条规则 问了一下群友 也是说有3W左右 今天又点了下更新等了俩小时 还是没更新动 底下的【国内IP段】数据库也是这个情况 GFW列表可以正常更新 不知道哪里的问题 看了下ssr的日志和系统日志 没有相关的条目记录...

hxlls, 2020-02-19:
国内IP段请全局后在更新

coolsnowwolf, 2020-02-19:
这个地址有两个订阅源，都试试

[image]

coolsnowwolf, 2020-02-19:
https://github.com/privacy-protection-tools/anti-AD

找兼容 ADG Home 的规则库即可

SSR Plus+ 这个是用的 ADG Home 的兼容格式

sky137818221, 2020-02-19:
非常感谢 可能是之前保留设置更新的原因 改为
https://github.com/privacy-protection-tools/anti-AD里面的规则之后就更新成功了
我之前的广告屏蔽列表是
https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt
看了一下好像是有8W+条目 但是只能更新5.9K
https://easylist-downloads.adblockplus.org/easylist.txt
6W行 只出了1K的规则
https://easylist-downloads.adblockplus.org/easylistchina.txt
这个也试了一下 1.7W行 更新出了 4.6K
这两个也是easylist的规则 不知道为什么无法使用 

感谢L大 我是一个小白 但是愿意一步一步按照你说的
不是只做一个伸手党 以后遇到问题 先自己测试找问题 尽自己所能
不只是把问题甩给你们 看到你之前感慨的那个issue今天关掉了
也只能在这里跟你说一下了 其实我之前也不小心按过很多次 pr
我以为那个就是git pull更新... 后来查了一下其实并不是 在此向你道歉..

closed 2020-02-19 by sky137818221