coolsnowwolf/lede #3198: 选择不同的硬件编译出来的UnblockNeteaseMusic功能不一样？

DongfeiSay, 2020-02-19:
如图片里 https://www.lanzous.com/i9hm7oj 
x84_64的是在路由的前一天编译的，路由的源码应比x86_64的新，看UnblockNeteaseMusic的主程序版本都是0.23.1，应该是一样的才对，可是看着64的UnblockNeteaseMusic功能更多？而且路由UnblockNeteaseMusic是用不了的。。64的UnblockNeteaseMusic能正常解锁播放灰色歌曲
因此表示疑惑

DongfeiSay, 2020-02-19:
路由的是mt7620a

zwillhill, 2020-02-19:
猜测是路由cpu没有FPU，路由编译的时候选Go版本试试

trackycn, 2020-02-19:
看了你截图, 用最新的代码编译出来就是路由器那张图的界面, 新版本不需要设置音源地址了。手头没有7620, 应该不支持FPU, 用GO版本吧

DongfeiSay, 2020-02-20:
> 看了你截图, 用最新的代码编译出来就是路由器那张图的界面, 新版本不需要设置音源地址了。手头没有7620, 应该不支持FPU, 用GO版本吧

2/20号，编译了L大最新源码，64的UnblockNeteaseMusic确实是和路由的一样没有音源地址，同时。。。也用不了了

DongfeiSay, 2020-02-20:
> 看了你截图, 用最新的代码编译出来就是路由器那张图的界面, 新版本不需要设置音源地址了。手头没有7620, 应该不支持FPU, 用GO版本吧

20/20号 编译了L大最新源码的64确实和路由的一样不需要设置音源地址了，也正常使用，也顺便装了go版的，不过go版的好像不能用，还是灰色的播放不了

DongfeiSay, 2020-02-20:
> > 看了你截图, 用最新的代码编译出来就是路由器那张图的界面, 新版本不需要设置音源地址了。手头没有7620, 应该不支持FPU, 用GO版本吧
> 
> 20/20号 编译了L大最新源码的64确实和路由的一样不需要设置音源地址了，也正常使用，也顺便装了go版的，不过go版的好像不能用，还是灰色的播放不了

go版网易云pc客户端手动设置代理可以用了，网页版用不了，原版不需要设置网页和客户端都可以用

DongfeiSay, 2020-02-20:
x86_64 2/20负载均衡保存配置一直在应用更改 /etc/config/mwan3 无法应用，返回去2/17的是能正常保存并应用的o(￣┰￣*)ゞ 都是全新环境下编译的

closed 2020-02-20 by coolsnowwolf

DongfeiSay, 2020-02-20:
> 猜测是路由cpu没有FPU，路由编译的时候选Go版本试试

经实验，尽管不知道路由cpu有没有FPU，不过我这里mt7620a是成功使用原版UnblockNeteaseMusic解锁网易云灰色歌曲了！版本为git-20.050.46852-a2b0e0c（2/20 17时源码）