coolsnowwolf/lede #3205: 致所有使用 KFERMercer/OpenWrt-CI 的人:

KFERMercer, 2020-02-19:
KFERMercer/OpenWrt-CI 是给正在维护自己的openwrt的库, 并且知道自己在做什么的人用来测试的测试环境, **而不是让只想得到固件文件的普通用户用来编译的工具**. 

### 建议所有自认为是后者的人停止使用 KFERMercer/OpenWrt-CI, 转而去使用 **[P3TERX/Actions-OpenWrt](https://github.com/P3TERX/Actions-OpenWrt)**. 

@P3TERX 创建的 workflow 不需要 fork 整个 OpenWrt 存储库, 只需要上传`.config`文件即可完成编译, 而且支持使用 SSH 连接 GitHub Actions 服务器, 更加直观的控制整个编译过程.

同时请所有尚不明白 Pull request 为何物的人, 在学习过网上随处可见的[Git教程](https://www.liaoxuefeng.com/wiki/896043488029600)后, 停止向 coolsnowwolf/lede, 以及 openwrt/openwrt 推送任何垃圾合并请求.

lqhorochi, 2020-02-19:
不用的路过
用自己电脑编译.

adaaaaaa, 2020-02-20:
我两者都有怎么办？？难道有人任意一次编译了之后都不下载固件的吗？🤪🤪🤪

9l, 2020-02-20:
> 我两者都有怎么办？？难道有人任意一次编译了之后都不下载固件的吗？🤪🤪🤪

不是不让下载固件，是不要瞎鸡脖的pr辣鸡配置文件给别人，你是拉新代码，不是推送辣鸡代码

segadrive, 2020-02-20:
我pork了不推送只修改不影响把？小白真不懂

z826540272, 2020-02-20:
我早就用P3TERX/Actions-OpenWrt了,我觉得你的脚本很难用

z826540272, 2020-02-20:
垃圾合并请求主要是因为github的辣鸡设计.

z826540272, 2020-02-20:
> > 我两者都有怎么办？？难道有人任意一次编译了之后都不下载固件的吗？🤪🤪🤪
> 
> 不是不让下载固件，是不要瞎鸡脖的pr辣鸡配置文件给别人，你是拉新代码，不是推送辣鸡代码

github的界面设计拉去代码比推送代码复杂

garypang13, 2020-02-20:
> 我早就用P3TERX/Actions-OpenWrt了,我觉得你的脚本很难用

同感

diy.sh  default-settings files大法以及文件覆盖等都可以用来修改 适合高阶玩家,哪需要拉取整个项目,在OpenWrt-CI中反而把diy.sh files ssh这些高阶玩法都去掉了.如果不是大规模修改那用p3terx的就完全够用了

把.config写在.yml中更是画蛇添足
.yml空格换行回车格式要求严格 .config放到.yml中还得按照cat >> .config <<EOF之类的语法
 增加了操作难度和出错可能性,反而不方便维护了,误导小白呀

Zxilly, 2020-02-21:
github这个fork之后，如果在网页端操作合并，很容易把merge发成PR。。。

ch1993128, 2020-02-21:
感谢，写得很详细，it matters

flin-X, 2020-02-22:
作为只要固件的小白，P3TERX/Actions-OpenWrt在固件的配置上确实要方便一些，openwrt软件包这么多，要配置出自己想要的，.config文件自己去写难度太大。
不得不说Microsoft是真的财大气粗，最近三个月同步过大佬代码的，不做修改，每天都会自动编译。

Clackz, 2020-02-22:
感谢大佬的建议，也感谢P3TERX大佬写的脚本，不用自己改配置好多了……

closed 2020-03-06 by coolsnowwolf