coolsnowwolf/lede #3210: 给项目开一个wiki是否具有可行性

Yi-Hunter, 2020-02-20:
- 之前有人提到过开一个 openwrt 的开发 wiki #2650 , 个人也感觉用处不大
- 这里我所谓的 wiki 是针对本 repo 的, 主要内容如下
  1. 项目背景及当前状态(如设备支持列表, 已知 Bug 等)
  2. 大版本发布及 Change Log
  3. 专有模块的使用及相关说明(如ssr+ 在 gfw 模式, 域名黑白模的优先级等)
  4. 正在开发或者修复的功能(如项目的 Roadmap)
- 这样的 wiki 可以带来的益处
  1. 新人或者想对项目有深入了解的人可以有迹可循
  2. 对 Bug 或功能的需求不用每次提 issue
  3. 有了使用说明可以减少关于某功能如何使用类的 issue

82kg, 2020-02-20:
蛮好的

Egolas, 2020-02-20:
挺好的

luckyyyyy, 2020-02-20:
支持

kl9800, 2020-02-20:
支持

hyird, 2020-02-21:
谁维护了 ，日更雕，忙的过来么

luodaoyi, 2020-02-21:
支持  有空了来写两句

Yi-Hunter, 2020-02-22:
> 谁维护了 ，日更雕，忙的过来么

我个人觉得不一定要大雕主导开展wiki, 这个项目目前除了大雕也有一些相对核心的开发者了, 建议可以由本项目相对核心的开发者开一个wiki repo, 然后链接添加到本repo的readme;
另外wiki的维护相对于开发专业技能及门槛要低很多, 这样也能分流一部分人群, 即相对小白或者只想使用的人可以更多关心wiki项目, 比较像一个使用者社群; 关注开发的人更多关心本repo

jackberry, 2020-02-22:
支持一下

madcat0513, 2020-02-23:
支持，不是核心开发者，甚至小白也有用武之地了。

liwenjie119, 2020-02-23:
支持这个想法

lizhenglin2002, 2020-02-25:
支持啦

yunmi323, 2020-02-26:
可以有，很多问题都本来可以很简单解决的，就是没有相关内容导致小白各种解决不了，wiki维护也是一个问题

closed 2020-03-06 by coolsnowwolf