coolsnowwolf/lede #3211: 关于网件R6800无线问题

Egolas, 2020-02-20:
似乎设置wifi密码，点击保存并应用第一次会应用，再点一次保存并应用会导致wifi速率不正常
[image]
[image]

Egolas, 2020-02-20:
测试了一下只改SSID不会有问题

Egolas, 2020-02-20:
系统日志：
```

Thu Feb 20 12:33:00 2020 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Thu Feb 20 12:33:00 2020 daemon.notice netifd: Interface 'wan_6' is enabled
Thu Feb 20 12:33:00 2020 daemon.notice netifd: Interface 'wan_6' has link connectivity
Thu Feb 20 12:33:00 2020 daemon.notice netifd: Interface 'wan_6' is setting up now
Thu Feb 20 12:33:03 2020 daemon.notice netifd: Interface 'wan_6' is now up
Thu Feb 20 12:33:04 2020 daemon.warn odhcpd[2594]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Thu Feb 20 12:33:04 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Thu Feb 20 12:33:26 2020 kern.info kernel: [  219.464900] device wlan1 left promiscuous mode
Thu Feb 20 12:33:26 2020 kern.info kernel: [  219.474141] br-lan: port 2(wlan1) entered disabled state
Thu Feb 20 12:33:26 2020 daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
Thu Feb 20 12:33:26 2020 daemon.notice hostapd: wlan1: AP-DISABLED
Thu Feb 20 12:33:26 2020 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Thu Feb 20 12:33:26 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Thu Feb 20 12:33:26 2020 daemon.notice hostapd: nl80211: Failed to remove interface wlan1 from bridge br-lan: Invalid argument
Thu Feb 20 12:33:27 2020 user.notice mac80211: Failed command: iw phy phy1 set antenna 0xffffffff 0xffffffff
Thu Feb 20 12:33:27 2020 daemon.notice netifd: radio1 (12581): command failed: Not supported (-122)
Thu Feb 20 12:33:27 2020 user.notice mac80211: Failed command: iw phy phy1 set distance 0
Thu Feb 20 12:33:28 2020 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Thu Feb 20 12:33:28 2020 kern.info kernel: [  221.456714] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Thu Feb 20 12:33:28 2020 kern.info kernel: [  221.483641] br-lan: port 2(wlan1) entered blocking state
Thu Feb 20 12:33:28 2020 kern.info kernel: [  221.494493] br-lan: port 2(wlan1) entered disabled state
Thu Feb 20 12:33:28 2020 kern.info kernel: [  221.505951] device wlan1 entered promiscuous mode
Thu Feb 20 12:33:28 2020 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Thu Feb 20 12:33:28 2020 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
Thu Feb 20 12:33:29 2020 daemon.err hostapd: Using interface wlan1 with hwaddr b0:39:56:59:ba:b8 and ssid "Karl-5G"
Thu Feb 20 12:33:30 2020 kern.info kernel: [  223.754630] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Thu Feb 20 12:33:30 2020 kern.info kernel: [  223.767806] br-lan: port 2(wlan1) entered blocking state
Thu Feb 20 12:33:30 2020 kern.info kernel: [  223.778466] br-lan: port 2(wlan1) entered forwarding state
Thu Feb 20 12:33:30 2020 daemon.notice hostapd: wlan1: interface state HT_SCAN->ENABLED
Thu Feb 20 12:33:30 2020 daemon.notice hostapd: wlan1: AP-ENABLED
Thu Feb 20 12:33:30 2020 daemon.notice netifd: Network device 'wlan1' link is up
Thu Feb 20 12:33:46 2020 daemon.info hostapd: wlan1: STA f0:42:1c:91:96:5a IEEE 802.11: authenticated
Thu Feb 20 12:33:46 2020 daemon.info hostapd: wlan1: STA f0:42:1c:91:96:5a IEEE 802.11: associated (aid 1)
Thu Feb 20 12:33:46 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED f0:42:1c:91:96:5a
Thu Feb 20 12:33:46 2020 daemon.info hostapd: wlan1: STA f0:42:1c:91:96:5a RADIUS: starting accounting session 5488E8FD54969544
Thu Feb 20 12:33:46 2020 daemon.info hostapd: wlan1: STA f0:42:1c:91:96:5a WPA: pairwise key handshake completed (RSN)
Thu Feb 20 12:34:27 2020 kern.info kernel: [  280.844886] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
Thu Feb 20 12:34:40 2020 kern.info kernel: [  293.507886] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
Thu Feb 20 12:39:40 2020 daemon.info hostapd: wlan1: STA 48:01:c5:5b:ed:22 IEEE 802.11: authenticated
Thu Feb 20 12:39:40 2020 daemon.info hostapd: wlan1: STA 48:01:c5:5b:ed:22 IEEE 802.11: associated (aid 2)
Thu Feb 20 12:39:40 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 48:01:c5:5b:ed:22
Thu Feb 20 12:39:40 2020 daemon.info hostapd: wlan1: STA 48:01:c5:5b:ed:22 RADIUS: starting accounting session 10B457FBE6D300B7
Thu Feb 20 12:39:40 2020 daemon.info hostapd: wlan1: STA 48:01:c5:5b:ed:22 WPA: pairwise key handshake completed (RSN)
Thu Feb 20 12:52:31 2020 daemon.info hostapd: wlan1: STA 48:01:c5:5b:ed:22 IEEE 802.11: disconnected due to excessive missing ACKs
Thu Feb 20 12:52:31 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 48:01:c5:5b:ed:22
Thu Feb 20 12:52:54 2020 daemon.info hostapd: wlan1: STA 48:01:c5:5b:ed:22 IEEE 802.11: authenticated
Thu Feb 20 12:52:54 2020 daemon.info hostapd: wlan1: STA 48:01:c5:5b:ed:22 IEEE 802.11: associated (aid 2)
Thu Feb 20 12:52:54 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 48:01:c5:5b:ed:22
Thu Feb 20 12:52:54 2020 daemon.info hostapd: wlan1: STA 48:01:c5:5b:ed:22 RADIUS: starting accounting session 10B457FBE6D300B7
Thu Feb 20 12:52:54 2020 daemon.info hostapd: wlan1: STA 48:01:c5:5b:ed:22 WPA: pairwise key handshake completed (RSN)
```

内核日志：
```
[    0.000000] Linux version 4.14.169 (test@test-virtual-machine) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r1975-02df852)) #0 SMP Sat Feb 15 06:56:15 2020
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Netgear R6800
[    0.000000] Determined physical RAM map:
[... cut ...]
[  221.483641] br-lan: port 2(wlan1) entered blocking state
[  221.494493] br-lan: port 2(wlan1) entered disabled state
[  221.505951] device wlan1 entered promiscuous mode
[  223.754630] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  223.767806] br-lan: port 2(wlan1) entered blocking state
[  223.778466] br-lan: port 2(wlan1) entered forwarding state
[  280.844886] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[  293.507886] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
```

coolsnowwolf, 2020-02-20:
MTK 目前开源驱动就这样

knightridersimon, 2020-02-21:
MT76的2.4G正常吗？发现有点惨不忍睹。在XWRT上编译，2.4G还可以接受，5G不行。现在还是转闭源。有空可以交流。
另外发现4.42比5.02好，有这样的感觉吗？

@coolsnowwolf 你的闭源mtk_wifi make file报错，麻烦看下。
[image]

knightridersimon, 2020-02-21:
> 
> 
> [image]
> MT76的2.4G正常吗？发现有点惨不忍睹。在XWRT上编译，2.4G还可以接受，5G不行。现在还是转闭源。有空可以交流。
> 另外发现4.42比5.02好，有这样的感觉吗？
> 
> @coolsnowwolf 你的闭源mtk_wifi make file报错，麻烦看下。
> WARNING: Makefile 'package/lean/mt/luci-app-mtwifi/Makefile' has a dependency on 'kmod-mt_wifi', which does not exist。
> 
> [image]

@coolsnowwolf 解决了，和我的4.42冲突，502没问题，不用处理了。谢谢！

Egolas, 2020-02-21:
@knightridersimon 您的意思是现在源码里并入了闭源驱动吗？
那我编译的时候把原本的开源驱动取消勾选然后把mt-wifi编译进去就可以了是吗

knightridersimon, 2020-02-21:
> 
> 
> @knightridersimon 您的意思是现在源码里并入了闭源驱动吗？
> 那我编译的时候把原本的开源驱动取消勾选然后把mt-wifi编译进去就可以了是吗

把开源MT76里面的7615去掉就可以了，但是云编译还会自动加进去，要把\target\linux\ramips\image下面的mt7651.mk 的 +=7615e去掉。
define Device/netgear_r6800
  $(Device/netgear_sercomm_nand)
  DEVICE_MODEL := R6800
  SERCOMM_HWNAME := R6950
  SERCOMM_HWID := BZV
  SERCOMM_HWVER := A001
  SERCOMM_SWVER := 0x0062
  IMAGE_SIZE := 40960k
#  DEVICE_PACKAGES += kmod-mt7615e kmod-pinctrl-sx150x
   DEVICE_PACKAGES += kmod-pinctrl-sx150x**
这样就不自动编译mt76了。
另外CONFIG_P....kmod-mt_wifi=y 加上就行了。
还有一些profile配置，make menuconfig里配吧。
目前遇到问题是luci的网卡名字和系统名对不上，以前xwrt等都是这样的。
luci的源码对应6个网卡，实际是2个，名字就错位了，看来还得修改它。
另外用脚本关联的官方的luci （config/wireless->uci2dat）的办法也不完美。
继续搞吧。。。
（公司网上传图看不见，看看这次吧）
[image]

knightridersimon, 2020-02-21:
> 
> 
> @knightridersimon 您的意思是现在源码里并入了闭源驱动吗？
> 那我编译的时候把原本的开源驱动取消勾选然后把mt-wifi编译进去就可以了是吗

这套闭源代码是没改过的，改配置文件，目前内核会崩溃，稍后我用自己调好的试试。

Egolas, 2020-02-22:
@knightridersimon 

> 另外CONFIG_P....kmod-mt_wifi=y 加上就行了。
请问这个config_p....是指的什么地方？

Egolas, 2020-02-22:
> @knightridersimon
> 
> > 另外CONFIG_P....kmod-mt_wifi=y 加上就行了。
> 
> 请问这个config_p....是指的什么地方？

明白了，这是workflow里makemenu的配置

Egolas, 2020-02-22:
> > @knightridersimon 您的意思是现在源码里并入了闭源驱动吗？
> > 那我编译的时候把原本的开源驱动取消勾选然后把mt-wifi编译进去就可以了是吗
> 
> 这套闭源代码是没改过的，改配置文件，目前内核会崩溃，稍后我用自己调好的试试。

好像编译完设置好无线设置找不到信号，我下了你repo里编译好的固件好像也是这样，是我设置得有问题吗，你是怎么设置的？

knightridersimon, 2020-02-22:
> 
> 
> > > @knightridersimon 您的意思是现在源码里并入了闭源驱动吗？
> > > 那我编译的时候把原本的开源驱动取消勾选然后把mt-wifi编译进去就可以了是吗
> > 
> > 
> > 这套闭源代码是没改过的，改配置文件，目前内核会崩溃，稍后我用自己调好的试试。
> 
> 好像编译完设置好无线设置找不到信号，我下了你repo里编译好的固件好像也是这样，是我设置得有问题吗，你是怎么设置的？

找不到信号是因为找不到配置文件不能开启wifi，makefile里文件位置和源码不对应，你可以去试试我的代码。
目前问题还很多。
[image]
[image]

jackcolentern, 2020-02-23:
Linksys 的 EA7500 和 8100 系列也是类似的问题， 同 MT7621 + 双 MT7615

5G 下载速度还行，上传速度是闭源驱动下的一半， 隔一堵墙就掉线了。 目前开源驱动就这样了，只能等大佬慢慢修

目前还是继续用闭源驱动吧， 可是闭源驱动好像只能在旧版内核上用，要不然会kernel panic

可以参考我repo里的 7500v2 的openwrt

（另外，我们撞头像了嘿嘿）

Egolas, 2020-02-25:
> Linksys 的 EA7500 和 8100 系列也是类似的问题， 同 MT7621 + 双 MT7615
> 
> 5G 下载速度还行，上传速度是闭源驱动下的一半， 隔一堵墙就掉线了。 目前开源驱动就这样了，只能等大佬慢慢修
> 
> 目前还是继续用闭源驱动吧， 可是闭源驱动好像只能在旧版内核上用，要不然会kernel panic
> 
> 可以参考我repo里的 7500v2 的openwrt
> 
> （另外，我们撞头像了嘿嘿）

hhhhhh，我这倒是下载比上传慢很多

closed 2020-03-06 by coolsnowwolf