coolsnowwolf/lede #3215: 2/20 接口填写了跃点数后负载均衡无法保存应用

DongfeiSay, 2020-02-20:
https://www.lanzous.com/i9ifckj
x86_64
2/17号的源码编译后能正常保存应用

DongfeiSay, 2020-02-20:
以为是机器问题，用云编译了最新源码的也还是不行
虚拟机文件下载：https://www.lanzous.com/i9iuzmj

coolsnowwolf, 2020-02-20:
直接给日志

DongfeiSay, 2020-02-20:
> 直接给日志

Thu Feb 20 20:30:29 2020 kern.debug kernel: [    0.008574] Calgary: detecting Calgary via BIOS EBDA area
Thu Feb 20 20:30:29 2020 kern.debug kernel: [    0.008575] Calgary: Unable to locate Rio Grande table in EBDA - bailing!
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.011804] Memory: 2041852K/2096696K available (10248K kernel code, 563K rwdata, 2044K rodata, 1084K init, 2368K bss, 54844K reserved, 0K cma-reserved)
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.011830] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.011831] Kernel/User page tables isolation: enabled
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.011876] rcu: Hierarchical RCU implementation.
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.011876] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.011877] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=2.
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.011878] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.011888] NR_IRQS: 4352, nr_irqs: 440, preallocated irqs: 16
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.018168] Console: colour VGA+ 80x25
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.038196] console [tty0] enabled
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.072167] console [ttyS0] enabled
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.072444] ACPI: Core revision 20180810
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.072823] APIC: Switch to symmetric I/O mode setup
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.073346] x2apic enabled
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.073759] Switched APIC routing to physical x2apic.
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.074973] ..TIMER: vector=0x30 apic1=0 pin1=2 apic2=-1 pin2=-1
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.077269] clocksource: tsc-early: mask: 0xffffffffffffffff max_cycles: 0x2879c7f065b, max_idle_ns: 440795303690 ns
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.077935] Calibrating delay loop (skipped) preset value.. 5616.00 BogoMIPS (lpj=28080020)
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.078512] pid_max: default: 32768 minimum: 301
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.078849] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.079268] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.079885] Last level iTLB entries: 4KB 128, 2MB 8, 4MB 8
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.080248] Last level dTLB entries: 4KB 64, 2MB 0, 4MB 0, 1GB 4
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.080644] Spectre V1 : Mitigation: usercopy/swapgs barriers and __user pointer sanitization
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.081218] Spectre V2 : Mitigation: Full generic retpoline
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.081584] Spectre V2 : Spectre v2 / SpectreRSB mitigation: Filling RSB on context switch
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.082144] Speculative Store Bypass: Vulnerable
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.082467] MDS: Vulnerable: Clear CPU buffers attempted, no microcode
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.082965] Freeing SMP alternatives memory: 28K
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.209263] smpboot: CPU0: Intel(R) Core(TM) i5-8400 CPU @ 2.80GHz (family: 0x6, model: 0x9e, stepping: 0xa)
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.209938] Performance Events: unsupported p6 CPU model 158 no PMU driver, software events only.
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.210507] rcu: Hierarchical SRCU implementation.
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.210897] smp: Bringing up secondary CPUs ...
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.211234] x86: Booting SMP configuration:
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.211518] .... node  #0, CPUs:      #1
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.075807] kvm-clock: cpu 1, msr 1fca041, secondary cpu clock
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.214062] smp: Brought up 1 node, 2 CPUs
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.214702] smpboot: Max logical packages: 1
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.214987] smpboot: Total of 2 processors activated (11232.00 BogoMIPS)
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.215543] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.216168] futex hash table entries: 512 (order: 3, 32768 bytes)
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.216554] pinctrl core: initialized pinctrl subsystem
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.216997] NET: Registered protocol family 16
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.217479] cpuidle: using governor ladder
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.217804] ACPI: bus type PCI registered
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.217934] acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.217934] PCI: Using configuration type 1 for base access
Thu Feb 20 20:30:29 2020 kern.info kernel: [    0.219512] cryptd: max_cpu_qlen set to 1000
[... cut ...]
Thu Feb 20 20:30:45 2020 daemon.notice procd: /etc/rc.d/S99v2ray_server:  * Running script '/etc/pptpd.include'
Thu Feb 20 20:30:45 2020 daemon.notice procd: /etc/rc.d/S99v2ray_server:  * Running script '/var/etc/unblockmusic.include'
Thu Feb 20 20:30:45 2020 daemon.notice procd: /etc/rc.d/S99v2ray_server:    ! Skipping due to path error: No such file or directory
Thu Feb 20 20:30:45 2020 daemon.info procd: - init complete -
Thu Feb 20 20:30:45 2020 user.info wsdd2[5818]: starting.
Thu Feb 20 20:30:57 2020 user.notice mwan3track[2835]: Stopping mwan3track for interface "wan2"

DongfeiSay, 2020-02-20:
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.237337] ACPI: Enabled 2 GPEs in block 00 to 07
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.240224] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-ff])
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.240615] acpi PNP0A03:00: _OSC: OS supports [Segments MSI]
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.241189] acpi PNP0A03:00: _OSC: not requesting OS control; OS requires [ExtendedConfig ASPM ClockPM MSI]
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.241988] PCI host bridge to bus 0000:00
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.242277] pci_bus 0000:00: root bus resource [io  0x0000-0x0cf7 window]
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.242697] pci_bus 0000:00: root bus resource [io  0x0d00-0xffff window]
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.243106] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.243605] pci_bus 0000:00: root bus resource [mem 0x80000000-0xfdffffff window]
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.244103] pci_bus 0000:00: root bus resource [bus 00-ff]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.244518] pci 0000:00:00.0: [8086:1237] type 00 class 0x060000
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.244966] pci 0000:00:01.0: [8086:7000] type 00 class 0x060100
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.245503] pci 0000:00:01.1: [8086:7111] type 00 class 0x01018a
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.245775] pci 0000:00:01.1: reg 0x20: [io  0xd000-0xd00f]
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.245886] pci 0000:00:01.1: legacy IDE quirk: reg 0x10: [io  0x01f0-0x01f7]
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.246325] pci 0000:00:01.1: legacy IDE quirk: reg 0x14: [io  0x03f6]
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.246723] pci 0000:00:01.1: legacy IDE quirk: reg 0x18: [io  0x0170-0x0177]
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.247149] pci 0000:00:01.1: legacy IDE quirk: reg 0x1c: [io  0x0376]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.247731] pci 0000:00:02.0: [15ad:0405] type 00 class 0x030000
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.248848] pci 0000:00:02.0: reg 0x10: [io  0xd010-0xd01f]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.249429] pci 0000:00:02.0: reg 0x14: [mem 0xf0000000-0xf0ffffff pref]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.250357] pci 0000:00:02.0: reg 0x18: [mem 0xf1000000-0xf11fffff]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.253614] pci 0000:00:03.0: [8086:100e] type 00 class 0x020000
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.254315] pci 0000:00:03.0: reg 0x10: [mem 0xf1200000-0xf121ffff]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.255563] pci 0000:00:03.0: reg 0x18: [io  0xd020-0xd027]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.259698] pci 0000:00:04.0: [80ee:cafe] type 00 class 0x088000
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.261194] pci 0000:00:04.0: reg 0x10: [io  0xd040-0xd05f]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.261909] pci 0000:00:04.0: reg 0x14: [mem 0xf1400000-0xf17fffff]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.262578] pci 0000:00:04.0: reg 0x18: [mem 0xf1800000-0xf1803fff pref]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.265272] pci 0000:00:05.0: [8086:2415] type 00 class 0x040100
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.265369] pci 0000:00:05.0: reg 0x10: [io  0xd100-0xd1ff]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.265416] pci 0000:00:05.0: reg 0x14: [io  0xd200-0xd23f]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.265829] pci 0000:00:06.0: [106b:003f] type 00 class 0x0c0310
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.266466] pci 0000:00:06.0: reg 0x10: [mem 0xf1804000-0xf1804fff]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.269556] pci 0000:00:07.0: [8086:7113] type 00 class 0x068000
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.269852] pci 0000:00:07.0: quirk: [io  0x4000-0x403f] claimed by PIIX4 ACPI
Thu Feb 20 20:41:11 2020 kern.info kernel: [    0.270368] pci 0000:00:07.0: quirk: [io  0x4100-0x410f] claimed by PIIX4 SMB
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.271040] pci 0000:00:08.0: [8086:100e] type 00 class 0x020000
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.271745] pci 0000:00:08.0: reg 0x10: [mem 0xf1820000-0xf183ffff]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.272991] pci 0000:00:08.0: reg 0x18: [io  0xd240-0xd247]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.275942] pci 0000:00:09.0: [8086:100e] type 00 class 0x020000
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.276683] pci 0000:00:09.0: reg 0x10: [mem 0xf1840000-0xf185ffff]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.277909] pci 0000:00:09.0: reg 0x18: [io  0xd248-0xd24f]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.280875] pci 0000:00:0a.0: [8086:100e] type 00 class 0x020000
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.281668] pci 0000:00:0a.0: reg 0x10: [mem 0xf1860000-0xf187ffff]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.282930] pci 0000:00:0a.0: reg 0x18: [io  0xd250-0xd257]
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.285526] pci 0000:00:0b.0: [8086:265c] type 00 class 0x0c0320
Thu Feb 20 20:41:11 2020 kern.debug kernel: [    0.286145] pci 0000:00:0b.0: reg 0x10: [mem 0xf1880000-0xf1880fff]
[... cut ...]
Thu Feb 20 20:41:27 2020 daemon.notice procd: /etc/rc.d/S99v2ray_server:    ! Skipping due to path error: No such file or directory
Thu Feb 20 20:41:27 2020 daemon.info procd: - init complete -
Thu Feb 20 20:41:27 2020 user.info wsdd2[5174]: starting.
Thu Feb 20 20:42:01 2020 daemon.err uhttpd[2187]: sh: write error: Broken pipe
Thu Feb 20 20:42:03 2020 user.notice mwan3track[2911]: Stopping mwan3track for interface "wan3"
Thu Feb 20 20:42:04 2020 user.notice mwan3track[3330]: Stopping mwan3track for interface "wan2"

coolsnowwolf, 2020-02-20:
DNS 问题

DongfeiSay, 2020-02-20:
> 直接给日志

L大还有一个关于ssr plus的问题，绕过大陆模式和全局模式不能用，绕过大陆模式=国外ip不能访问，全局模式=全部木大，全部网页无法连接互联网，访问控制里设置走全局代理的lan ip倒是能正常的使用全局代理模式，这个问题是我路由和虚拟机使用L大的openwrt乳酸菌都有出现的问题 GFW模式是正常的 别告诉我也是dns问题,dns是114dns

以下是日志
Thu Feb 20 21:27:54 2020 daemon.warn pdnsd[21984]: Caught signal 15. Exiting.
Thu Feb 20 21:27:57 2020 cron.info crond[22799]: crond (busybox 1.30.0) started, log level 8
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22990]: initializing ciphers... chacha20-ietf
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22990]: tcp port reuse enabled
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22990]: listening at 0.0.0.0:1234
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22990]: running from root user
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22992]: initializing ciphers... chacha20-ietf
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22992]: tcp port reuse enabled
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22992]: listening at 0.0.0.0:1234
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22992]: running from root user
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[23032]: initializing ciphers... chacha20-ietf
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[23032]: UDP relay enabled
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[23032]: udp port reuse enabled
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[23032]: TCP relay disabled
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[23032]: listening at 0.0.0.0:1234
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[23032]: running from root user
Thu Feb 20 21:27:57 2020 daemon.info pdnsd[23049]: pdnsd-1.2.9b-par starting.

然后检查连通性
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available...无数个

DongfeiSay, 2020-02-20:
> DNS 问题

L大还有一个关于ssr plus的问题，绕过大陆模式和全局模式不能用，绕过大陆模式=国外ip不能访问，全局模式=全部木大，全部网页无法连接互联网，访问控制里设置走全局代理的lan ip倒是能正常的使用全局代理模式，这个问题是我路由和虚拟机使用L大的openwrt一直以来都有出现的问题 GFW模式是正常的 别告诉我也是dns问题,dns是114dns 一样的的dns，17号的openwrt虚拟机是正常的

以下是日志
Thu Feb 20 21:27:54 2020 daemon.warn pdnsd[21984]: Caught signal 15. Exiting.
Thu Feb 20 21:27:57 2020 cron.info crond[22799]: crond (busybox 1.30.0) started, log level 8
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22990]: initializing ciphers... chacha20-ietf
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22990]: tcp port reuse enabled
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22990]: listening at 0.0.0.0:1234
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22990]: running from root user
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22992]: initializing ciphers... chacha20-ietf
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22992]: tcp port reuse enabled
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22992]: listening at 0.0.0.0:1234
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[22992]: running from root user
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[23032]: initializing ciphers... chacha20-ietf
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[23032]: UDP relay enabled
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[23032]: udp port reuse enabled
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[23032]: TCP relay disabled
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[23032]: listening at 0.0.0.0:1234
Thu Feb 20 21:27:57 2020 kern.info /usr/bin/ssr-redir[23032]: running from root user
Thu Feb 20 21:27:57 2020 daemon.info pdnsd[23049]: pdnsd-1.2.9b-par starting.

然后检查连通性
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available
Thu Feb 20 21:27:54 2020 kern.err /usr/bin/ssr-redir[21925]: accept: No file descriptors available...还有无数个

DongfeiSay, 2020-02-20:
在这里总结一下吧，负载均衡无法保存应用(一直在正在应用更改)的问题，无法保存应用的情况是因为openwrt使用dns ip上级路由器ip 一开始是动态获取ip地址，动态获取到的dns是上级路由ip(上级路由wan口设置的dns为114dns)并且openwrt wan口设置了跃点数，没有设置跃点数能正常保存应用，将openwrt的几个wan口设置为静态地址，dns设置为各大公共dns并且设置了跃点数可以正常保存应用，17号编译的openwrt动态获取dns同样是上级路由ip192.168.10.1并设置了跃点数，可以正常保存应用（不知道为什么。不过重启openwrt后接口全部是禁用状态，每次重启后需要手动进入管理页面在负载均衡里应用一次才会全部变绿，设置为静态地址 公共dns，重启直接全部绿色）

ssr plus大陆白名单和全局模式不能用不知道是不是个人问题

closed 2020-03-06 by coolsnowwolf